beagleboard/bb.org-overlays #155: Beagle Bone Blue - 4.19.103-bone-rt-r47 - Crash

Andrewiski, 2020-02-25:
@RobertCNelson 

I follow the instructions found at [imfatant/test](https://github.com/imfatant/test), seting up aurdopilot on a Beagle Bone Blue.  If I upgrade to 4.19.103-bone-rt-r47  kernel the board locks up about 3 minuts after boot.  If I stick with 4.14.108-ti-r124 no crash but no DSM I assume missing because lacking realtime kernal.

Any ideas why the beagle board would just crash and no longer accept here is the out put of version.sh before the board crashes.

debian@beagleblueandy:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[e457c010efc76c4e6fecccb7a5f7ff9b0597c4c0]
eeprom:[A335BNLTBLA21708EL000063]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-12-01]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g3d8c979660]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblue.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
kernel:[4.19.103-bone-rt-r47]
device-tree-override:[dtb=am335x-boneblue.dtb]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-I2C1-00A0.dtbo]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-UART4-00A0.dtbo]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-ADC-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200131.0-0rcnee0~stretch+20200131]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep remoteproc admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.022854] remoteproc remoteproc0: wkup_m3 is available
[    1.128232] remoteproc remoteproc0: powering up wkup_m3
[    1.128251] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.130469] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.741406] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    0.791372] pinctrl-single 44e10800.pinmux: pin PIN102 already requested by 48300180.eqep; cannot claim for 481a0000.spi
[    0.802333] pinctrl-single 44e10800.pinmux: pin-102 (481a0000.spi) status -22
[    0.809508] pinctrl-single 44e10800.pinmux: could not request pin 102 (PIN102) from group pinmux_spi1_pins  on device pinctrl-single
dmesg | grep gpio-of-helper
END

Andrewiski, 2020-02-26:
The only realtime kernel I could get to work on the beagle bone blue with ardupilot 4.0.3 is
 
sudo /opt/scripts/tools/update_kernel.sh --lts-4_9 --ti-rt-channel

 sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[8d8aa8dd017a0e9564a405f44e5590e47f7006b9]
eeprom:[A335BNLTBLA21708EL000063]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-12-01]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g3d8c979660]:[location: dd MBR]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
kernel:[4.9.147-ti-rt-r121]
device-tree-override:[dtb=am335x-boneblue.dtb]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-I2C1-00A0.dtbo]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-UART4-00A0.dtbo]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-ADC-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200131.0-0rcnee0~stretch+20200131]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep remoteproc admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.810534] remoteproc remoteproc0: wkup_m3 is available
[    1.958533] remoteproc remoteproc0: powering up wkup_m3
[    1.959155] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.965280] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.480036] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
END

RobertCNelson, 2020-02-27:
Sorry, haven't had much free time to work on these issue..

But just to follow up on this, been running some test on a BeagleBone Black Wireless..

With, 4.19.103-bone-rt-r47 installed, i can't even login..

```beaglebone login: [   67.592871] Bluetooth: HCI UART driver ver 2.3
[   67.600767] Bluetooth: HCI UART protocol H4 registered
[   67.613342] Bluetooth: HCI UART protocol LL registered
[   67.624450] Bluetooth: HCI UART protocol ATH3K registered
[   67.638821] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   86.487712] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   86.497162] Bluetooth: BNEP filters: protocol multicast
[   86.512440] Bluetooth: BNEP socket layer initialized
[   87.735270] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   87.803805] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   87.833213] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[   87.849114] cfg80211: failed to load regulatory.db
[   88.836526] wlcore: wl18xx HW: 183x or 180x, PG 2.2 (ROM 0x11)
[   88.910489] wlcore: loaded
[   89.473030] wlcore: PHY firmware version: Rev 8.2.0.0.242

Debian GNU/Linux 9 beaglebone ttyS0

BeagleBoard.org Debian Image 2018-10-07

Support/FAQ: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian

default username:password is [debian:temppwd]

beaglebone login: debian
temppwd
```

and just waiting. :(

Regards,

RobertCNelson, 2020-02-27:
If someone has some free time, here is the full list of kernels built in 4.19.x..

You can specify kernels via:

sudo /opt/scripts/tools/version.sh --kernel 4.19.1-bone-rt-r5

```
4.19.1-bone-rt-r5
4.19.2-bone-rt-r6
4.19.4-bone-rt-r7
4.19.4-bone-rt-r8
4.19.5-bone-rt-r9
4.19.6-bone-rt-r10
4.19.8-bone-rt-r11
4.19.9-bone-rt-r12
4.19.9-bone-rt-r13
4.19.10-bone-rt-r14
4.19.11-bone-rt-r15
4.19.12-bone-rt-r16
4.19.13-bone-rt-r17
4.19.14-bone-rt-r18
4.19.15-bone-rt-r19
4.19.17-bone-rt-r20
4.19.19-bone-rt-r21
4.19.23-bone-rt-r22
4.19.23-bone-rt-r23
4.19.24-bone-rt-r24
4.19.25-bone-rt-r25
4.19.26-bone-rt-r26
4.19.27-bone-rt-r27
4.19.28-bone-rt-r28
4.19.29-bone-rt-r29
4.19.31-bone-rt-r30
4.19.31-bone-rt-r31
4.19.37-bone-rt-r32
4.19.37-bone-rt-r33
4.19.37-bone-rt-r34
4.19.50-bone-rt-r35
4.19.59-bone-rt-r36
4.19.69-bone-rt-r37
4.19.71-bone-rt-r38
4.19.72-bone-rt-r39
4.19.72-bone-rt-r40
4.19.82-bone-rt-r41
4.19.90-bone-rt-r42
4.19.94-bone-rt-r43
4.19.94-bone-rt-r44
4.19.98-bone-rt-r45
4.19.100-bone-rt-r46
4.19.103-bone-rt-r47
```

4.19.103-bone-rt-r47 - locks up..

Regards,

Andrewiski, 2020-02-27:
@RobertCNelson Thanks didn't know I could be that specific I will work my way backwards see what works.

RobertCNelson, 2020-02-27:
It kinda looks like the RT patch broke this, now testing these two patches:

4.19.94-rt38 4.19.94-rt39

Andrewiski, 2020-02-27:
I do get few missing package errors well trying to change to the older releases of the kernels.  

E: Unable to locate package ti-sgx-ti335x-modules-4.19.26-bone-rt-r26
E: Couldn't find any package by glob 'ti-sgx-ti335x-modules-4.19.26-bone-rt-r26'
E: Couldn't find any package by regex 'ti-sgx-ti335x-modules-4.19.26-bone-rt-r26

The kernel does seem to build and load regardless and no crashes on r26

 sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[8d8aa8dd017a0e9564a405f44e5590e47f7006b9]
eeprom:[A335BNLTBLA21708EL000063]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-12-01]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g3d8c979660]:[location: dd MBR]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
kernel:[4.19.26-bone-rt-r26]
device-tree-override:[dtb=am335x-boneblue.dtb]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-I2C1-00A0.dtbo]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-UART4-00A0.dtbo]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-ADC-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200131.0-0rcnee0~stretch+20200131]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep remoteproc admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    0.970964] remoteproc remoteproc0: wkup_m3 is available
[    1.075578] remoteproc remoteproc0: powering up wkup_m3
[    1.075600] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.077760] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.719576] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
END

RobertCNelson, 2020-02-27:
Yeah, rt patech 4.19.94-rt39 broke it..

Only about 20 patches to diff..

https://git.kernel.org/pub/scm/linux/kernel/git/rt/linux-stable-rt.git/log/?h=v4.19.94-rt39-rebase

The latest safest for you to use is:

```sudo /opt/scripts/tools/update_kernel.sh --kernel 4.19.94-bone-rt-r43```

Regards,

RobertCNelson, 2020-02-27:
These two reverts on 4.19.94-rt39 make it work again..

https://github.com/RobertCNelson/bb-kernel/commit/f760168ba1efa4731e50bec0af0506143716e894

Regards,

Andrewiski, 2020-02-27:
Thank you for all the help and all you do for the Beaglebone community.

closed 2020-02-27 by Andrewiski

RobertCNelson, 2020-02-27:
I tagged 4.19.103-bone-rt-r48, it should be built out by later tonight..

RobertCNelson, 2020-02-28:
4.19.103-bone-rt-r48 is now in the repo, it's now safe to run:

```
sudo /opt/scripts/tools/update_kernel.sh --lts-4_19 --bone-rt-channel
```

Regards,