beagleboard/bb.org-overlays #156: Beagle Bone Blue /dev/ttyO4 missing

Andrewiski, 2020-03-01:
Even if I load BB-UART4-00A0.dtbo  an ls /dev/ty*  does not return a ttyO4.

Aurdiopilot is looking for DSM on /dev/tty04 

Line 156  
https://github.com/ArduPilot/ardupilot/blob/05a0fe615b50fc2fa8dba11e541ba4f5740a1181/libraries/AP_HAL_Linux/HAL_Linux_Class.cpp

I saw a really old port from Jason and you https://github.com/ArduPilot/ardupilot/issues/6641
but in the post the dsm port is refered to as /dev/ttyS2 ????

Schematic for BeableBone blue show DSM as uart4 RX  so I assume its /dev/ttyO4 

Any ways what am I missing that is preventing /dev/ttyO4  from being an device?

 sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[8d8aa8dd017a0e9564a405f44e5590e47f7006b9]
eeprom:[A335BNLTBLA21708EL000063]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-12-01]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g3d8c979660]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblue.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
kernel:[4.19.90-bone-rt-r42]
device-tree-override:[dtb=am335x-boneblue.dtb]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-I2C1-00A0.dtbo]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-UART4-00A0.dtbo]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-ADC-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200131.0-0rcnee0~stretch+20200131]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep remoteproc admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    0.987374] remoteproc remoteproc0: wkup_m3 is available
[    1.095633] remoteproc remoteproc0: powering up wkup_m3
[    1.095653] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.097816] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.740494] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
END

Andrewiski, 2020-03-02:
If I use the ti  versions /dev/ttyO4 is there but not in the bone versions of the kernel, is that intentional.   

I thought that I had to use the bone kernels for the pru code to work  AM335X-PRU-UIO-00A0.dtbo?

RobertCNelson, 2020-03-02:
@Andrewiski, no, you can use AM335X-PRU-UIO-00A0.dtbo on *-ti... Initially there was always bugs, but enough users eventually fixed it.. (head versions of v4.4.x-ti, v4.9.x-ti, v4.14.x-ti, v4,19.x-ti, all work with PRU-UIO..)

The "ti" bone dts files  "where" maintained by: https://github.com/StrawsonDesign/librobotcontrol/tree/master/device_tree

The bone ones are based on what I submitted mainline, and then with some features added... Right up to the point where i got tired of frying BeagleBone Blue's...

Regards,

Andrewiski, 2020-03-02:
Should the /dev/ttyO4 be there on the bone kernel or only the ti kernel?   The 4.19.x.ti kernel still has issues where it starts to run slow after about 3 minutes.  Your rollback did fix the 4.19.x.bone kernel so assume those roll backs haven't happen in the ti kernel as maintained by Strawson Design.

RobertCNelson, 2020-03-02:
He graduated last year, and got a full time job, so the Blue hasn't really had a maintainer for the last year.  I didn't see the same issue with v4.19.x-ti, but i'll work on a roll back.

Andrewiski, 2020-03-02:
Maybe I messed up my test but did a git pull in /opt/scripts then a 

sudo /opt/scripts/tools/update_kernel.sh --lts-4_19 --ti-rt-channel

but everything went to a crawl even the console was laggy.

I am happy to stick with the bone_rt kernel if I can get /dev/ttyO4 to work.  Almost wonder if there shouldn't be an Overlay for UART04_DSM where only the rx pin is enabled.

Andy

RobertCNelson, 2020-03-02:
No, you can't mess up the git pull... -ti has SMP enabled, so maybe it just doesn't show up as fast as bone (no SMP), which i only saw the issue on startup, so if it crawls after 3 minutes, i might have missed it..

Regards,

RobertCNelson, 2020-03-02:
Revert pushed: https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/ed16426f961fd0acab8f7491af8cb25a50a1a69b  now just waiting for the build test to finish, before i push to the build arm..

http://gfnd.rcn-ee.org:8080/view/Builds-v4.19.x-ti/

Regards,

RobertCNelson, 2020-03-02:
Tagged and pushed out: https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/ae94bbbd82eb2fdf2514c61ea57750d7779445c3