beagleboard/bb.org-overlays #157: BB-SPI0-AT86RF233-00A0.dtbo enabled results in pin-84 (48030000.spi) status -22

hideflolife, 2020-03-11:
I'm trying to enabled the BB-SPI)-AT86RF233 overlay within my /boot/uEnv.txt:
`#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.108-ti-r113
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
uboot_overlay_addr4=/lib/firmware/BB-SPI0-AT86RF233-00A0.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_rproc (4.19.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti, 4.19.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet overlayroot=tmpfs

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
`

However, I get the following error:
`
Loaded environment from /boot/uEnv.txt
Checking if uname_r is set in /boot/uEnv.txt...
gpio: pin 56 (gpio 56) value is 1
Running uname_boot ...
loading /boot/vmlinuz-4.14.108-ti-r113 ...
9929216 bytes read in 659 ms (14.4 MiB/s)
debug: [enable_uboot_overlays=1] ...
debug: [enable_uboot_cape_universal=1] ...
debug: [uboot_base_dtb_univ=] ...
loading /boot/dtbs/4.14.108-ti-r113/am335x-bonegreen.dtb ...
58411 bytes read in 69 ms (826.2 KiB/s)
uboot_overlays: [fdt_buffer=0x60000] ...
uboot_overlays: loading /lib/firmware/BB-SPI0-AT86RF233-00A0.dtbo ...
2390 bytes read in 171 ms (12.7 KiB/s)
uboot_overlays: loading /lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo ...
3675 bytes read in 339 ms (9.8 KiB/s)
loading /boot/initrd.img-4.14.108-ti-r113 ...
4696522 bytes read in 327 ms (13.7 MiB/s)
debug: [console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet] ...
debug: [bootz 0x82000000 0x88080000:47a9ca 88000000] ...
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Ramdisk to 8fb85000, end 8ffff9ca ... OK
   Loading Device Tree to 8fb13000, end 8fb84fff ... OK

Starting kernel ...

[    0.000813] timer_probe: no matching timers found
[    0.970421] wkup_m3_ipc 44e11324.wkup_m3_ipc: could not get rproc handle
[    1.009733] pinctrl-single 44e10800.pinmux: pin PIN84 already requested by 48024000.serial; cannot claim for 48030000.spi
[    1.021340] pinctrl-single 44e10800.pinmux: pin-84 (48030000.spi) status -22
[    1.028483] pinctrl-single 44e10800.pinmux: could not request pin 84 (PIN84) from group spi0_at86_s0  on device pinctrl-single
[    1.040017] omap2_mcspi 48030000.spi: Error applying setting, reverse things back
`

In looking within my am33xxdtsi, I do see the following - which appears to be conflicting but is disabled???

`uart2: serial@48024000 {
            compatible = "ti,am3352-uart", "ti,omap3-uart";
            ti,hwmods = "uart3";
            clock-frequency = <48000000>;
            reg = <0x48024000 0x2000>;
            interrupts = <74>;
            status = "disabled";
            dmas = <&edma 30 0>, <&edma 31 0>;
            dma-names = "tx", "rx";
        };`

Any guidance on what I might be doing wrong is appreciated?

RobertCNelson, 2020-03-12:
@hideflolife please include the output of:

```sudo /opt/scripts/tools/version.sh```

Regards,

hideflolife, 2020-03-12:
It's a new board that is based on BBG that I'm trying to help bring up, hence the EEPROM garbage:

```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[109f74fb87e6034ae1a8971a244064a8d5e090a5]
eeprom:[▒▒▒▒*]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Image 2019-08-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-dirty]:[location: dd MBR]
kernel:[4.14.108-ti-r113]
nodejs:[v6.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-SPI0-AT86RF233-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190801.0-0rcnee0~stretch+20190801]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.260778] remoteproc remoteproc0: wkup_m3 is available
[    1.350381] remoteproc remoteproc0: powering up wkup_m3
[    1.350517] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.354985] remoteproc remoteproc0: remote processor wkup_m3 is now up
[    7.482181] remoteproc remoteproc1: 4a334000.pru is available
[    7.498215] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[    7.415394] pruss 4a300000.pruss: creating PRU cores and other child platform devices
[    7.482181] remoteproc remoteproc1: 4a334000.pru is available
[    7.482339] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@34000 probed successfully
[    7.498215] remoteproc remoteproc2: 4a338000.pru is available
[    7.498392] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    0.965819] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    1.009932] pinctrl-single 44e10800.pinmux: pin PIN84 already requested by 48024000.serial; cannot claim for 48030000.spi
[    1.021551] pinctrl-single 44e10800.pinmux: pin-84 (48030000.spi) status -22
[    1.028677] pinctrl-single 44e10800.pinmux: could not request pin 84 (PIN84) from group spi0_at86_s0  on device pinctrl-single
dmesg | grep gpio-of-helper
[    0.967303] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

RobertCNelson, 2020-03-12:
@hideflolife with a fully up-to date stack, we get:

```debian@bbg-pwr07-ser15:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[ff7b834e4ea2608185c68aad6ce9da35f2ac4918]
eeprom:[A335BNLTBBG1BBG219102322]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian IoT Image 2020-03-02]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-14-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-SPI0-AT86RF233-00A0]
UBOOT: Loaded Overlay:[M-BB-BBG-00A0]
kernel:[4.14.108-ti-r127]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-SPI0-AT86RF233-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200306.0-0rcnee0~buster+20200306]
pkg:[bb-wl18xx-firmware]:[1.20200226.1-0rcnee0~buster+20200226]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100]
dmesg | grep remote
[    2.961104] remoteproc remoteproc0: wkup_m3 is available
[    3.215658] remoteproc remoteproc0: powering up wkup_m3
[    3.225170] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    3.259575] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   80.345810] remoteproc remoteproc1: 4a334000.pru is available
[   80.437839] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   79.903208] pruss 4a300000.pruss: creating PRU cores and other child platform devices
[   80.345810] remoteproc remoteproc1: 4a334000.pru is available
[   80.377847] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@34000 probed successfully
[   80.437839] remoteproc remoteproc2: 4a338000.pru is available
[   80.466025] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    1.104265] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.106319] gpio-of-helper ocp:cape-universal: Failed to get gpio property of 'P8_03'
[    1.106340] gpio-of-helper ocp:cape-universal: Failed to create gpio entry
[    3.208638] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

[    1.106319] gpio-of-helper ocp:cape-universal: Failed to get gpio property of 'P8_03'

RobertCNelson, 2020-03-12:
With the overlay disabled, 

```
debian@bbg-pwr07-ser15:~$ dmesg | grep gpio-of-helper
[    1.115044] gpio-of-helper ocp:cape-universal: ready
```

Okay, P8_03 wasn't specified..

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-SPI0-AT86RF233-00A0.dts#L52-L60

Let's convert the magic numbers to real values..

```
			bb_at86_pins: bb_at86_pins {
				pinctrl-single,pins = <
					0x040 0x27  /* irq | P9.15 gpio1[16] | MODE7 | INPUT | PULLDOWN */
					0x078 0x07  /* rst | P9.12 gpio1[28] | MODE7 | OUTPUT | PULLDOWN */
					0x044 0x07  /* slp | P9.23 gpio1[17] | MODE7 | OUTPUT | PULLDOWN */
				>;
			};
			spi0_at86_s0: spi0_at86_s0 {
				pinctrl-single,pins = <
					0x150 0x30  /* spi0_sclk, INPUT_PULLUP | MODE0 */
					0x154 0x30  /* spi0_d0, INPUT_PULLUP | MODE0 */
					0x158 0x10  /* spi0_d1, OUTPUT_PULLUP | MODE0 */
					0x15c 0x10  /* spi0_cs0, OUTPUT_PULLUP | MODE0 */
```

hideflolife, 2020-03-12:
@RobertCNelson I'm confused by the link you provided above for L52-L60, did you mean the lines associated with the AM33XX_PADCONF changes in the recent commit?

RobertCNelson, 2020-03-12:
Okay, cleaned up.. 

https://github.com/beagleboard/bb.org-overlays/commit/f6cdb3f7bb269d0644f8f1cc2822011cd38edecf

and

https://github.com/beagleboard/bb.org-overlays/commit/d93cd690ed2f1ef2ccbc176f4acb2868cf144b46

i think this was breaking things, it was also wrong...
```
	fragment@4 {
		reg = <2>;
		target = <&gpio1>;
		__overlay__{
			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <&bb_at86_pins>;
		};
	};
```

```debian@bbg-pwr07-ser15:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[5fa30ffcf068996ec08e034d2af5c5a133f10709]
eeprom:[A335BNLTBBG1BBG219102322]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian IoT Image 2020-03-02]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-14-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-SPI0-AT86RF233-00A0]
UBOOT: Loaded Overlay:[M-BB-BBG-00A0]
kernel:[4.14.108-ti-r127]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-SPI0-AT86RF233-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200306.0-0rcnee0~buster+20200306]
pkg:[bb-wl18xx-firmware]:[1.20200226.1-0rcnee0~buster+20200226]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100]
dmesg | grep remote
[    2.946252] remoteproc remoteproc0: wkup_m3 is available
[    3.183543] remoteproc remoteproc0: powering up wkup_m3
[    3.189013] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    3.201478] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   76.188011] remoteproc remoteproc1: 4a334000.pru is available
[   76.275858] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   75.639847] pruss 4a300000.pruss: creating PRU cores and other child platform devices
[   76.188011] remoteproc remoteproc1: 4a334000.pru is available
[   76.225875] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@34000 probed successfully
[   76.275858] remoteproc remoteproc2: 4a338000.pru is available
[   76.311310] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    1.103076] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.115054] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END```

Remember I DONT have the module...

```debian@bbg-pwr07-ser15:~$ dmesg | grep at86
[    8.142125] at86rf230 spi1.0: Non-Atmel dev found (MAN_ID 00 00)
[    8.236916] at86rf230: probe of spi1.0 failed with error -22
```

RobertCNelson, 2020-03-12:
@hideflolife , okay all pushed out, please test with:

```
sudo apt update
sudo apt install --only-upgrade bb-cape-overlays
sudo reboot
```

Regards,

hideflolife, 2020-03-12:
What you did is probably good, I think my problem is somewhere ttyS2 is being enabled - hence the conflict, I'm thinking in my u-boot build, though I'm not sure where yet:

```
debian@beaglebone:~$ dmesg | grep at86
[    1.027586] pinctrl-single 44e10800.pinmux: could not request pin 84 (PIN84) from group spi0_at86_s0  on device pinctrl-single
debian@beaglebone:~$
```
```
debian@beaglebone:~$ dmesg | grep [Ss]erial
[    0.971293] Serial: 8250/16550 driver, 6 ports, IRQ sharing disabled
[    0.976027] 44e09000.serial: ttyS0 at MMIO 0x44e09000 (irq = 30, base_baud = 3000000) is a 8250
[    0.989446] 48024000.serial: ttyS2 at MMIO 0x48024000 (irq = 31, base_baud = 3000000) is a 8250
[    1.009056] pinctrl-single 44e10800.pinmux: pin PIN84 already requested by 48024000.serial; cannot claim for 48030000.spi
[    1.139387] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.139411] usb usb1: SerialNumber: musb-hdrc.1
debian@beaglebone:~$
```

RobertCNelson, 2020-03-13:
@hideflolife, yeah sorry, i should have also checked the kernel version..

So i installed 4.14.108-ti-r113 to match your version, and it still works..

So... What did you change?

You did say a "custom" board based off BBG..

```
debian@bbg-pwr06-ser14:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[ff7b834e4ea2608185c68aad6ce9da35f2ac4918]
eeprom:[A335BNLTBBG1BBG219101925]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian IoT Image 2020-03-02]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-14-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-SPI0-AT86RF233-00A0]
UBOOT: Loaded Overlay:[M-BB-BBG-00A0]
kernel:[4.14.108-ti-r113]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-SPI0-AT86RF233-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200312.0-0rcnee0~buster+20200312]
pkg:[bb-wl18xx-firmware]:[1.20200226.1-0rcnee0~buster+20200226]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100]
dmesg | grep remote
[    2.966309] remoteproc remoteproc0: wkup_m3 is available
[    3.203481] remoteproc remoteproc0: powering up wkup_m3
[    3.208954] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    3.221430] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   56.826507] remoteproc remoteproc1: 4a334000.pru is available
[   56.911138] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   56.439047] pruss 4a300000.pruss: creating PRU cores and other child platform devices
[   56.826507] remoteproc remoteproc1: 4a334000.pru is available
[   56.853827] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@34000 probed successfully
[   56.911138] remoteproc remoteproc2: 4a338000.pru is available
[   56.945930] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    1.127953] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.140089] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

PS, since this is a custom board and it works for me, i'm going to close it, for our BeagleBoard.org issue tracker.. But the communication can continue on this thread.. ;)

Regards,

closed 2020-03-13 by RobertCNelson