beagleboard/bb.org-overlays #158: CAN1 doesn't work in BBB

Gregarieitor, 2020-03-12:
Hi, I have an issue with my beaglebone. I have tried both BB-CAN1 overlay and the cape-universal overlay, but I cannot see any transmissions on P9.24 and P9.26, when I load BB-CAN1 and I try to verify the pins I get the following error.

### root@beaglebone:/# config-pin -q P9.24
P9_24 pinmux file not found!
Cannot read pinmux file: /sys/devices/platform/ocp/ocp*P9_24_pinmux/state

And my /boot/uEnv.txt:

### #Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.108-ti-r113
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
uboot_overlay_addr4=/lib/firmware/BB-CAN1-00A0.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_rproc (4.19.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti, 4.19.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet

If you have any suggestion, I don't understand what's wrong in my configuration.

Thanks

RobertCNelson, 2020-03-12:
@Gregarieitor, that's expected..

```config-pin -q P9.24``` will not show any information, as you've enabled thru: uboot_overlay_addr4=/lib/firmware/BB-CAN1-00A0.dtbo

The can pins need loading, do you have a bus connected, do you have a termination resistor?

Please include the output of:

sudo /opt/scripts/tools/version.sh

Regards,

Gregarieitor, 2020-03-16:
Sorry, It is the first time that I work with the Beagle.

The output of sudo /opt/scripts/tools/version.sh:

### root@beaglebone:/# /opt/scripts/tools/version.sh
git:/opt/scripts/:[109f74fb87e6034ae1a8971a244064a8d5e090a5]
eeprom:[A335BNLT00C04219BBBK0577]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2019-08-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gbb4af0f50f]:[location: dd MBR]
kernel:[4.14.108-ti-r113]
nodejs:[v6.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-CAN1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[disable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190801.0-0rcnee0~stretch+20190801]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.118235] remoteproc remoteproc0: wkup_m3 is available
[    1.326513] remoteproc remoteproc0: powering up wkup_m3
[    1.326630] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.330561] remoteproc remoteproc0: remote processor wkup_m3 is now up
[    8.220157] remoteproc remoteproc1: 4a334000.pru is available
[    8.229565] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[    8.192869] pruss 4a300000.pruss: creating PRU cores and other child platform devices
[    8.220157] remoteproc remoteproc1: 4a334000.pru is available
[    8.220270] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@34000 probed successfully
[    8.229565] remoteproc remoteproc2: 4a338000.pru is available
[    8.229690] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    0.872475] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.873719] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
root@beaglebone:/#

I have connected a tja1040 transceiver to its resitor to canalyzer which is read, but it doesn't show any message. I don't know what I can be doing wrong.

Thanks for helping.

RobertCNelson, 2020-03-17:
@Gregarieitor , sorry for the delay.  The software side of the BBB looks fine, so It should be working.  That just leaves your hardware connections. Sorry i don't have the hardware in my posession at the moment to test this. I'll have to look into my junk box..

Regards,

Gregarieitor, 2020-03-18:
Thank you very much for your help,
I will also check again all the hardware to verify that everything is ok.

sigttou, 2020-09-15:
I have a comms cape and run into a similar issue:
```
$ sudo config-pin p9.24 can
P9_24 pinmux file not found!
bash: /sys/devices/platform/ocp/ocp*P9_24_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P9_24_pinmux/state
```
```
$ sudo config-pin p9.26 can
P9_26 pinmux file not found!
bash: /sys/devices/platform/ocp/ocp*P9_26_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P9_26_pinmux/state
```

```
$ cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.108-ti-r131
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
uboot_overlay_addr4=/lib/firmware/BB-CAN1-00A0.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_rproc (4.19.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
###pru_uio (4.14.x-ti, 4.19.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet video=HDMI-A-1:1024x768@60e

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
```

I've tried so far:
```
enable_uboot_cape_universial=0
```
```
uboot_overlay_addr4=/lib/firmware/BBORG_COMMS-00A2.dtbo
```

and /enabledisabling emmc, video, audio, wireless, adc.

Also using a different ISO (newer/older debian version) doesn't help.

```
$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLT000C2013BBBG0823]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Stretch imgtec Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-14-TI-00A0]
UBOOT: Loaded Overlay:[BB-CAN1-00A0]
UBOOT: Loaded Overlay:[BBORG_COMMS-00A2]
kernel:[4.14.108-ti-r131]
nodejs:[v6.17.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-CAN1-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200403.0-0rcnee0~stretch+20200403]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~stretch+20200322]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~stretch+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait uboot_detected_capes=BBORG_COMMS, coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    0.999931] remoteproc remoteproc0: wkup_m3 is available
[    1.083441] remoteproc remoteproc0: powering up wkup_m3
[    1.083562] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.086202] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   72.881380] remoteproc remoteproc1: 4a334000.pru is available
[   72.890269] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   72.040738] pruss 4a300000.pruss: creating PRU cores and other child platform devices
[   72.881380] remoteproc remoteproc1: 4a334000.pru is available
[   72.881507] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@34000 probed successfully
[   72.890269] remoteproc remoteproc2: 4a338000.pru is available
[   72.890392] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    0.678617] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.691133] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

RobertCNelson, 2020-09-15:
When you load a specific overlay, such as can1, those pins are no longer available thru config-pin.

Which cape do you have installed on the  board?

sigttou, 2020-09-15:
Ah, okay. So using the overlay should already provide me with the CAN1?

I have the BBORG_COMMS connected to the BBB. So I guess that Loaded overlay is enough and I can comment out the BB-CAN1 again?

RobertCNelson, 2020-09-15:
The BBORG_COMMS cape should have an eeprom, so you don't need to specify anything to load can1:

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BBORG_COMMS-00A2.dts

Are you not getting the "can" device under "sudo ifconfig -a"?

Regards,

sigttou, 2020-09-16:
thx. that's it. I was trying to follow the steps at https://github.com/beagleboard/capes/tree/master/beaglebone/Comms and didn't check there. Sorry for taking your time.

RobertCNelson, 2020-09-16:
That page is a little out of date.. if u-boot didn't pick up the overlay, then you could use those directions to enable it with no overlay..