beagleboard/bb.org-overlays #162: Config-pin -f gives No such file error

michieltjampens, 2020-04-20:
So i've been getting 
ERROR: open() for /sys/firmware/devicetree/base/ocp/P9_18_pinmux/pinctrl-names failed, No such file or directory

when trying to apply a config with config-pin -g .bbio. Decided to install the latest console image i found, but the result stays the same (meaning i didn't mess something up trying other stuff).

Output of version.h, only thing i changed on the uEnv is disable hdmi,audio, adc.
`eeprom:[A335BNLTEIA04219BBBK0E9A]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster Console Image 2020-04-06]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
kernel:[4.19.94-ti-r42]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200403.0-0rcnee0~buster+20200403]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~buster+20200322]
pkg:[kmod]:[26-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   26.244953] remoteproc remoteproc0: wkup_m3 is available
[   26.345536] remoteproc remoteproc0: powering up wkup_m3
[   26.345564] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   26.345824] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   28.861442] remoteproc remoteproc1: 4a334000.pru is available
[   28.893576] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   28.861442] remoteproc remoteproc1: 4a334000.pru is available
[   28.861597] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   28.893576] remoteproc remoteproc2: 4a338000.pru is available
[   28.893786] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.819100] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.820359] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 0bda:8178 Realtek Semiconductor Corp. RTL8192CU 802.11n WLAN Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END`

Is there something i forget to enable/disable/do?

RobertCNelson, 2020-04-20:
I don't have this model, looks like it's backordered:

https://www.newark.com/element14/bbone-black-ind-4g/silicon-manufacturer-texas-instruments/dp/76Y2810

RobertCNelson, 2020-04-20:
@michieltjampens ,

```
UBOOT: Booted Device-Tree:[am335x-boneblack.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
```

So cape-universal never was loaded..

I don't have this board on hand, sorry, i can't really do anything today..

Regards,

michieltjampens, 2020-04-20:
Hm, but uboot_overlay_options:[enable_uboot_cape_universal=1] does make a difference when i check the state of pins. So i thought it was loaded... Maybe something else instead then?

RobertCNelson, 2020-04-20:
@michieltjampens in u-boot it maps to a beaglebone black:

Default Black settings:

https://github.com/RobertCNelson/Bootloader-Builder/blob/master/patches/v2019.04/0002-U-Boot-BeagleBone-Cape-Manager.patch#L98-L106

EIA: basically u-boot prints out what it is..

https://github.com/RobertCNelson/Bootloader-Builder/blob/master/patches/v2019.04/0002-U-Boot-BeagleBone-Cape-Manager.patch#L154-L156

So yeah, it should be working, based on your /boot/uEnv.txt settings..

The only hail-marry option i can think of..
```
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
```

Maybe there's another older version of u-boot actually loading.. Fire up a serial port on J1 to quickly double check..

Regards,

michieltjampens, 2020-04-20:
> uboot_overlays: [uboot_base_dtb=am335x-boneblack-uboot.dtb] ...
> uboot_overlays: Switching too: dtb=am335x-boneblack-uboot.dtb ...
> loading /boot/dtbs/4.19.94-ti-r42/am335x-boneblack-uboot.dtb ...
> 57419 bytes read in 40 ms (1.4 MiB/s)
> uboot_overlays: [fdt_buffer=0x60000] ...
> uboot_overlays: uboot loading of [/lib/firmware/BB-ADC-00A0.dtbo] disabled by /b                                                  oot/uEnv.txt [disable_uboot_overlay_adc=1]...
> uboot_overlays: loading /lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo ...
> 1584 bytes read in 51 ms (30.3 KiB/s)
> uboot_overlays: uboot loading of [/lib/firmware/BB-HDMI-TDA998x-00A0.dtbo] disab                                                  led by /boot/uEnv.txt [disable_uboot_overlay_video=1]...
> uboot_overlays: loading /lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo ...
> 3801 bytes read in 29 ms (127.9 KiB/s)
> uboot_overlays: add [enable_uboot_cape_universal=1] to /boot/uEnv.txt to enable.                                                  ..
> loading /boot/initrd.img-4.19.94-ti-r42 ...
> 6585227 bytes read in 433 ms (14.5 MiB/s)
> debug: [console=ttyO0,115200n8 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait c                                                  oherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet] ..                                                  .
> debug: [bootz 0x82000000 0x88080000:647b8b 88000000] ...
> ## Flattened Device Tree blob at 88000000
>    Booting using the fdt blob at 0x88000000
>    Loading Ramdisk to 8f9b8000, end 8ffffb8b ... OK
>    Loading Device Tree to 8f946000, end 8f9b7fff ... OK
> 
> Starting kernel ...
> 
> [    0.000757] timer_probe: no matching timers found
> [    0.115695] l4_wkup_cm:clk:0010:0: failed to disable
> [    1.210502] omap_voltage_late_init: Voltage driver support not added
> 
> Debian GNU/Linux 10 beaglebone ttyS0
> 
> BeagleBoard.org Debian Buster Console Image 2020-04-06
> 
> Support: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian
> 
> default username:password is [debian:temppwd]
> 
> beaglebone login: [   26.135264] wkup_m3_ipc 44e11324.wkup_m3_ipc: could not get rproc handle
> [  122.864193] reboot: Power down
> 
> U-Boot SPL 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600)
> Trying to boot from MMC2
> Loading Environment from EXT4... Card did not respond to voltage select!
> 
> 
> U-Boot 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600), Build: jenkins-github_Bootloader-Builder-137
> 
> CPU  : AM335X-GP rev 2.1
> I2C:   ready
> DRAM:  512 MiB
> No match for driver 'omap_hsmmc'
> No match for driver 'omap_hsmmc'
> Some drivers were not found
> Reset Source: Power-on reset has occurred.
> RTC 32KCLK Source: External.
> MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
> Loading Environment from EXT4... Card did not respond to voltage select!
> Board: BeagleBone Black
> <ethaddr> not set. Validating first E-fuse MAC
> BeagleBone Black:
> Model: Element14 BeagleBone Black Industrial:
> BeagleBone: cape eeprom: i2c_probe: 0x54:
> BeagleBone: cape eeprom: i2c_probe: 0x55:
> BeagleBone: cape eeprom: i2c_probe: 0x56:
> BeagleBone: cape eeprom: i2c_probe: 0x57:
> Net:   eth0: MII MODE
> cpsw, usb_ether
> Press SPACE to abort autoboot in 0 seconds
> board_name=[A335BNLT] ...
> board_rev=[EIA0] ...
> Card did not respond to voltage select!
> Card did not respond to voltage select!
[... 37 lines cut ...]
> uboot_overlays: uboot loading of [/lib/firmware/BB-ADC-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot_overlay_adc=1]...
> uboot_overlays: loading /lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo ...
> 1584 bytes read in 51 ms (30.3 KiB/s)
> uboot_overlays: uboot loading of [/lib/firmware/BB-HDMI-TDA998x-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot_overlay_video=1]...
> uboot_overlays: loading /lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo ...
> 3801 bytes read in 30 ms (123 KiB/s)
> loading /boot/initrd.img-4.19.94-ti-r42 ...
> 6585227 bytes read in 433 ms (14.5 MiB/s)
> debug: [console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet] ...
> debug: [bootz 0x82000000 0x88080000:647b8b 88000000] ...
> ## Flattened Device Tree blob at 88000000
>    Booting using the fdt blob at 0x88000000
>    Loading Ramdisk to 8f9b8000, end 8ffffb8b ... OK
>    Loading Device Tree to 8f92d000, end 8f9b7fff ... OK
> 
> Starting kernel ...
> 
> [    0.002116] timer_probe: no matching timers found
> [    0.191318] l4_wkup_cm:clk:0010:0: failed to disable
> [    1.474372] omap_voltage_late_init: Voltage driver support not added
> 
> Debian GNU/Linux 10 beaglebone ttyS0
> 
> BeagleBoard.org Debian Buster Console Image 2020-04-06
>

RobertCNelson, 2020-04-20:
Okay that's the correct u-boot, did find a documentation bug:

```
UBOOT: Booted Device-Tree:[am335x-boneblack.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
kernel:[4.19.94-ti-r42]
```

```
debug: [uboot_base_dtb_univ=am335x-boneblack-uboot-univ.dtb] ...
uboot_overlays: [uboot_base_dtb=am335x-boneblack-uboot-univ.dtb] ...
uboot_overlays: Switching too: dtb=am335x-boneblack-uboot-univ.dtb ...
loading /boot/dtbs/4.19.94-ti-r42/am335x-boneblack-uboot-univ.dtb ...
162266 bytes read in 45 ms (3.4 MiB/s)
```

So it should say: UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]

so i wonder if it's a problem with config-pin

michieltjampens, 2020-04-20:
Hm, i've got a regular black, so tried to install the image on that one, all seemed to go fine but this appeared.
```
[  148.764305] systemd-udevd[1961]: Failed to wait for spawned command '/bin/chgrp -R gpio '/sys/devices/platform/ocp/44e0b000.i2c                                                     '': Input/output error
[  148.778412] systemd-udevd[1951]: Failed to wait for spawned command '/bin/chmod -R g=u '/sys/devices/platform/leds'': Input/out                                                     put error
[  148.792197] systemd-udevd[1952]: Failed to wait for spawned command '/bin/chgrp -R gpio '/sys/devices/platform/ocp/44e09000.ser                                                     ial'': Input/output error
[  148.808222] systemd-udevd[1950]: Failed to wait for spawned command '/bin/chgrp -R gpio '/sys/devices/platform/ocp/44e07000.gpi                                                     o'': Input/output error
[  148.821961] systemd-udevd[1965]: Failed to wait for spawned command '/bin/chmod -R g=u '/sys/devices/platform/4b000000.pmu'': I                                                     nput/output error
[  148.835641] systemd-udevd[1962]: Failed to wait for spawned command '/bin/chgrp -R gpio '/sys/devices/platform/ocp/40300000.ocm                                                     cram'': Input/output error
[  148.854712] systemd-udevd[1960]: Failed to wait for spawned command '/bin/chmod -R g=u '/sys/devices/platform/clk_mcasp0'': Inp                                                     ut/output error
[  148.868324] systemd-udevd[1967]: Failed to wait for spawned command '/bin/chmod -R g=u '/sys/devices/platform/fixedregulator0''                                                     : Input/output error
```
Is that normal?

RobertCNelson, 2020-04-20:
I think it is, we've got a few udev rules, but let me add "systemd-udevd" to my test farm..

http://gfnd.rcn-ee.org:81/farm/uptime/

RobertCNelson, 2020-04-20:
Oh right, console image.. let's check..

RobertCNelson, 2020-04-20:
ps, 'config-pin -f ' doesn't exist anymore:

```
debian@bbb-pwr03-ser11:~$ config-pin

GPIO Pin Configurator

Usage: config-pin -c <filename>
       config-pin -l <pin>
       config-pin -q <pin>
       config-pin <pin> <mode>
```

Currently installed version is: (c re-rewrite of a bash script..)

https://github.com/beagleboard/bb.org-overlays/tree/master/tools/pmunts_muntsos

The "old" bash version is:

https://github.com/beagleboard/bb.org-overlays/blob/master/tools/beaglebone-universal-io/config-pin

Regards,

michieltjampens, 2020-04-20:
Using config -c gives:
> ERROR: open() for /sys/devices/platform/ocp/ocp:OVERLAY_pinmux/state failed, No such file or directory
So slightly different error (this is on the regular bbb).

RobertCNelson, 2020-04-20:
Please attach your file, such that i can run it locally with 'config-pin -c'..

michieltjampens, 2020-04-20:
[BBIO.txt](https://github.com/beagleboard/bb.org-overlays/files/4504304/BBIO.txt)

RobertCNelson, 2020-04-20:
```
debian@bbb-pwr03-ser11:~$ config-pin -c BBIO.txt 
ERROR: open() for /sys/devices/platform/ocp/ocp:OVERLAY_pinmux/state failed, No such file or directory
```

Nuke these lines: 
```
overlay cape-univ-hdmi
overlay cape-universal
```

Kernel overlays are NOT supported..

Next change:
```
P8_7 default
P8_8 default
P8_9 default
```
```
P8_07 default
P8_08 default
P8_09 default
```

RobertCNelson, 2020-04-20:
```
debian@bbb-pwr03-ser11:~$ config-pin 

GPIO Pin Configurator

Usage: config-pin -c <filename>
       config-pin -l <pin>
       config-pin -q <pin>
       config-pin <pin> <mode>

debian@bbb-pwr03-ser11:~$ config-pin -l P9.28

Available modes for P9_28 are: default gpio gpio_pu gpio_pd gpio_input spi_cs pwm pwm2 pruout pruin

debian@bbb-pwr03-ser11:~$ config-pin -c BBIO.txt 
ERROR: write() to /sys/devices/platform/ocp/ocp:P9_28_pinmux/state failed, No such device
debian@bbb-pwr03-ser11:~$ config-pin -l P9.28

Available modes for P9_28 are: default gpio gpio_pu gpio_pd gpio_input spi_cs pwm pwm2 pruout pruin

debian@bbb-pwr03-ser11:~$ config-pin -l P9_28

Available modes for P9_28 are: default gpio gpio_pu gpio_pd gpio_input spi_cs pwm pwm2 pruout pruin

debian@bbb-pwr03-ser11:~$ cat /sys/devices/platform/ocp/ocp\:P9_28_pinmux/state 
default
```

michieltjampens, 2020-04-20:
Tried removing those overlay lines before posting, didnt make a difference. But didnt pay attention to those missing 0's.
Ill test later.

RobertCNelson, 2020-04-20:
```
debian@bbb-pwr03-ser11:~$ config-pin -l P9_28

Available modes for P9_28 are: default gpio gpio_pu gpio_pd gpio_input spi_cs pwm pwm2 pruout pruin
```

```
P9_28 spi -> spi_cs
```

```
debian@bbb-pwr03-ser11:~$ config-pin -c BBIO.txt 
ERROR: write() to /sys/devices/platform/ocp/ocp:P9_31_pinmux/state failed, No such device
```

```
debian@bbb-pwr03-ser11:~$ config-pin -l P9_31

Available modes for P9_31 are: default gpio gpio_pu gpio_pd gpio_input spi_sclk pwm pruout pruin
```

```
P9_31 spi -> spi_sclk
```

All fixed..

```
P8_07 default
P8_08 default
P8_09 default
P8_10 default
P8_11 default
P8_12 default
P8_13 default
P8_14 default
P8_15 default
P8_16 default
P8_17 default
P8_18 default
P8_19 default
P8_26 default
P8_27 default
P8_28 default
P8_29 default
P8_30 default
P8_31 gpio
P8_32 gpio
P8_33 gpio
P8_34 gpio
P8_35 gpio
P8_36 gpio
P8_37 uart
P8_38 uart
P8_39 default
P8_40 default
P8_41 default
P8_42 default
P8_43 default
P8_44 default
P8_45 default
P8_46 default
P9_11 uart
P9_12 default
P9_13 uart
P9_14 default
P9_15 default
P9_16 default
P9_17 i2c
P9_18 i2c
P9_21 uart
P9_22 uart
P9_23 default
P9_24 uart
P9_25 default
P9_26 uart
P9_27 default
P9_28 spi_cs
P9_29 spi
P9_30 spi
P9_31 spi_sclk
P9_41 default
P9_42 default
P9_91 default
P9_92 default
```

RobertCNelson, 2020-04-20:
P9_28/P9_31 was renamed years ago...  That config only ever worked on 3.8.x based kernels...

RobertCNelson, 2020-04-20:
```
debian@bbb-pwr03-ser11:~$ config-pin -c BBIO.txt 
debian@bbb-pwr03-ser11:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[b61125c1485bee929340cacc06c85c6fcfd678bc]
eeprom:[A335BNLT0A5C3313BBBK2518]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
kernel:[4.19.94-ti-r42]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200403.0-0rcnee0~buster+20200403]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~buster+20200322]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100]
dmesg | grep remote
[   66.679315] remoteproc remoteproc0: wkup_m3 is available
[   66.854179] remoteproc remoteproc0: powering up wkup_m3
[   66.863737] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   66.884627] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   68.615941] remoteproc remoteproc1: 4a334000.pru is available
[   68.682819] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   68.615941] remoteproc remoteproc1: 4a334000.pru is available
[   68.639814] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   68.682819] remoteproc remoteproc2: 4a338000.pru is available
[   68.706320] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.926562] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.940051] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

RobertCNelson, 2020-04-20:
```
debian@bbb-pwr03-ser11:~$ sudo /opt/scripts/device/bone/show-pins.pl 
P8.25 / eMMC d0                    0  U7 fast rx  up  1 mmc 1 d0         mmc@481d8000 (pinmux_emmc_pins)
P8.24 / eMMC d1                    1  V7 fast rx  up  1 mmc 1 d1         mmc@481d8000 (pinmux_emmc_pins)
P8.05 / eMMC d2                    2  R8 fast rx  up  1 mmc 1 d2         mmc@481d8000 (pinmux_emmc_pins)
P8.06 / eMMC d3                    3  T8 fast rx  up  1 mmc 1 d3         mmc@481d8000 (pinmux_emmc_pins)
P8.23 / eMMC d4                    4  U8 fast rx  up  1 mmc 1 d4         mmc@481d8000 (pinmux_emmc_pins)
P8.22 / eMMC d5                    5  V8 fast rx  up  1 mmc 1 d5         mmc@481d8000 (pinmux_emmc_pins)
P8.03 / eMMC d6                    6  R9 fast rx  up  1 mmc 1 d6         mmc@481d8000 (pinmux_emmc_pins)
P8.04 / eMMC d7                    7  T9 fast rx  up  1 mmc 1 d7         mmc@481d8000 (pinmux_emmc_pins)
P8.19                              8 U10 fast rx down 7 gpio 0.22        ocp/P8_19_pinmux (pinmux_P8_19_default_pin)
P8.13                              9 T10 fast rx down 7 gpio 0.23        ocp/P8_13_pinmux (pinmux_P8_13_default_pin)
P8.14                             10 T11 fast rx down 7 gpio 0.26        ocp/P8_14_pinmux (pinmux_P8_14_default_pin)
P8.17                             11 U12 fast rx down 7 gpio 0.27        ocp/P8_17_pinmux (pinmux_P8_17_default_pin)
P8.12                             12 T12 fast rx down 7 gpio 1.12        ocp/P8_12_pinmux (pinmux_P8_12_default_pin)
P8.11                             13 R12 fast rx down 7 gpio 1.13        ocp/P8_11_pinmux (pinmux_P8_11_default_pin)
P8.16                             14 V13 fast rx down 7 gpio 1.14        ocp/P8_16_pinmux (pinmux_P8_16_default_pin)
P8.15                             15 U13 fast rx down 7 gpio 1.15        ocp/P8_15_pinmux (pinmux_P8_15_default_pin)
P9.15                             16 R13 fast rx down 7 gpio 1.16        ocp/P9_15_pinmux (pinmux_P9_15_default_pin)
P9.23                             17 V14 fast rx down 7 gpio 1.17        ocp/P9_23_pinmux (pinmux_P9_23_default_pin)
P9.14                             18 U14 fast rx down 7 gpio 1.18        ocp/P9_14_pinmux (pinmux_P9_14_default_pin)
P9.16                             19 T14 fast rx down 7 gpio 1.19        ocp/P9_16_pinmux (pinmux_P9_16_default_pin)
P9.11                             28 T17 fast rx  up  6 uart 4 rxd       ocp/P9_11_pinmux (pinmux_P9_11_uart_pin)
P9.13                             29 U17 fast rx  up  6 uart 4 txd       ocp/P9_13_pinmux (pinmux_P9_13_uart_pin)
P9.12                             30 U18 fast rx  up  7 gpio 1.28        ocp/P9_12_pinmux (pinmux_P9_12_default_pin)
P8.26                             31  V6 fast rx  up  7 gpio 1.29        ocp/P8_26_pinmux (pinmux_P8_26_default_pin)
P8.21 / eMMC clk                  32  U9 fast rx  up  2 mmc 1 clk        mmc@481d8000 (pinmux_emmc_pins)
P8.20 / eMMC cmd                  33  V9 fast rx  up  2 mmc 1 cmd        mmc@481d8000 (pinmux_emmc_pins)
P9.15                             34 T13 fast rx  up  7 gpio 2.00
P8.18                             35 V12 fast rx down 7 gpio 2.01        ocp/P8_18_pinmux (pinmux_P8_18_default_pin)
P8.07                             36  R7 fast rx  up  7 gpio 2.02        ocp/P8_07_pinmux (pinmux_P8_07_default_pin)
P8.08                             37  T7 fast rx  up  7 gpio 2.03        ocp/P8_08_pinmux (pinmux_P8_08_default_pin)
P8.10                             38  U6 fast rx  up  7 gpio 2.04        ocp/P8_10_pinmux (pinmux_P8_10_default_pin)
P8.09                             39  T6 fast rx  up  7 gpio 2.05        ocp/P8_09_pinmux (pinmux_P8_09_default_pin)
P8.45 / hdmi / sysboot 0          40  R1 fast rx down 7 gpio 2.06        ocp/P8_45_pinmux (pinmux_P8_45_default_pin)
P8.46 / hdmi / sysboot 1          41  R2 fast rx down 7 gpio 2.07        ocp/P8_46_pinmux (pinmux_P8_46_default_pin)
P8.43 / hdmi / sysboot 2          42  R3 fast rx down 7 gpio 2.08        ocp/P8_43_pinmux (pinmux_P8_43_default_pin)
P8.44 / hdmi / sysboot 3          43  R4 fast rx down 7 gpio 2.09        ocp/P8_44_pinmux (pinmux_P8_44_default_pin)
P8.41 / hdmi / sysboot 4          44  T1 fast rx down 7 gpio 2.10        ocp/P8_41_pinmux (pinmux_P8_41_default_pin)
P8.42 / hdmi / sysboot 5          45  T2 fast rx down 7 gpio 2.11        ocp/P8_42_pinmux (pinmux_P8_42_default_pin)
P8.39 / hdmi / sysboot 6          46  T3 fast rx down 7 gpio 2.12        ocp/P8_39_pinmux (pinmux_P8_39_default_pin)
P8.40 / hdmi / sysboot 7          47  T4 fast rx down 7 gpio 2.13        ocp/P8_40_pinmux (pinmux_P8_40_default_pin)
P8.37 / hdmi / sysboot 8          48  U1 fast rx  up  4 uart 5 txd       ocp/P8_37_pinmux (pinmux_P8_37_uart_pin)
P8.38 / hdmi / sysboot 9          49  U2 fast rx  up  4 uart 5 rxd       ocp/P8_38_pinmux (pinmux_P8_38_uart_pin)
P8.36 / hdmi / sysboot 10         50  U3 fast rx      7 gpio 2.16        ocp/P8_36_pinmux (pinmux_P8_36_gpio_pin)
P8.34 / hdmi / sysboot 11         51  U4 fast rx      7 gpio 2.17        ocp/P8_34_pinmux (pinmux_P8_34_gpio_pin)
P8.35 / hdmi / sysboot 12         52  V2 fast rx      7 gpio 0.08        ocp/P8_35_pinmux (pinmux_P8_35_gpio_pin)
P8.33 / hdmi / sysboot 13         53  V3 fast rx      7 gpio 0.09        ocp/P8_33_pinmux (pinmux_P8_33_gpio_pin)
P8.31 / hdmi / sysboot 14         54  V4 fast rx      7 gpio 0.10        ocp/P8_31_pinmux (pinmux_P8_31_gpio_pin)
P8.32 / hdmi / sysboot 15         55  T5 fast rx      7 gpio 0.11        ocp/P8_32_pinmux (pinmux_P8_32_gpio_pin)
[... cut ...]
P9.42b                           104 B12 fast rx down 7 gpio 3.18        ocp/P9_92_pinmux (pinmux_P9_92_default_pin)
P9.27                            105 C13 fast rx down 7 gpio 3.19        ocp/P9_27_pinmux (pinmux_P9_27_default_pin)
P9.41                            106 D13 fast rx down 7 gpio 3.20        ocp/P9_91_pinmux (pinmux_P9_91_default_pin)
P9.25 / audio osc                107 A14 fast rx down 7 gpio 3.21        ocp/P9_25_pinmux (pinmux_P9_25_default_pin)
P9.41 / jtag emu3                109 D14 fast rx down 7 gpio 0.20        ocp/P9_41_pinmux (pinmux_P9_41_default_pin)
```

michieltjampens, 2020-04-20:
Thanks a lot! Now it indeed seems to be applied.
Cape universal is responsible for not being able to use mem sleep properly (works if not enabled).
Can config-pin make a pin 'unclaimed' again? (only way that mem sleep seems to work if cape universal is not applied).

RobertCNelson, 2020-04-20:
Oh, by using cape-universal, we throw away any chance of gpio, mem sleep, or ANY power management working..

If you think about it, every "Pinmux" possible is enable with cape-universal... Thus All usarts, all spi, all pwm, all timers, all i2c, all can, Everything is running on the core..  All cape-universal does, connects the physical pin to the pinmux connection on the peripherals. ;)

If you want real power management, you need to just create your own overlay, apply it to the base tree, and boot that.

Your choice, ease of use, or power management, by default we've chosen ease of use for BeagleBoard.org.

Regards,

michieltjampens, 2020-04-20:
Ok, saves me the trying to figure it out then :p
So if the use case needs sleep etc i have to make the appropriate cape.