beagleboard/bb.org-overlays #163: unable to configuration SPI pins

aljandali, 2020-04-29:
Unable to configure pins BBB

I am beginner with Beaglebone and would like to use SPI to transfer data to a Microcontroller , but i have a problem
When I try to configure any of the pins I get the following errors:

GPIO:
debian@beaglebone:~$ config-pin P9_17 spi_cs
P9_17 pinmux file not found!
bash: /sys/devices/platform/ocp/ocp*P9_17_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P9_17_pinmux/state

debian@beaglebone:~$ config-pin P9_18 spi                                                                                                                        
P9_18 pinmux file not found!
bash: /sys/devices/platform/ocp/ocp*P9_18_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P9_18_pinmux/state
debian@beaglebone:~$

i use this Image : AM3358 Debian 9.12 2020-04-06 4GB SD ImgTec
the Kernel : 4.4.155-ti-r155

the output of sudo /opt/scripts/tools/version.sh : 

git:/opt/scripts/:[b61125c1485bee929340cacc06c85c6fcfd678bc]
eeprom:[A335BNLT000C1818BBBG1658]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Stretch imgtec Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-14-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
UBOOT: Loaded Overlay:[BB-SPIDEV0-00A0]
UBOOT: Loaded Overlay:[BB-SPIDEV1-00A0]
kernel:[4.4.155-ti-r155]
nodejs:[v6.17.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr5=/lib/firmware/BB-SPIDEV1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200403.2-0~stretch+20200424]
pkg:[bb-wl18xx-firmware]:[1.20200420.0-0rcnee0~stretch+20200420]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~stretch+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.258967]  remoteproc0: wkup_m3 is available
[    1.258984]  remoteproc0: Note: remoteproc is still under development and considered experimental.
[    1.258992]  remoteproc0: THE BINARY FORMAT IS NOT YET FINALIZED, and backward compatibility isn't yet guaranteed.
[    1.363565]  remoteproc0: powering up wkup_m3
[    1.364244]  remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.372252]  remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.055156] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[   45.234871] pinctrl-single 44e10800.pinmux: pin 44e1099c.0 already requested by 481a0000.spi; cannot claim for 48038000.mcasp
[   45.246461] pinctrl-single 44e10800.pinmux: pin-103 (48038000.mcasp) status -22
[   45.253891] pinctrl-single 44e10800.pinmux: could not request pin 103 (44e1099c.0) from group mcasp0_pins  on device pinctrl-single
dmesg | grep gpio-of-helper
[    1.056238] gpio-of-helper ocp:cape-universal: Allocated GPIO id=0
[    1.056391] gpio-of-helper ocp:cape-universal: Allocated GPIO id=1
[    1.056402] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

What am I doing wrong?

Best regards,

Jan

RobertCNelson, 2020-04-29:
@aljandali you already loaded them:

```
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-14-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
UBOOT: Loaded Overlay:[BB-SPIDEV0-00A0]
UBOOT: Loaded Overlay:[BB-SPIDEV1-00A0]
```

```
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr5=/lib/firmware/BB-SPIDEV1-00A0.dtbo]
```
Thus don't use "config-pin" they are already set...

But.....
```
[ 1.055156] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[ 45.234871] pinctrl-single 44e10800.pinmux: pin 44e1099c.0 already requested by 481a0000.spi; cannot claim for 48038000.mcasp
[ 45.246461] pinctrl-single 44e10800.pinmux: pin-103 (48038000.mcasp) status -22
[ 45.253891] pinctrl-single 44e10800.pinmux: could not request pin 103 (44e1099c.0) from group
```
You have a conflict, probably HDMI-AUDIO..  see:

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Disable_on-board_devices

```
kernel:[4.4.155-ti-r155]
```
Really? Sorry, i no longer test anything before "v4.14.x-ti".. Why can't you use v4.14.x-ti?

RobertCNelson, 2020-04-29:
```
i use this Image : AM3358 Debian 9.12 2020-04-06 4GB SD ImgTec
the Kernel : 4.4.155-ti-r155
```
WHY?

The "ImgTec" image was custom built for the v4.14.x-ti provided kernel and v4.14.x-ti SGX modules..

There's atleast ~ 1-2Gig's of wasted space that won't work on the v4.4.x-ti!!!!

aljandali, 2020-04-29:
thanks for replay 
Ok what should i to do ? !!!

RobertCNelson, 2020-04-29:
Why did you first downgrade to v4.4.x?

SPI works fine in v4.14.x...

Are you following some guide?

You set:

```
uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo
uboot_overlay_addr5=/lib/firmware/BB-SPIDEV1-00A0.dtbo
```

Which would give you SPI out of the box without changing the kernel..

But you have to disable hdmi-audio as that conflicets with spidev1..

```
disable_uboot_overlay_audio=1
```

So no reason to downgrade to v4.4.x

Regards,

aljandali, 2020-04-29:
yes i'm following some guide , 
thanks for your fast replay , now if i test the SPI interface , i have a strange false data and the output is [0,0,0,0,] 

> from Adafruit_BBIO.SPI import SPI
> 
> print("SPI(1,0)")
> spi = SPI(1,0)
> print(spi.xfer2([32, 11, 110, 22, 220]))
> spi.close() 

the output is : 

> SPI(1,0)
> [0, 0, 0, 0, 0]

if i connect it with logic analyser i get a strange data 

What am I doing wrong?

Best regards,

jan

RobertCNelson, 2020-04-29:
@aljandali sorry, wrong bug list, please talk to @pdp7  at : https://github.com/adafruit/adafruit-beaglebone-io-python/issues

closed 2020-04-29 by RobertCNelson

aljandali, 2020-04-29:
> Why did you first downgrade to v4.4.x?
how can i upgrade to  v4.14.x... ? 
should i do it ?

aljandali, 2020-04-29:
> Why did you first downgrade to v4.4.x?
>
> SPI works fine in v4.14.x...
>
> Are you following some guide?
>
> You set:
>
> ```
> uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo [...]

> Why did you first downgrade to v4.4.x?
how can i upgrade to v4.14.x... ?
should i do it ?

RobertCNelson, 2020-04-29:
Just grab the microSD iot image from:

https://elinux.org/Beagleboard:Latest-images-testing#Debian_9_.28Stretch.29_IoT

then in /boot/uEnv.txt set:

```
uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo
uboot_overlay_addr5=/lib/firmware/BB-SPIDEV1-00A0.dtbo
disable_uboot_overlay_audio=1
```

Regards,

aljandali, 2020-04-29:
thanks RobertCNelson 
regards ,

dayrenefrometa, 2020-12-18:
> yes i'm following some guide ,
> thanks for your fast replay , now if i test the SPI interface , i have a strange false data and the output is [0,0,0,0,]
>
>
> the output is :
>
>
> if i connect it with logic [...]

Hi aljandali, 

I am facing the same problem that you mentioned here. Could you please indicate if you solve it? I would really appreciate any help from your side, I have been working on solving this problem but the information on the Internet is not that clear and no success so far.

Thank you very much in advance for your help,

Regards,

Dayrene