beagleboard/bb.org-overlays #166: RS485

erji44, 2020-05-14:
I have purchased a cape made by GHI have been trying to make it work as a serial port RS485 but so far no avail.
In the file uEnv.txt I have enabled the following line:
uboot_overlay_addr0=/lib/firmware/BBORG_COMMS-00A2.dtbo
BBORG_COMMS-00A2.dtbo is present in /lib/firmware

When I try to send a message over serial echo message > /dev/ttyO4 on other end where I have another 485 usb to serial all I get is unrelated single letters or digits such as 0, H, (, ) etc but not the message I'm sending.
When attaching a USB to rs485 on the beaglebone I'm getting the desired results on the other end which sits on another machine i.e
echo message > /dev/ttyUSB0   on other machine message is being received exactly as send.
I'm having exactly the same results on latest debian image 10.3 and previous debian release 9.8
Any ideas as I've spent 3 days with this but just can't get the cape to work.

Thank you for your time

RobertCNelson, 2020-05-14:
@erji44 the overlay looks in-complete for rs485:

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BBORG_COMMS-00A2.dts#L101-L104

For reference, which usb rs 485 adapter are you using? i really need to setup some kind of rs485 test device...

Regards,

erji44, 2020-05-15:
Hi Nelson, thank you for the answer
I'm using two different types one is based on the prolific chip PL2303
And the other is newer based on CH340 chip
They look like this:
https://www.ebay.co.uk/itm/402262001885?hash=item5da8af08dd:g:hHwAAOSw7thbdR0S

Regards