beagleboard/bb.org-overlays #167: Beaglebone can't boot from emmc

jerryyip, 2020-05-22:
A clear and concise description of what the bug is.
Recently(May 2020) I built a image for my beaglebone green. After I flashed the new image to the emmc, I found that my board couldn't boot from emmc and stayed at u-boot.
Here is something strange I found in dmesg:
```
[    1.739502] omap_hsmmc 48060000.mmc: Got CD GPIO
[    1.780527] omap_hsmmc 481d8000.mmc: prop pinctrl-0 index 0 invalid phandle
[    1.787723] omap_hsmmc: probe of 481d8000.mmc failed with error -22
```

Can anyone tell me what is happening to my board? Would it be a problem of the device trees? I build this image with `apt install overlayroot bb-cape-overlays` from rcn-ee. And when I reflashed the previous built image to emmc, it could work again.

RobertCNelson, 2020-05-22:
@jerryyip please include the output of:

```
sudo /opt/scripts/tools/version.sh 
```
to help aid in debugging.

Regards,

PS, while i no longer test v4.4.x/v4.9.x, (the recommended minimal is now v4.14.x) they should still work at this point.

jerryyip, 2020-05-24:
@RobertCNelson As my borad can't boot from emmc, I also build a image which boots from SD card with the save config file. And here is the output of version.sh:
```
sudo /opt/scripts/tools/version.sh
eeprom:[A335BNLTGW1A]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[Seeed SenseCap Gateway Debian Image 2020-05-17-891422]
kernel:[4.9.80-ti-r98]
nodejs:[v8.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200430.0-0~stretch+20200430]
pkg:[bb-wl18xx-firmware]:[1.20200420.1-0~stretch+20200424]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p3 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet root=/dev/mmcblk0p3 rw rootfstype=ext4 rootwait cape_universal=enable]
dmesg | grep remote
[    1.805791] remoteproc remoteproc0: wkup_m3 is available
[    1.937823] remoteproc remoteproc0: powering up wkup_m3
[    1.937852] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217148
[    1.944101] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   64.196792] remoteproc remoteproc1: 4a334000.pru0 is available
[   64.204957] remoteproc remoteproc2: 4a338000.pru1 is available
dmesg | grep pru
[   63.769349] ti-pruss 4a300000.pruss: creating PRU cores and other child platform devices
[   63.809029] irq: no irq domain found for /ocp/pruss_soc_bus@4a326000/pruss@4a300000/intc@4a320000 !
[   63.810815] irq: no irq domain found for /ocp/pruss_soc_bus@4a326000/pruss@4a300000/intc@4a320000 !
[   64.196792] remoteproc remoteproc1: 4a334000.pru0 is available
[   64.196919] pru-rproc 4a334000.pru0: PRU rproc node /ocp/pruss_soc_bus@4a326000/pruss@4a300000/pru@4a334000 probed successfully
[   64.204957] remoteproc remoteproc2: 4a338000.pru1 is available
[   64.205081] pru-rproc 4a338000.pru1: PRU rproc node /ocp/pruss_soc_bus@4a326000/pruss@4a300000/pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    1.483161] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.495152] gpio-of-helper ocp:cape-universal: ready
END
```

RobertCNelson, 2020-05-25:
@jerryyip 

```
pkg:[bb-cape-overlays]:[4.14.20200430.0-0~stretch+20200430] leads to the problem? It this means that I have to update kernel to 4.14?
```
v4.14.x added a new Bluetooth driver and substantial hdmi device tree changes..  For v4.4.x/v4.9.x we made Bluetooth work with a "fake" led and a bootup-script, very fragile..  With v4.14.x we could use the serdev driver and it would just work.. (but this doesn't exist on v4.4.x/v4.9.x) So yes v4.14.x is highly recommended.. The next 'large' breakage for our bb-cape-overlays package shouldn't occur till now (v5.8.x) as TI changed the MMC/SDIO driver..  So that'll take a few years become the new default..

```
sudo /opt/scripts/tools/version.sh
eeprom:[A335BNLTGW1A]
model:[TI_AM335x_BeagleBone_Black]
```
This look wrong...  "GW1A" is a BeagleBone Green Wireless.. Can you please confirm which board you actually have, might have to re-flash the firmware..

HEre are some pretty pictures to help:

Green:
https://www.seeedstudio.com/Seeed-Studio-BeagleBoner-Green.html

Green Wireless:
https://www.seeedstudio.com/Seeed-Studio-BeagleBoner-Green-Wireless-Development-Board-TI-AM335x-WiFi-BT.html

Regards,

jerryyip, 2020-05-27:
@RobertCNelson Thanks for your guidance.
I got around this problem(with kernel 4.9.80-ti-r98) by replacing the newly installed /lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo with the old one. But [BB-BONE-eMMC1-01-00A0.dts](https://github.com/RobertCNelson/bb.org-overlays/blob/master/src/arm/BB-BONE-eMMC1-01-00A0.dts) hasn't been modified recently, I still don't understand how it fixed the problem.

And for the wrong model name, I think it should be caused by the lack of am335x-bonegreen-wireless-uboot-univ.dtb ?
```
Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
BeagleBone Black:
Model: SeeedStudio BeagleBone Green Wireless:
...
debug: [enable_uboot_overlays=1] ...
debug: [enable_uboot_cape_universal=1] ...
debug: [uboot_base_dtb_univ=am335x-bonegreen-wireless-uboot-univ.dtb] ...
uboot_overlays: [uboot_base_dtb=am335x-bonegreen-wireless-uboot-univ.dtb] ...
debug: unable to find [am335x-bonegreen-wireless-uboot-univ.dtb] using [am335x-boneblack-uboot.dtb] instead ...
debug: [uboot_base_dtb_univ=am335x-boneblack-uboot.dtb] ...
uboot_overlays: [uboot_base_dtb=am335x-boneblack-uboot.dtb] ...
uboot_overlays: Switching too: dtb=am335x-boneblack-uboot.dtb ...
loading /boot/dtbs/4.9.80-ti-r98/am335x-boneblack-uboot.dtb ...
61626 bytes read in 22 ms (2.7 MiB/s)
uboot_overlays: [fdt_buffer=0x60000] ...
uboot_overlays: loading /lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo ...
1606 bytes read in 42 ms (37.1 KiB/s)
uboot_overlays: uboot loading of [/lib/firmware/BB-BBGW-WL1835-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot_overlay_wireless=1]...
uboot_overlays: loading /lib/firmware/BB-ADC-00A0.dtbo ...
867 bytes read in 118 ms (6.8 KiB/s)
uboot_overlays: loading /lib/firmware/univ-bbgw-Ex-00A0.dtbo ...
112752 bytes read in 71 ms (1.5 MiB/s)
```

Thanks again,