beagleboard/bb.org-overlays #181: dpulls to manage your dependent PRs

dpulls-support, 2020-06-20:
Hi, Walid the founder of [dpulls](https://github.com/marketplace/dpulls) here.

We noticed that you are working with dependent PRs https://github.com/beagleboard/bb.org-overlays/pull/180 and we thought that maybe [dpulls](https://github.com/marketplace/dpulls) could be interesting for you.
[dpulls](https://github.com/marketplace/dpulls) will help working with dependent PRs by offering these features:

- A submission check to make sure your PR cannot be merged before the dependency PR gets merged first.
- It has support for cross-repository dependency, which means a PR can depend on a PR from another repository (as long as they belong to the same organisation/owner), like this one https://github.com/beagleboard/bb.org-overlays/pull/180
- Conditional merge: If your PR is ready to merge but waiting for a dependency PR to resolve, just put a comment and dpulls will merge it for you once the dependency gets resolved.
- And [more](https://www.dpulls.com/#features)

So if you are interested, give it a try and let me know what you think. The setup process doesn't take more than 1 min.
cc @RobertCNelson @lorforlinux 

If you have questions, feel free to send me an email:
- support email: support@dpulls.com
- personal email: wberrahal@dpulls.com

Thanks and have a good day :)

pdp7, 2020-06-20:
PR spam :disappointed:

closed 2020-06-20 by pdp7

RobertCNelson, 2020-06-20:
@dpulls-support spam is not appreciated.

dpulls-support, 2020-06-20:
Sorry for the inconvenience :(