beagleboard/bb.org-overlays #190: U-boot Overlay does not work (pin setup does not change)

mfariasp, 2020-07-20:
Hi, 

I have two BBB.
Both of them run over ubuntu 18.04.4 LTS

BBB X has kernel 4.14.108-ti-r104 arm71
BBB Y has kernel 5.4.24-arm7-x20

On the BBB X, on uEnv.txt I have registered on uboot_overlay_addr4 a custom dtbo that works.

Both BBB has the same uEnv.txt file and both boot from MicroSD card.

I have added to the BBB Y (uEnv.txt) the same dtbo as BBB X, but after reboot, I have the default pin configuration.

This is the /opt/scripts/tools/version.sh output for the BBB X (this one works), i see the pin config i need.

/opt/scripts/tools/version.sh
git:/opt/scripts/:[9df8deb6b2ae7570c9102ab58f795d3bed592194]
eeprom:[A335BNLT000C1837BBBG0932]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net console Ubuntu Image 2019-04-10]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.01-00002-g6de8c74429]:[location: dd MBR]
kernel:[4.14.108-ti-r104]
nodejs:[v8.10.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/MY-PINMUX-SETUP.dtbo]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190404.0-0rcnee0~bionic+20190404]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~bionic+20190227]
pkg:[kmod]:[24-1ubuntu3.2rcnee0~bionic+20190208]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet cape_universal=enable]
dmesg | grep remote
[    2.397252] remoteproc remoteproc0: wkup_m3 is available
[    2.522793] remoteproc remoteproc0: powering up wkup_m3
[    2.522911] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    2.527023] remoteproc remoteproc0: remote processor wkup_m3 is now up
.
.
.
.

This is the /opt/scripts/tools/version.sh output for the BBB Y (this one do not works) no ping config change
/opt/scripts/tools/version.sh
git:/opt/scripts/:[5fa30ffcf068996ec08e034d2af5c5a133f10709]
eeprom:[A335BNLT000C1908BBBG0612]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net Ubuntu Console Image 2020-03-12]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
kernel:[5.4.24-armv7-x20]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/MY-PINMUX-SETUP.dtbo]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200312.0-0rcnee0~bionic+20200312]
pkg:[bb-wl18xx-firmware]:[1.20200226.1-0rcnee0~bionic+20200226]
pkg:[kmod]:[24-1ubuntu3.2rcnee0~bionic+20190208]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p2 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet cape_universal=enable]
dmesg | grep remote
[    2.316134] remoteproc remoteproc0: wkup_m3 is available
[    2.594102] remoteproc remoteproc0: powering up wkup_m3
[    2.594218] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    2.597986] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.817126] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 002: ID 2357:011e
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

What I'm missing??.

Thx

RobertCNelson, 2020-07-20:
@mfariasp sorry, from log we can't really tell anything, can you please share the source of:

MY-PINMUX-SETUP.dts

Regards,

mfariasp, 2020-07-20:
@RobertCNelson thx for the answer,

This is the dts that works on the BBB X but not on the BBB Y (5.4.24-arm7-x20)

/*
*/
/dts-v1/;
/plugin/;

/{
    compatible = "ti,beaglebone", "ti,beaglebone-black";
    part-number = "BB-FESTO";
    version = "00A0";
    fragment@0 {
        target = <&am33xx_pinmux>;
        __overlay__ {
            bb_festo_pins: pinmux_bb_festo_pins {
                pinctrl-single,pins = <
                                0x000 0x30 /*U7  -- GPMC_AD0 P8_25 U7*/
                                0x004 0x30 /*V7  -- GPMC_AD1 P8_24 V7*/
                                0x008 0x30 /*R8  -- GPMC_AD2 P8_5 R8*/
                                0x00C 0x30 /*T8  -- GPMC_AD3 P8_6 T8*/
                                0x010 0x30 /*U8  -- GPMC_AD4 P8_23 U8*/
                                0x014 0x30 /*V8  -- GPMC_AD5 P8_22 V8*/
                                0x018 0x30 /*R9  -- GPMC_AD6 P8_3 R9*/
                                0x01C 0x30 /*T9  -- GPMC_AD7 P8_4 T9*/
                                0x020 0x30 /*U10  -- GPMC_ADH0 8 P8_19 U10*/
                                0x024 0x30 /*T10  -- GPMC_ADH1 9 P8_13 T10*/
                                0x028 0x30 /*T11  -- GPMC_ADH2 10 P8_14 T11*/
                                0x02C 0x30 /*U12  -- GPMC_ADH3 11 P8_17 U12*/
                                0x030 0x30 /*T12  -- GPMC_ADH4 12 P8_12 T12*/
                                0x034 0x30 /*R12  -- GPMC_ADH5 13 P8_11 R12*/
                                0x038 0x30 /*V13  -- GPMC_ADH6 14 P8_16 V13*/
                                0x03C 0x30 /*U13  -- GPMC_ADH7 15 P8_15 U13*/
                                0x0A0 0x31 /*R1  -- GPMCA0 P8_45 R1*/
                                0x0A4 0x31 /*R2  -- GPMCA1 P8_46 R2*/
                                0x0A8 0x31 /*R3  -- GPMCA2 P8_43 R3*/
                                0x0AC 0x31 /*R4  -- GPMCA3 P8_44 R4*/
                                0x0B0 0x31 /*T1  -- GPMCA4 P8_41 T1*/
                                0x0B4 0x31 /*T2  -- GPMCA5 P8_42 T2*/
                                0x0B8 0x31 /*T3  -- GPMCA6 P8_39 T3*/
                                0x0BC 0x31 /*T4  -- GPMCA7 P8_40 T4*/
                                0x00C 0x30 /*T8  -- GPMC_AD3 P8_6 T8*/
                                0x010 0x30 /*U8  -- GPMC_AD4 P8_23 U8*/
                                0x014 0x30 /*V8  -- GPMC_AD5 P8_22 V8*/
                                0x018 0x30 /*R9  -- GPMC_AD6 P8_3 R9*/
                                0x01C 0x30 /*T9  -- GPMC_AD7 P8_4 T9*/
                                0x020 0x30 /*U10  -- GPMC_ADH0 8 P8_19 U10*/
                                0x024 0x30 /*T10  -- GPMC_ADH1 9 P8_13 T10*/
                                0x028 0x30 /*T11  -- GPMC_ADH2 10 P8_14 T11*/
                                0x02C 0x30 /*U12  -- GPMC_ADH3 11 P8_17 U12*/
                                0x030 0x30 /*T12  -- GPMC_ADH4 12 P8_12 T12*/
                                0x034 0x30 /*R12  -- GPMC_ADH5 13 P8_11 R12*/
                                0x038 0x30 /*V13  -- GPMC_ADH6 14 P8_16 V13*/
                                0x03C 0x30 /*U13  -- GPMC_ADH7 15 P8_15 U13*/
                                0x0A0 0x31 /*R1  -- GPMCA0 P8_45 R1*/
                                0x0A4 0x31 /*R2  -- GPMCA1 P8_46 R2*/
                                0x0A8 0x31 /*R3  -- GPMCA2 P8_43 R3*/
                                0x0AC 0x31 /*R4  -- GPMCA3 P8_44 R4*/
                                0x0B0 0x31 /*T1  -- GPMCA4 P8_41 T1*/
                                0x0B4 0x31 /*T2  -- GPMCA5 P8_42 T2*/
                                0x0B8 0x31 /*T3  -- GPMCA6 P8_39 T3*/
                                0x0BC 0x31 /*T4  -- GPMCA7 P8_40 T4*/
                                0x0E0 0x9 /*U5  -- GPMC_A8 P8_27 U5*/
                                0x0E4 0x9 /*R5  -- GPMC_A9 P8_29 R5*/
                                0x0E8 0x9 /*V5  -- GPMC_A10 P8_28 V5*/
                                0x0EC 0x11 /*R6  -- GPMC_A11 P8_30 R6*/
                                0x070 0x30 /*T17  -- GPMC_WAIT0 P9_11 T17*/
                                0x080 0x30 /*U9  -- GPMC_CSN1 P8_21 U9*/
                                0x084 0x30 /*V9  -- GPMC_CSN2 P8_20 V9*/
                                0x08C 0x10 /*V12  -- GPMC_CLK P8_18 V12*/
                                0x090 0x30 /*R7  -- GPMC_ADVN_ALE P8_7 R7*/
                                0x094 0x30 /*T7  -- GPMC_OE_REN P8_8 T7*/
                                0x098 0x30 /*U6  -- GPMC_WEN P8_10 U6*/
                                0x044 0x37 /*V14 ---PIN_GPMC_A1 --- V14*/
                                0x048 0x37 /*U14 ---PIN_GPMC_A2 --- U14*/
                                0x04C 0x37 /*T14 ---PIN_GPMC_A3 --- T14*/
                               >;
                       };
                };
        };
        fragment@1 {
                target = <&ocp>; /* FESTO */
                __overlay__ {
                    test_helper: helper {
                        compatible = "bone-pinmux-helper";
                        status = "okay";
                        pinctrl-names = "default";
                        pinctrl-0 = <&bb_festo_pins>;
                };
        };
   };
};

RobertCNelson, 2020-07-20:
bone-pinmux-helper isn't build in the "5.4.x" branch aka 5.4.24-arm7-x20

https://github.com/RobertCNelson/armv7-multiplatform/blob/v5.4.x/patches/defconfig#L2177

You need to use "bone.."

```
sudo apt install linux-image-5.4.47-bone30
```

Regards,

mfariasp, 2020-07-20:
> bone-pinmux-helper isn't build in the "5.4.x" branch aka 5.4.24-arm7-x20
> 
> https://github.com/RobertCNelson/armv7-multiplatform/blob/v5.4.x/patches/defconfig#L2177
> 
> You need to use "bone.."
> 
> ```
> sudo apt install linux-image-5.4.47-bone30
> ```
> 
> Regards,

and just like that,  it works..... 

Thank you.

closed 2020-07-20 by mfariasp