beagleboard/bb.org-overlays #191: config-pin fails to set pin mux for p9.42

VedantParanjape, 2020-07-20:
**Describe the bug**
I am trying to use ecap module in PRU-ICSS to generate PWM. It is connected to P9.42, even after setting it's mode to pru_ecap. I am unable to get a PWM output on that pin. I am sure my code is correct.
**Describe which image you are running (example: bone-debian-10.2-iot-armhf-2020-02-03-4gb.img.xz) **
Image: bone-eMMC-flasher-debian-10.3-iot-armhf-2020-04-06-4gb.img.xz
**Describe your hardware**
BeagleBone Black Wireless
**Describe any Capes Installed**
None
**Please run this built-in script**
```sudo /opt/scripts/tools/version.sh```
git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLTBWA52010BBWG0621]
model:[TI_AM335x_BeagleBone_Black_Wireless]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BBBW-WL1835-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
kernel:[4.19.94-ti-r45]
nodejs:[v10.21.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200709.0-0~buster+20200709]
pkg:[bb-wl18xx-firmware]:[1.20200702.0-0~buster+20200702]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   11.045528] remoteproc remoteproc0: 4a334000.pru is available
[   11.061712] remoteproc remoteproc1: 4a338000.pru is available
[   59.735303] remoteproc remoteproc2: wkup_m3 is available
[   60.287649] remoteproc remoteproc2: powering up wkup_m3
[   60.287683] remoteproc remoteproc2: Booting fw image am335x-pm-firmware.elf, size 217168
[   60.287972] remoteproc remoteproc2: remote processor wkup_m3 is now up
dmesg | grep pru
[   11.045528] remoteproc remoteproc0: 4a334000.pru is available
[   11.045716] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   11.061712] remoteproc remoteproc1: 4a338000.pru is available
[   11.061899] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.957116] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.970499] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

RobertCNelson, 2020-07-20:
P9.42 is one of the special dual pins, can you share both config-pin commands you are using to enable pwm.

Regards,

VedantParanjape, 2020-07-21:
`config-pin p9.42 pru_ecap`

This is the command i am using. This is the output of show-pins:https://pastebin.com/r2128R9b

VedantParanjape, 2020-07-24:
Hello @RobertCNelson any updates ?

alefsouto, 2025-02-03:
@VedantParanjape have you solve this problem? I'm facing the same issue.

RobertCNelson, 2025-02-03:
P9_42 on 4.19.x:

```
	P9_42_PRU="39"
	P9_42_GPIO="7"
	P9_42_PIN="gpio"
	P9_42_PINMUX="default gpio gpio_pu gpio_pd gpio_input spi_cs spi_sclk uart pwm pru_ecap"
	P9_42_INFO="gpio0_7 default gpio0_7 gpio0_7 gpio0_7 gpio0_7 spi1_cs1 spi1_sclk uart3_txd ecap0_in_pwm0_out pru_ecap"
	P9_42_CAPE=""

	P9_92_PRU="146"
	P9_92_GPIO="114"
	P9_92_PIN="gpio"
	P9_92_PINMUX="default gpio gpio_pu gpio_pd gpio_input qep pruout pruin"
	P9_92_INFO="gpio3_18 default gpio3_18 gpio3_18 gpio3_18 gpio3_18 eqep0a_in pru0_out4 pru0_in4"
	P9_92_CAPE=""
```