beagleboard/bb.org-overlays #192: BBB does no load after update bootloader to solve overlay issue.

mfariasp, 2020-07-21:
I think have made a huge mistake, 

I'm working with multiple BBB at the same time and I had a problem loading the overlay on one of them.

I though the BBB bootloader was too old to support overlay and I found this 

https://github.com/beagleboard/bb.org-overlays/issues/87#issuecomment-473680205 

The problem was that I got confused and run it on the BBB that was booting from eMMC (no card MicroSD). after reboot the BBB does nothing.

any advice? (other than being more careful)...

thx.

jadonk, 2020-07-21:
Boot from microSD, then re-install the bootloader to the eMMC using the [update_bootloader.sh script](https://github.com/RobertCNelson/boot-scripts/blob/master/tools/developers/update_bootloader.sh). You can find it in /opt/scripts/tools/developer.

mfariasp, 2020-07-21:
@jadonk thank you for the answer, but unfortunately, it does not boot from anywhere, I've inserted a microSD card and it does not boot. As I've written... just the power led is on.

mfariasp, 2020-07-22:
@jadonk ... I found why I was not capable to boot from the microSD..... I think the image was corrupted ... now i'm capable to boot from microSD thx

closed 2020-07-22 by mfariasp