beagleboard/bb.org-overlays #198: flash eMMC with kernel 5.4.47-bone30

mfariasp, 2020-08-04:
Hi guys,

I've MicroSD with a version that I would like to flash into the eMMC.

Once I try this:
`cd /opt/scripts/tools/eMMC/`
`./init-eMMC-flasher-v3.sh`

I have this output

Checking for Valid bbb-eeprom header
Traceback (last called is first):
     do_we_have_eeprom() in ./functions.sh:483
     do_we_have_am335x_eeprom() in ./functions.sh:491
     check_am335x_eeprom() in ./functions.sh:499
     check_eeprom() in ./functions.sh:535
     main() in ./init-eMMC-flasher-v3.sh:46
The command ls /sys/devices/ocp*/44e0b000.i2c/i2c-0/0-0050/eeprom 2> /dev/null exited with exit code 2.
umount: /boot: not mounted.
100000+0 records in
100000+0 records out
51200000 bytes (51 MB, 49 MiB) copied, 1.32503 s, 38.6 MB/s
Traceback (last called is first):
     do_we_have_eeprom() in ./functions.sh:483
     do_we_have_am335x_eeprom() in ./functions.sh:491
     check_am335x_eeprom() in ./functions.sh:499
     check_eeprom() in ./functions.sh:535
     main() in ./init-eMMC-flasher-v3.sh:46
The command eeprom_location=$(ls /sys/devices/ocp*/44e0b000.i2c/i2c-0/0-0050/eeprom 2> /dev/null) exited with exit code 2.

Tearing Down script environment
==> Unmounting /tmp
umount: /tmp: not mounted.
==> Unmounting /boot
umount: /boot: not mounted.
==> Stopping Cylon LEDs ...
==> Stopping Cylon LEDs ...
==> Setting LEDs to

I was looking for answers and I found this:
[https://www.mail-archive.com/beagleboard@googlegroups.com/msg51333.html](url)

I think the problem is related to the kernel version I'm using ... 

What I'm missing?

this is /opt/scripts/tools/version.sh output:

git:/opt/scripts/:[5fa30ffcf068996ec08e034d2af5c5a133f10709]
eeprom:[A335BNLT000C1837BBBG0932]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net Ubuntu Console Image 2020-03-12]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00001-g21a26066ea]:[location: fatfs /boot/uboot/MLO]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack.dts]
kernel:[5.4.47-bone30]
/boot/uEnv.txt Settings:
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200722.0-0~bionic+20200722]
pkg:[bb-wl18xx-firmware]:[1.20200702.0-0~bionic+20200702]
pkg:[kmod]:[24-1ubuntu3.4rcnee0~bionic+20200525]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p2 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet cape_universal=enable]
dmesg | grep remote
[    1.582673] remoteproc remoteproc0: wkup_m3 is available
[    1.929052] remoteproc remoteproc0: powering up wkup_m3
[    1.929073] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.933073] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.391040] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

Thx.

RobertCNelson, 2020-08-04:
This was fixed with: https://github.com/RobertCNelson/boot-scripts/commit/9b2c9d6a090fe468944b97408b3dcda2d2ec9fce

```
cd /opt/scripts/
git pull
```

Regards,

mfariasp, 2020-08-04:
> This was fixed with: [RobertCNelson/boot-scripts@9b2c9d6](https://github.com/RobertCNelson/boot-scripts/commit/9b2c9d6a090fe468944b97408b3dcda2d2ec9fce)
> 
> ```
> cd /opt/scripts/
> git pull
> ```
> 
> Regards,
Hi Robert,

After this update this is the output i had:

Writing bootloader to [/dev/mmcblk1]

==> Figuring out options for SPL U-Boot copy ...
===> Will use : conv=notrunc
==> Figuring out options for U-Boot copy ...
===> Will use : conv=notrunc
==> Copying SPL U-Boot with dd if=/opt/backup/uboot/ of=/dev/mmcblk1 conv=notrunc

dd: failed to open '/opt/backup/uboot/': No such file or directory
Traceback (last called is first):
     _dd_bootloader() in ./functions.sh:809
     prepare_drive() in ./functions.sh:1402
     main() in ./init-eMMC-flasher-v3.sh:49
The command dd if=${dd_spl_uboot_backup} of=${destination} ${dd_spl_uboot} exited with exit code 1.

Tearing Down script environment
==> Unmounting /tmp
==> Unmounting /boot
umount: /boot: not mounted.
==> Stopping Cylon LEDs ...
==> Setting LEDs to
==> Force writeback of eMMC buffers by Syncing: /dev/mmcblk1
./functions.sh: line 256:  7534 Terminated              cylon_leds

100000+0 records in
100000+0 records out
51200000 bytes (51 MB, 49 MiB) copied, 1.23586 s, 41.4 MB/s

===> Syncing: /dev/mmcblk1 complete
==> This script took 35 seconds to run

==> Setting LEDs to

RobertCNelson, 2020-08-04:
@mfariasp 

```
Writing bootloader to [/dev/mmcblk1]

==> Figuring out options for SPL U-Boot copy ...
===> Will use : conv=notrunc
==> Figuring out options for U-Boot copy ...
===> Will use : conv=notrunc
==> Copying SPL U-Boot with dd if=/opt/backup/uboot/ of=/dev/mmcblk1 conv=notrunc

dd: failed to open '/opt/backup/uboot/': No such file or directory
Traceback (last called is first):
```

The script was unable to find a backup copy of U-Boot to install into the eMMC.

All factory images, have this backup stored under /opt/backup/uboot/

```
debian@bbb-pwr02-ser10:~$ ls -lha /opt/backup/uboot/*
-rw-r--r-- 1 root root  91K Aug  3 16:48 /opt/backup/uboot/MLO
-rw-r--r-- 1 root root 464K Aug  3 16:48 /opt/backup/uboot/u-boot.img
```

Your base image is just really old, so let's create those backup's now.. 

```
wget https://rcn-ee.com/repos/bootloader/am335x_evm/MLO-am335x_evm-v2019.04-r15
wget https://rcn-ee.com/repos/bootloader/am335x_evm/u-boot-am335x_evm-v2019.04-r15.img

sudo mkdir -p /opt/backup/uboot/
sudo cp -v ./MLO-am335x_evm-v2019.04-r15 /opt/backup/uboot/MLO
sudo cp -v ./u-boot-am335x_evm-v2019.04-r15.img /opt/backup/uboot/u-boot.img
```
Regards,

mfariasp, 2020-08-04:
@RobertCNelson thanks, but this is the new output. 

Writing bootloader to [/dev/mmcblk1]

==> Figuring out options for SPL U-Boot copy ...
===> Will use : conv=notrunc
==> Figuring out options for U-Boot copy ...
===> Will use : conv=notrunc
==> Copying SPL U-Boot with dd if=/opt/backup/uboot/ of=/dev/mmcblk1 conv=notrunc

**dd: error reading '/opt/backup/uboot/': Is a directory**
0+0 records in
0+0 records out
0 bytes copied, 0.011922 s, 0.0 kB/s
Traceback (last called is first):
     _dd_bootloader() in ./functions.sh:809
     prepare_drive() in ./functions.sh:1402
     main() in ./init-eMMC-flasher-v3.sh:49
The command dd if=${dd_spl_uboot_backup} of=${destination} ${dd_spl_uboot} exited with exit code 1.

Tearing Down script environment
==> Unmounting /tmp
==> Unmounting /boot
umount: /boot: not mounted.
==> Stopping Cylon LEDs ...
==> Setting LEDs to
==> Force writeback of eMMC buffers by Syncing: /dev/mmcblk1
./functions.sh: line 256:  7850 Terminated              cylon_leds

100000+0 records in
100000+0 records out
51200000 bytes (51 MB, 49 MiB) copied, 1.23597 s, 41.4 MB/s

===> Syncing: /dev/mmcblk1 complete
==> This script took 36 seconds to run

==> Setting LEDs to

Regards,

RobertCNelson, 2020-08-04:
that's exactly the same.. are you sure?

side note do you have a /boot/SOC.sh file?

```
#!/bin/sh
format=1.0

board=am335x_evm

bootloader_location=dd_spl_uboot_boot
bootrom_gpt=

dd_spl_uboot_count=2
dd_spl_uboot_seek=1
dd_spl_uboot_conf=notrunc
dd_spl_uboot_bs=128k
dd_spl_uboot_backup=/opt/backup/uboot/MLO

dd_uboot_count=4
dd_uboot_seek=1
dd_uboot_conf=notrunc
dd_uboot_bs=384k
dd_uboot_backup=/opt/backup/uboot/u-boot.img

boot_fstype=ext4
conf_boot_startmb=4
conf_boot_endmb=
sfdisk_fstype=L

boot_label=BOOT
rootfs_label=rootfs

#Kernel
dtb=
serial_tty=ttyO0
usbnet_mem=

#Advanced options
#disable_ssh_regeneration=true
```

mfariasp, 2020-08-04:
@RobertCNelson thanks for the help, but it's not the same 

first one:

dd: failed to open '/opt/backup/uboot/': No such file or directory
Traceback (last called is first):

last one:
dd: error reading '/opt/backup/uboot/': Is a directory

And yes I have the SOC.sh:

#!/bin/sh
format=1.0

board=omap3_beagle

bootloader_location=fatfs_boot
bootrom_gpt=

dd_spl_uboot_count=
dd_spl_uboot_seek=
dd_spl_uboot_conf=notrunc
dd_spl_uboot_bs=
dd_spl_uboot_backup=/opt/backup/uboot/

dd_uboot_count=
dd_uboot_seek=
dd_uboot_conf=notrunc
dd_uboot_bs=
dd_uboot_backup=/opt/backup/uboot/

boot_fstype=fat
conf_boot_startmb=1
conf_boot_endmb=12
sfdisk_fstype=0xE

boot_label=BOOT
rootfs_label=rootfs

#Kernel
dtb=
serial_tty=ttyO2
usbnet_mem=16384

#Advanced options
#disable_ssh_regeneration=true

RobertCNelson, 2020-08-04:
Oh! 

```
board=omap3_beagle
```
you've installed the bbxm image.. yeah...

Okay, change, well everything to the version i posted.. https://github.com/beagleboard/bb.org-overlays/issues/198#issuecomment-668778557

Regards,

mfariasp, 2020-08-04:
Thanks @RobertCNelson, it seems to work ..... but ..

Copying: Current rootfs to /dev/mmcblk1p1

==> rsync: / -> /tmp/rootfs
          3.48G  99%    4.48MB/s    0:12:22 (xfr#121637, to-chk=0/154005)
rsync: write failed on "/tmp/rootfs/var/lib/snapd/cache/fccfbc34b117e37e8d9b2956d5c5f1a5cde6cb6fc782bae5d2965130afdcf3763e863eb7de6b7b86cdb7d89799e00da3": No space left on device (28)
rsync error: error in file IO (code 11) at receiver.c(393) [receiver=3.1.2]
writing to [/dev/mmcblk1] failed...
==> Stopping Cylon LEDs ...
==> Setting LEDs to
./functions.sh: line 391:  8184 Terminated              cylon_leds

I imaging that it's because the MicroSD I'm using is an 8GB card.

RobertCNelson, 2020-08-04:
Okay, well there is the main problem.

```
No space left on device
```

So your FileSystem on the microSD is bigger then the eMMC..

Regards,

mfariasp, 2020-08-05:
@RobertCNelson thanks for your help, basically I've deleted some files I don't need anymore and it works.

closed 2020-08-05 by mfariasp