beagleboard/bb.org-overlays #207: RS485 DE/RE pin does not work

ffeldbauer, 2020-11-02:
Hey,

I'm using a beaglebone black with the [Debian 10.3 image](https://rcn-ee.com/rootfs/2020-04-09/flasher/) with Kernel version 5.4.66-ti-r18.
Based on the BB-UART4-RS485 overlay, I wrote an overlay for UART5 using P8.37 (Tx), P8.38 (Rx), P8.32 (de/re).
I have the problem that gpio0_11 doesn't go high when transmitting.

In an older issue with a similar problem, it was mentioned, one should rebuild the kernel with
```
CONFIG_SERIAL_8250_OMAP=n
CONFIG_SERIAL_8250_OMAP_TTYO_FIXUP=n
CONFIG_SERIAL_OMAP=y
```

Is this still true for the 5.4.x Kernel releases?
If so, is there any guide how to rebuild the kernel under Debian 10 (amd64)?

Regards,
Florian

hanusek, 2021-02-09:
I have a similar problem. Also trying to turn on RS485. Config-pin doesn't work. My overlay doesn't work either.

My dts file:
```
/dts-v1/;
/plugin/;

#include <dt-bindings/board/am335x-bbw-bbb-base.h>
#include <dt-bindings/pinctrl/am33xx.h>

/ {
	compatible = "ti,beaglebone", "ti,beaglebone-black", "ti,beaglebone-green";

	// identification
	part-number = "BB-UART4-RS485";
	version = "00A0";

	// resources this cape uses
	exclusive-use =
		"P9.13",		// uart4_txd
		"P9.11",		// uart4_rxd
		"P8.33",		// uart4_de/re
                "gpio0_9",

		"uart4";		// hardware ip used

	/*
	 * Helper to show loaded overlays under: /proc/device-tree/chosen/overlays/
	 */
	fragment@0 {
		target-path="/";
		__overlay__ {

			chosen {
				overlays {
					BB-UART4-RS485-00A0 = __TIMESTAMP__;
				};
			};
		};
	};

	/*
	 * Free up the pins used by the cape from the pinmux helpers.
	 */
	fragment@1 {
		target = <&ocp>;
		__overlay__ {
			P9_13_pinmux { status = "disabled"; };	/* P9_13: gpmc_wpn.uart4_txd_mux2 */
			P9_11_pinmux { status = "disabled"; };	/* P9_11: gpmc_wait0.uart4_rxd_mux2 */
			P8_33_pinmux { status = "disabled"; };	/* P8_33: PIN_OUTPUT_PULLDOWN | MUX_MODE7 (GPIO3_19) */
		};
	};

	fragment@2 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			bb_uart4_rs485_pins: pinmux_bb_uart4_rs485_pins {
				pinctrl-single,pins = <
					BONE_P9_13 (PIN_OUTPUT | MUX_MODE6)	/* P9_13: gpmc_wpn.uart4_txd_mux2 */
					BONE_P9_11 (PIN_INPUT  | MUX_MODE6)	/* P9_11: gpmc_wait0.uart4_rxd_mux2 */
					BONE_P8_33 0x07				/* P9_27: PIN_OUTPUT_PULLDOWN | MUX_MODE7 (GPIO3_19) */
				>;
			};
		};
	};

	fragment@3 {
		target = <&uart4>;
		__overlay__ {
			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <&bb_uart4_rs485_pins>;
			rs485-rts-delay = <0 0>;

			rts-gpio = <&gpio0 9 0>; 
			rs485-rts-active-low;
			linux,rs485-enabled-at-boot-time;
		};
	};
};
```

ffeldbauer, 2021-02-09:
I solved this issue by compiling my own kernel with the modifications metioned in my original post.
You can find the kernel sources with build scripts [here](https://github.com/RobertCNelson/bb-kernel.git)

You just need to do this modification
```diff
diff --git a/patches/defconfig b/patches/defconfig
index 79f6691c..6d10ebe4 100644
--- a/patches/defconfig
+++ b/patches/defconfig
@@ -2860,8 +2860,7 @@ CONFIG_SERIAL_8250_FSL=y
 # CONFIG_SERIAL_8250_DW is not set
 # CONFIG_SERIAL_8250_EM is not set
 # CONFIG_SERIAL_8250_RT288X is not set
-CONFIG_SERIAL_8250_OMAP=y
-CONFIG_SERIAL_8250_OMAP_TTYO_FIXUP=y
+# CONFIG_SERIAL_8250_OMAP is not set
 CONFIG_SERIAL_OF_PLATFORM=y

 #
@@ -2873,7 +2872,8 @@ CONFIG_SERIAL_MAX310X=m
 # CONFIG_SERIAL_UARTLITE is not set
 CONFIG_SERIAL_CORE=y
 CONFIG_SERIAL_CORE_CONSOLE=y
-# CONFIG_SERIAL_OMAP is not set
+CONFIG_SERIAL_OMAP=y
+CONFIG_SERIAL_OMAP_CONSOLE=y
 # CONFIG_SERIAL_SIFIVE is not set
 # CONFIG_SERIAL_SCCNXP is not set
 # CONFIG_SERIAL_SC16IS7XX is not set
 ```

closed 2022-03-29 by ffeldbauer