beagleboard/bb.org-overlays #212: Problem of migrating from kernel overlays to uboot overlay

BrentCC, 2021-02-12:
Hi,
I have packet sniffing device that consists of BBG and a cape(sniffer)
The device was running with an old image from 2017, and I updated it to the latest image.
However after the update, I couldn't initialize BBG any more. After some debugging, the issue is related to the overlays.
The device was configured with kernel overlays with bone cape manager and slots, but these are removed from the latest kernel.

I tried several methods to re-configure it.
_**i) Disable the uboot overlay and stick to the old solution.**_
As instructed in [here](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Where_did_the_slots_file_go.3F), I put #enable_uboot_overlays=1 in the boot up file. However, it is automatically written into the file when BBG boots up
```
debug: [enable_uboot_overlays=] ...
loading /boot/dtbs/4.19.94-ti-r42/am335x-bonegreen-overlay-ni2c2.dtb ...
54543 bytes read in 72 ms (739.3 KiB/s)
uboot_overlays: add [enable_uboot_overlays=1] to /boot/uEnv.txt to enable...
```
_**ii) Override U-Boot external capes**_ 
As instructed [here](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Override_external_capes), I put overrode the uboot overlay with my customized ones.(using dtc to compile dts into dtbo)
```
uboot_overlay_addr0=/lib/firmware/am335x-bonegreen-overlay-ni2c2.dtbo
uboot_overlay_addr1=/lib/firmware/BB-GIRTC-00A0.dtbo
uboot_overlay_addr2=/lib/firmware/BBHDME8DMAPALL-00A0.dtbo
uboot_overlay_addr3=/lib/firmware/BBHDME8GPIOPALL-00A0.dtbo
uboot_overlay_addr4=/lib/firmware/lastditch-00A0.dtbo
uboot_overlay_addr5=/lib/firmware/RECLAIMI2C2-00A0.dtbo
```
Afterwards, I couldn't boot up BBG because, I have this error:
```
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
failed on fdt_overlay_apply(): FDT_ERR_BADMAGIC
base fdt does did not have a /__symbols__ node
make sure you've compiled with -@
```
I googled this error, and I found someone had encountered similar [issue](https://github.com/beagleboard/bb.org-overlays/issues/87) as well.
I tried to follow the [suggestion](https://github.com/beagleboard/bb.org-overlays/issues/87#issuecomment-397331309) to make my dtbo file, but nothing need to be done. **Do you know why it's not working in my case?**
```
debian@beaglebone:~/bb.org-overlays$ make /home/debian/dtree/src/am335x-bonegreen-overlay-ni2c2.dtbo
make: Nothing to be done for '/home/debian/dtree/src/am335x-bonegreen-overlay-ni2c2.dtbo'.
```
Therefore, the issue still remains. **Could you give me some hints about migrating from kernel overlays to uboot overlays?**
(By the way, the old DT files also contain slots)

**Image**: bone-eMMC-flasher-debian-10.3-iot-armhf-2020-04-06-4gb.img
**Hardware**: SeeedStudio BeagleBone Green
**Capes**: Customized cape(packet sniffer)

```
sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[e4e4854ef8ff9ada5c85553376043ee7679167ca]
eeprom:[A335BNLTBBG1BBG217083985]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot SPL 2019.04-00002-gc9b3922522                                                                                                              (Aug 24 2020 - 16:42:18 -0500)]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-gc9b3922522]:[l                                                                                                             ocation: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[M-BB-BBG-00A0]
kernel:[4.19.94-ti-r59]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-                                                                                                             00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20201221.0-0~buster+20201221]
pkg:[bb-customizations]:[1.20200306.0-0rcnee0~buster+20200306]
pkg:[bb-usb-gadgets]:[1.20200322.0-0rcnee0~buster+20200322]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~buster+20200322]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev us                                                                                                             ers systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio                                                                                                              docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/m                                                                                                             mcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656                                                                                                              rng_core.default_quality=100 quiet]
dmesg | grep remote
[   44.087108] remoteproc remoteproc0: wkup_m3 is available
[   44.186052] remoteproc remoteproc0: powering up wkup_m3
[   44.186080] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf,                                                                                                              size 217168
[   44.186332] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   45.821693] remoteproc remoteproc1: 4a334000.pru is available
[   45.829753] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   45.821693] remoteproc remoteproc1: 4a334000.pru is available
[   45.821861] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed succes                                                                                                             sfully
[   45.829753] remoteproc remoteproc2: 4a338000.pru is available
[   45.829932] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed succes                                                                                                             sfully
dmesg | grep pinctrl-single
[    0.987636] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.001404] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

Thank you so much in advance!

RobertCNelson, 2021-02-12:
@BrentCC 

> However, it is automatically written into the file when BBG boots up

No, that statement is incorrect.. U-Boot just "reports" it is disabled and recommends for you to enable it..

In your /boot/uEnv.txt just make sure you have these set..

```
dtb=<your custom kernel>.dtb
#enable_uboot_overlays=1
```

and it'll boot the same way as before u-boot overlays...

> ```
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
failed on fdt_overlay_apply(): FDT_ERR_BADMAGIC
base fdt does did not have a /__symbols__ node
make sure you've compiled with -@
```
Yeah, your going to have to post the problem overlay, i can't help you with just that information..

Regards,

BrentCC, 2021-02-12:
@RobertCNelson 
> No, that statement is incorrect.. U-Boot just "reports" it is disabled and recommends for you to enable it..
>
> In your /boot/uEnv.txt just make sure you have these set..
>
> ```
> dtb=<your custom [...]
Is it possible to put multiple customized customize there? I have six DT.
```
/lib/firmware/am335x-bonegreen-overlay-ni2c2.dtbo
/lib/firmware/BB-GIRTC-00A0.dtbo
/lib/firmware/BBHDME8DMAPALL-00A0.dtbo
/lib/firmware/BBHDME8GPIOPALL-00A0.dtbo
/lib/firmware/lastditch-00A0.dtbo
/lib/firmware/RECLAIMI2C2-00A0.dtbo
```

Thanks!

RobertCNelson, 2021-02-13:
@BrentCC no... that's not how it works.

For u-boot overlays we start with a base ".dtb"  this usually comes from the kernel tree.

For the beaglebone black this would be:

```
am335x-boneblack.dtb
```

Then for u-boot overlays you list the "overlays" you'd like to apply on top of those..

Which would be found under /lib/firmware/

and set via:

```
uboot_overlay_addrX=/lib/firmware/my-overlay.dtbo
```

Pre u-boot overlays, you'd just boot:

```
am335x-boneblack.dtb
```

Then in your case, you'd load these via 'slots'..

/lib/firmware/am335x-bonegreen-overlay-ni2c2.dtbo
/lib/firmware/BB-GIRTC-00A0.dtbo
/lib/firmware/BBHDME8DMAPALL-00A0.dtbo
/lib/firmware/BBHDME8GPIOPALL-00A0.dtbo
/lib/firmware/lastditch-00A0.dtbo
/lib/firmware/RECLAIMI2C2-00A0.dtbo

With u-boot overlays, the "load by slot" is done by u-boot instead of by the kernel..

Regards,

RobertCNelson, 2021-02-13:
Here's another example to explain it:

U-Boot overlay's:

U-Boot Load: am335x-boneblack.dtb
U-Boot apply: BB-GIRTC-00A0.dtbo
U-Boot apply: RECLAIMI2C2-00A0.dtbo
Boot into Linux..

Linux Overlay:

U-Boot Load: am335x-boneblack.dtb
Boot into Linux..
Linux load: BB-GIRTC-00A0.dtbo
Linux load: RECLAIMI2C2-00A0.dtbo

See, we just "moved" where the device overlay load happens.. and best of all, since we do it u-boot, you don't have a random kernel "runtime" issues..

Regards,

BrentCC, 2021-02-14:
Hi Robert

Thanks for the reply! It looks clearer to me.
> am335x-boneblack.dtb

Where can I find the path of this file? so I can modify it.
In my case it is _am335x-boneblack-uboot-univ.dts_

```
debian@beaglebone:/opt/scripts/tools$ sudo ./version.sh | grep UBOOT
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[M-BB-BBG-00A0]
```
I have a another question about correctly using [bb.org-overlays](https://github.com/beagleboard/bb.org-overlays/) to make the dtbo file.
Should I just download the repo and _make_ with the exisiting dtbo file?

Thanks again!

RobertCNelson, 2021-02-15:
@BrentCC the repo with the 4.19.94-ti device tree's can be found here:

https://github.com/beagleboard/BeagleBoard-DeviceTrees

just use the Makefile to build and install your *.dtb

Regards,

BrentCC, 2021-02-15:
Hi @RobertCNelson 

I

BrentCC, 2021-02-15:
@RobertCNelson 
I modified my /boot/uEnv.txt as below by disabling uboot
```
uname_r=4.19.94-ti-r59
#uuid=
dtb=/boot/dtbs/4.19.94-ti-r59/am335x-bonegreen-overlay-ni2c2.dtb
#enable_uboot_overlays=1
uboot_overlay_addr0=/lib/firmware/BB-GIRTC-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BBHDME8DMAPALL-00A0.dtbo
uboot_overlay_addr2=/lib/firmware/BBHDME8GPIOPALL-00A0.dtbo
uboot_overlay_addr3=/lib/firmware/lastditch-00A0.dtbo
uboot_overlay_addr4=/lib/firmware/RECLAIMI2C2-00A0.dtbo
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
#enable_uboot_cape_universal=1
cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet
```
However, I couldn't boot up afterwards. The boot up log is attached below.
[boot up without uboot.txt](https://github.com/beagleboard/bb.org-overlays/files/5981674/boot.up.without.uboot.txt)

RobertCNelson, 2021-02-15:
@BrentCC 
```
dtb=/boot/dtbs/4.19.94-ti-r59/am335x-bonegreen-overlay-ni2c2.dtb
```
I have no idea what that would do.. you already have "uname_r=4.19.94-ti-r59" defined, so it'll auto pre-pend the path: "/boot/dtbs/4.19.94-ti-r59/"...

So just:

dtb=am335x-bonegreen-overlay-ni2c2.dtb

Second:
```
#enable_uboot_overlays=1
```
Disables all uboot_overlays_addr*=

So if you want to use: uboot_overlays_addr*=

```
enable_uboot_overlays=1
```

Regards,

BrentCC, 2021-02-15:
Hi @RobertCNelson 
Maybe I misunderstood what you said before. 
This time I tried below settings, without overriding uboot overlays.
```
uname_r=4.19.94-ti-r42
#uuid=
dtb=am335x-bonegreen-overlay-ni2c2.dtb
#enable_uboot_overlays=1
```
But I run into another error that stops me from booting.
```
Gave up waiting for root file system device.  Common problems:
 - Boot args (cat /proc/cmdline)
   - Check rootdelay= (did the system wait long enough?)
 - Missing modules (cat /proc/modules; ls /dev)
ALERT!  /dev/mmcblk1p1 does not exist.  Dropping to a shell!
(initramfs)
```
[disable_uboot.txt](https://github.com/beagleboard/bb.org-overlays/files/5984330/disable_uboot.txt)

Next step, I will try 
```
enable_uboot_overlays=1
```
And add _uboot_overlay_addrX_

RobertCNelson, 2021-02-16:
@BrentCC do you have both microSD/eMMC defined in your custom am335x-bonegreen-overlay-ni2c2.dtb ?

Waiting for /dev/mmcblk1p1 usually means it's waiting for eMMC.

Since i don't have access to your your am335x-bonegreen-overlay-ni2c2.dtb, you'll need to look at it to see if you have the eMMC enabled...

Regards,

BrentCC, 2021-02-18:
Hi, @RobertCNelson 
> @BrentCC do you have both microSD/eMMC defined in your custom am335x-bonegreen-overlay-ni2c2.dtb ?
>
> Waiting for /dev/mmcblk1p1 usually means it's waiting for eMMC.
>
> Since i don't have access to your [...]
No, I only have eMMC to boot up. I flashed the image into eMMC through SDcard. I followed the [guide here](https://linuxpropaganda.wordpress.com/2018/06/26/backup-and-restore-your-beaglebone-black/). BBG works perfectly before I used the custom kernel. I upload [am335x-bonegreen-overlay-ni2c2.dts](https://github.com/BrentCC/vigilant-bassoon/blob/main/am335x-bonegreen-overlay-ni2c2.dts) here. 

In addition, I also tried to enable uboot overlays as below.
```
uname_r=4.19.94-ti-r42
dtb=am335x-bonegreen-overlay-ni2c2.dtb
enable_uboot_overlays=1
uboot_overlay_addr0=/lib/firmware/BB-GIRTC-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BBHDME8DMAPALL-00A0.dtbo
uboot_overlay_addr2=/lib/firmware/BBHDME8GPIOPALL-00A0.dtbo
uboot_overlay_addr3=/lib/firmware/lastditch-00A0.dtbo
uboot_overlay_addr4=/lib/firmware/RECLAIMI2C2-00A0.dtbo
```
However, maybe there is a conflict in i/o definition....
```
U-Boot SPL 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600)
Trying to boot from MMC2
Loading Environment from EXT4... Card did not respond to voltage select!

U-Boot 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600), Build: jenkins-github_Bootloader-Builder-137

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
No match for driver 'omap_hsmmc'
No match for driver 'omap_hsmmc'
Some drivers were not found
Reset Source: Global external warm reset has occurred.
Reset Source: Global warm SW reset has occurred.
Reset Source: Power-on reset has occurred.
RTC 32KCLK Source: External.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Loading Environment from EXT4... Card did not respond to voltage select!
Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
BeagleBone Black:
Model: SeeedStudio BeagleBone Green:
BeagleBone: cape eeprom: i2c_probe: 0x54:
BeagleBone: cape eeprom: i2c_probe: 0x55:
BeagleBone: cape eeprom: i2c_probe: 0x56:
BeagleBone: cape eeprom: i2c_probe: 0x57:
Net:   eth0: MII MODE
cpsw, usb_ether
Press SPACE to abort autoboot in 0 seconds
board_name=[BBG1] ...
Card did not respond to voltage select!
Card did not respond to voltage select!
Card did not respond to voltage select!
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
Card did not respond to voltage select!
Card did not respond to voltage select!
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc1(part 0) is current device
gpio: pin 54 (gpio 54) value is 1
[... 56 lines cut ...]
[    0.922647] pinctrl-single 44e10800.pinmux: could not request pin 95 (PIN95) from group pinmux_P9_19_default_pin  on device pinctrl-single
[    0.935224] bone-pinmux-helper ocp:P9_19_pinmux: Error applying setting, reverse things back
[    0.944316] pinctrl-single 44e10800.pinmux: pin PIN94 already requested by ocp:valversled_helper; cannot claim for ocp:P9_20_pinmux
[    0.956329] pinctrl-single 44e10800.pinmux: pin-94 (ocp:P9_20_pinmux) status -22
[    0.963860] pinctrl-single 44e10800.pinmux: could not request pin 94 (PIN94) from group pinmux_P9_20_default_pin  on device pinctrl-single
[    0.976432] bone-pinmux-helper ocp:P9_20_pinmux: Error applying setting, reverse things back
[    1.188501] omap_voltage_late_init: Voltage driver support not added
[    4.941876] cc2520 spi2.7: fifo gpio is not valid
[    4.948071] cc2520 spi2.6: fifo gpio is not valid
[    4.953169] pinctrl-single 44e10800.pinmux: pin PIN51 already requested by i2c@0; cannot claim for spi2.5
[    4.962826] pinctrl-single 44e10800.pinmux: pin-51 (spi2.5) status -22
[    4.969414] pinctrl-single 44e10800.pinmux: could not request pin 51 (PIN51) from group pinmux_bb_cc_6_pins  on device pinctrl-single
[    4.981491] cc2520 spi2.5: Error applying setting, reverse things back
[    4.988254] cc2520 spi2.4: fifo gpio is not valid
[    4.993163] pinctrl-single 44e10800.pinmux: pin PIN5 already requested by 481d8000.mmc; cannot claim for spi2.3
[    5.003331] pinctrl-single 44e10800.pinmux: pin-5 (spi2.3) status -22
[    5.009830] pinctrl-single 44e10800.pinmux: could not request pin 5 (PIN5) from group pinmux_bb_cc_4_pins  on device pinctrl-single
[    5.021750] cc2520 spi2.3: Error applying setting, reverse things back
[    5.028541] cc2520 spi2.2: fifo gpio is not valid
[    5.033493] cc2520 spi2.1: fifo gpio is not valid
[    5.038361] pinctrl-single 44e10800.pinmux: pin PIN7 already requested by 481d8000.mmc; cannot claim for spi2.0
[    5.048534] pinctrl-single 44e10800.pinmux: pin-7 (spi2.0) status -22
[    5.055036] pinctrl-single 44e10800.pinmux: could not request pin 7 (PIN7) from group pinmux_bb_cc_1_pins  on device pinctrl-single
[    5.067026] cc2520 spi2.0: Error applying setting, reverse things back
```

RobertCNelson, 2021-02-18:
> No, I only have eMMC to boot up. I flashed the image into eMMC through SDcard. I followed the guide here. BBG works perfectly before I used the custom kernel. I upload am335x-bonegreen-overlay-ni2c2.dts here.

If we look at that file:
```
        mmc@48060000 {
            status = "okay";
        };
        mmc@481d8000 {
            status = "disabled";
        };
        mmc@47810000 {
            status = "disabled";
        };
```
Only microSD is enabled, thus your eMMC is disabled..  you need to pass ```root=/dev/mmcblk0p1```

What was your old kernel?

> [    0.903073] pinctrl-single 44e10800.pinmux: pin PIN95 already requested by ocp:valversled_helper; cannot claim for ocp:P9_19_pinmux
[    0.915123] pinctrl-single 44e10800.pinmux: pin-95 (ocp:P9_19_pinmux) status -22
[    0.922647] pinctrl-single 44e10800.pinmux: could not request pin 95 (PIN95) from group pinmux_P9_19_default_pin  on device pinctrl-single
[    0.935224] bone-pinmux-helper ocp:P9_19_pinmux: Error applying setting, reverse things back
[    0.944316] pinctrl-single 44e10800.pinmux: pin PIN94 already requested by ocp:valversled_helper; cannot claim for ocp:P9_20_pinmux
[    0.956329] pinctrl-single 44e10800.pinmux: pin-94 (ocp:P9_20_pinmux) status -22
[    0.963860] pinctrl-single 44e10800.pinmux: could not request pin 94 (PIN94) from group pinmux_P9_20_default_pin  on device pinctrl-single
[    0.976432] bone-pinmux-helper ocp:P9_20_pinmux: Error applying setting, reverse things back

You need to disable pin's your overlay uses..

That's this section:

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-SPIDEV0-00A0.dts#L49-L60

Regards,

RobertCNelson, 2021-02-18:
PS, this is a bug:

```
debug: [dtb=am335x-bonegreen-overlay-ni2c2.dtb] ...
Using: dtb=am335x-bonegreen-overlay-ni2c2.dtb ...
Checking if uname_r is set in /boot/uEnv.txt...
gpio: pin 56 (gpio 56) value is 1
Running uname_boot ...
loading /boot/vmlinuz-4.19.94-ti-r42 ...
10095592 bytes read in 652 ms (14.8 MiB/s)
debug: [enable_uboot_overlays=1] ...
debug: [enable_uboot_cape_universal=] ...
debug: [uboot_base_dtb_univ=am335x-boneblack-uboot.dtb] ...
uboot_overlays: [uboot_base_dtb=am335x-boneblack-uboot.dtb] ...
uboot_overlays: Switching too: dtb=am335x-boneblack-uboot.dtb ...
loading /boot/dtbs/4.19.94-ti-r42/am335x-boneblack-uboot.dtb ...
```
I'll work on that it, it should not have swapped over...

BrentCC, 2021-02-19:
Hi @RobertCNelson 
Thanks for the reply! 
> If we look at that file:
>
> ```
>         mmc@48060000 {
>             status = "okay";
>         };
>         mmc@481d8000 {
>             status = "disabled";
>         };
>         mmc@47810000 { [...]
Yes, you are right!  Originally, the system was running on SD card. After too many read/write, I found out that some I/O on SD card were dead. I wasn't sure if those dead I/O caused the problem I had, so I decided to move the system to eMMC. 

> What was your old kernel?
I didn't record the exact version, but I know it's an old one from 2017.

> You need to disable pin's your overlay uses..
>
> That's this section:
>
> https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-SPIDEV0-00A0.dts#L49-L60
Yesterday, I contacted the original developer. He said the sniffer cape needs nearly every possible pin on the beaglebone to be configured as a GPIO to function properly, and more pins are available for this when booting off of the SD card. (eMMC mode reserves some pins for itself). 
Is there a way to use those pins while the system running on eMMC?

Thanks!

BrentCC, 2021-02-22:
Hi @RobertCNelson 

This time I tried to boot up the system from SD card and it seemed working fine. 
I tried to disable uboot overlay and stick to the old kernel overlay solution.
```
uname_r=4.19.94-ti-r42
dtb=am335x-bonegreen-overlay-ni2c2.dtb
#enable_uboot_overlays=1
```
But I found out that service cannot be initialized since `/sys/devices/platform/bone_capemgr/slots` is missing. Do I have to switch to uboot overlay now?
```
● bb_sniff.service - sniff 802.15.4, get 802.11 SSIDs, get GPS traces and timestamps
   Loaded: loaded (/lib/systemd/system/bb_sniff.service; enabled; vendor preset: enabled)
   Active: activating (start) since Mon 2021-02-22 13:51:36 UTC; 27s ago
Cntrl PID: 2478 (bb_sniff_marco.)
    Tasks: 2 (limit: 1029)
   Memory: 424.0K
   CGroup: /system.slice/bb_sniff.service
           ├─2478 /bin/bash - /usr/bin/bb_sniff_marco.sh
           └─2496 sleep 10

Feb 22 13:51:56 beaglebone bb_sniff_marco.sh[2478]: /usr/bin/bb_sniff_marco.sh: line 24: /sys/devices/platform/bone_capemgr/slots: Permission denied
Feb 22 13:51:58 beaglebone bb_sniff_marco.sh[2478]: cat: /sys/devices/platform/bone_capemgr/slots: No such file or directory
Feb 22 13:51:58 beaglebone bb_sniff_marco.sh[2478]: attempting to remove:
Feb 22 13:51:58 beaglebone bb_sniff_marco.sh[2478]: /usr/bin/bb_sniff_marco.sh: line 30: /sys/devices/platform/bone_capemgr/slots: Permission denied
Feb 22 13:52:00 beaglebone bb_sniff_marco.sh[2478]: attempting to add UART5
Feb 22 13:52:00 beaglebone bb_sniff_marco.sh[2478]: cat: /sys/devices/platform/bone_capemgr/slots: No such file or directory
Feb 22 13:52:00 beaglebone bb_sniff_marco.sh[2478]: /usr/bin/bb_sniff_marco.sh: line 40: /sys/devices/platform/bone_capemgr/slots: Permission denied
Feb 22 13:52:00 beaglebone bb_sniff_marco.sh[2478]: attempting to add SEIZEMOTEG
Feb 22 13:52:00 beaglebone bb_sniff_marco.sh[2478]: cat: /sys/devices/platform/bone_capemgr/slots: No such file or directory
```

RobertCNelson, 2021-02-22:
Kernel overlays don't exist post 4.14.x, so in 4.19.x you need to run u-boot overlays..

If you want kernel overlays just run:
```
sudo /opt/scripts/tools/update_kernel.sh --ti-channel --lts-4_14
```
Kernel overlays are un-supported so good luck!

Regards,

BrentCC, 2021-02-23:
Hi Robert 
I used `sudo /opt/scripts/tools/update_kernel.sh --ti-channel --lts-4_14` to downgrade the kernel to 4.14.108-ti-r137 , but the latest version is 4.14.222. May I know how to get this version?

Thanks!

RobertCNelson, 2021-02-23:
TI moved to 4.19.x after 4.14.108:

https://git.ti.com/gitweb/?p=ti-linux-kernel/ti-linux-kernel.git;a=shortlog;h=refs/heads/ti-linux-4.14.y

if you aren't relying on the TI PRU patch-set, then use the bone channel..

```
sudo /opt/scripts/tools/update_kernel.sh --bone-channel --lts-4_14
```

Which will pull in 4.14.215-bone44..

Why 215 vs 222 -> https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/4.14/

So when the RT patch get's bumped i'll rebuild the old 4.14.x branch..

Regards,