beagleboard/bb.org-overlays #216: RoboticsCape-00A0 is not working in the fresh install of debian on Beaglebone

parzival2, 2021-05-28:
**Describe the bug**
RoboticsCape-00A0 overlay is not working on the fresh install of debian. The board fails to boot with `failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND`

**Describe which image you are running (example: bone-debian-10.2-iot-armhf-2020-02-03-4gb.img.xz) **
The image is the fresh install of https://debian.beagleboard.org/images/bone-debian-10.3-console-armhf-2020-04-06-1gb.img.xz and the board is booting from SD-Card. 

**Describe your hardware**
Hardware is beaglebone black

**Describe any Capes Installed**
I wanted to use RoboticsCape

**Please run this built-in script**
```sudo /opt/scripts/tools/version.sh```

The log output while the u-boot is booting

_**U-boot output**_
```

U-Boot SPL 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600)
Trying to boot from MMC1
Loading Environment from EXT4... ** File not found /boot/uboot.env **

** Unable to read "/boot/uboot.env" from mmc0:1 **

U-Boot 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600), Build: jenkins-github_Bootloader-Builder-137

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
No match for driver 'omap_hsmmc'
No match for driver 'omap_hsmmc'
Some drivers were not found
Reset Source: Power-on reset has occurred.
RTC 32KCLK Source: External.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Loading Environment from EXT4... ** File not found /boot/uboot.env **

** Unable to read "/boot/uboot.env" from mmc0:1 **
Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
BeagleBone Black:
Model: Element14 BeagleBone Black Industrial:
BeagleBone: cape eeprom: i2c_probe: 0x54:
BeagleBone: cape eeprom: i2c_probe: 0x55:
BeagleBone: cape eeprom: i2c_probe: 0x56:
BeagleBone: cape eeprom: i2c_probe: 0x57:
Net:   eth0: MII MODE
cpsw, usb_ether
Press SPACE to abort autoboot in 0 seconds
board_name=[A335BNLT] ...
board_rev=[EIA0] ...
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc0 is current device
gpio: pin 54 (gpio 54) value is 1
Checking for: /uEnv.txt ...
Checking for: /boot.scr ...
Checking for: /boot/boot.scr ...
Checking for: /boot/uEnv.txt ...
gpio: pin 55 (gpio 55) value is 1
2066 bytes read in 33 ms (60.5 KiB/s)
Loaded environment from /boot/uEnv.txt
Checking if uname_r is set in /boot/uEnv.txt...
gpio: pin 56 (gpio 56) value is 1
Running uname_boot ...
loading /boot/vmlinuz-4.19.94-ti-r42 ...
10095592 bytes read in 663 ms (14.5 MiB/s)
debug: [enable_uboot_overlays=1] ...
debug: [enable_uboot_cape_universal=1] ...
debug: [uboot_base_dtb_univ=am335x-boneblack-uboot-univ.dtb] ...
uboot_overlays: [uboot_base_dtb=am335x-boneblack-uboot-univ.dtb] ...
uboot_overlays: Switching too: dtb=am335x-boneblack-uboot-univ.dtb ...
loading /boot/dtbs/4.19.94-ti-r42/am335x-boneblack-uboot-univ.dtb ...
162266 bytes read in 105 ms (1.5 MiB/s)
uboot_overlays: [fdt_buffer=0x60000] ...
uboot_overlays: uboot loading of [/lib/firmware/BB-ADC-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot_overlay_adc=1]...
uboot_overlays: uboot loading of [/lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot_overlay_emmc=1]...
uboot_overlays: uboot loading of [/lib/firmware/BB-HDMI-TDA998x-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot_overlay_video=1]...
uboot_overlays: loading /lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo ...
3801 bytes read in 222 ms (16.6 KiB/s)
uboot_overlays: [dtb_overlay=/lib/firmware/RoboticsCape-00A0.dtbo] ...
uboot_overlays: loading /lib/firmware/RoboticsCape-00A0.dtbo ...
7834 bytes read in 31 ms (246.1 KiB/s)
failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND
loading /boot/initrd.img-4.19.94-ti-r42 ...
6585227 bytes read in 446 ms (14.1 MiB/s)
debug: [console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet] ...
debug: [bootz 0x82000000 0x88080000:647b8b 88000000] ...
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
** Invalid partition 2 **
** Invalid partition 3 **
** Invalid partition 4 **
** Invalid partition 5 **
** Invalid partition 6 **
** Invalid partition 7 **
switch to partitions #0, OK
mmc1(part 0) is current device
** No partition table - mmc 1 **
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc1(part 0) is current device
gpio: pin 54 (gpio 54) value is 1
** No partition table - mmc 1 **
Checking for: /uEnv.txt ...
** No partition table - mmc 1 **
Checking for: /boot.scr ...
** No partition table - mmc 1 **
Checking for: /boot/boot.scr ...
** No partition table - mmc 1 **
Checking for: /boot/uEnv.txt ...
** No partition table - mmc 1 **
** No partition table - mmc 1 **
** No partition table - mmc 1 **
** No partition table - mmc 1 **
** No partition table - mmc 1 **
** No partition table - mmc 1 **
** No partition table - mmc 1 **
```

_**UEnv.txt**_

```
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.19.94-ti-r42
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
dtb_overlay=/lib/firmware/RoboticsCape-00A0.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.14.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_rproc (4.19.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
###pru_uio (4.14.x-ti, 4.19.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet video=HDMI-A-1:1024x768@60e

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
```

Any help regarding this issue would be greatly appreciated. Thanks.

RobertCNelson, 2021-05-28:
Hi @parzival2 please grab this image:

https://rcn-ee.net/rootfs/bb.org/testing/2021-05-28/buster-iot/bone-debian-10.9-iot-armhf-2021-05-28-4gb.img.xz

Then set, in /boot/uEnv.txt 

dtb=am335x-boneblack-roboticscape.dtb
#enable_uboot_overlays=1

Overlays need to be disabled with this cape..

Regards,

parzival2, 2021-05-29:
Thanks for the input. 
Is there any possibility that I can get away with using an overlay? Is the software problem or is it because of the Hardware and we can't do anything about it?
I am asking because I want to try out new Kernel versions, but the Device tree sources are only available for 4.14 versions [here](https://github.com/beagleboard/linux/blob/4.14/arch/arm/boot/dts/am335x-boneblack-roboticscape.dts). Do you think the Device tree sources are compatible with newer Linux versions? I tried compiling them with the newer versions but had some build errors. 
I also want to try out the Yocto builds as I can customize the OS.

RobertCNelson, 2021-05-30:
@parzival2 and generic overlay would have to be rewritten..

As far as it no in any tree other then v4.14.x/v4.19.x, feel free to do a pull request. ;)  I'll gladly merge it..

Regards,

parzival2, 2021-05-30:
> @parzival2 and generic overlay would have to be rewritten..
> 
> As far as why it's not in any tree other then v4.14.x/v4.19.x, feel free to do a pull request. ;) I'll gladly merge it.. Just because i don't use it day to day, doesn't mean other's don't!
> 
> Regards,

Thanks. That was good to hear that it is possible to write an overlay for that board. 
I worked on it a bit and came to know that the DTB files in 4.19 are not compatible with the newer versions(5.4 and so on).
I tried to compile them by checking out the newer version of Linux( I tried 5.4 and 5.10) and they had some build problems but I solved them looking at the videos of how the DTS file should be written and how it works. Here is the solution that I came up with if any one is interested.

### Compiling DTS files to DTB
```
make ARCH=arm CROSS_COMPILE=arm-linux.gnueabihf- am335x-boneblack-roboticscape.dtb
```
for the above command to work, we need to modify the `Makefile` in the `/arch/arm/boot/dts` folder in the Linux Kernel.

```
dtb-$(CONFIG_SOC_AM33XX) += \
	am335x-boneblack-roboticscape.dtb \
```
We have to add the DTS file that we are trying to compile so that it will be switched on when the particular SOC is switched on using the `defconfig`. I used the `bb.org_defconfig` to generate the configuration for the Beaglebone.

The compilation failed due to some missing elements in `am33xx.dts` file. Those are particularly related to `eqep` and `pruss` subsystems.  
 
So I copied and pasted carefully the lines related to PRUSS subsystem to the `am33xx.dtsi` file

```
		pruss_soc_bus: pruss-soc-bus@4a326004 {
			compatible = "ti,am3356-pruss-soc-bus";
			reg = <0x4a326004 0x4>;
			ti,hwmods = "pruss";
			#address-cells = <1>;
			#size-cells = <1>;
			ranges;
			status = "disabled";

			pruss: pruss@4a300000 {
				compatible = "ti,am3356-pruss";
				reg = <0x4a300000 0x80000>;
				interrupts = <20 21 22 23 24 25 26 27>;
				interrupt-names = "host2", "host3", "host4",
						  "host5", "host6", "host7",
						  "host8", "host9";
				#address-cells = <1>;
				#size-cells = <1>;
				ranges;
				status = "disabled";

				pruss_mem: memories@4a300000 {
					reg = <0x4a300000 0x2000>,
					      <0x4a302000 0x2000>,
					      <0x4a310000 0x3000>;
					reg-names = "dram0", "dram1",
						    "shrdram2";
				};

				pruss_cfg: cfg@4a326000 {
					compatible = "syscon";
					reg = <0x4a326000 0x2000>;
				};

				pruss_iep: iep@4a32e000 {
					compatible = "syscon";
					reg = <0x4a32e000 0x31c>;
				};

				pruss_mii_rt: mii-rt@4a332000 {
					compatible = "syscon";
					reg = <0x4a332000 0x58>;
				};

				pruss_intc: interrupt-controller@4a320000 {
					compatible = "ti,am3356-pruss-intc";
					reg = <0x4a320000 0x2000>;
					interrupt-controller;
					#interrupt-cells = <1>;
				};

				pru0: pru@4a334000 {
					compatible = "ti,am3356-pru";
					reg = <0x4a334000 0x2000>,
					      <0x4a322000 0x400>,
					      <0x4a322400 0x100>;
					reg-names = "iram", "control", "debug";
					firmware-name = "am335x-pru0-fw";
					interrupt-parent = <&pruss_intc>;
					interrupts = <16>, <17>;
					interrupt-names = "vring", "kick";
				};

				pru1: pru@4a338000 {
					compatible = "ti,am3356-pru";
					reg = <0x4a338000 0x2000>,
					      <0x4a324000 0x400>,
					      <0x4a324400 0x100>;
					reg-names = "iram", "control", "debug";
					firmware-name = "am335x-pru1-fw";
					interrupt-parent = <&pruss_intc>;
					interrupts = <18>, <19>;
					interrupt-names = "vring", "kick";
				};

				pruss_mdio: mdio@4a332400 {
					compatible = "ti,davinci_mdio";
					reg = <0x4a332400 0x90>;
					clocks = <&dpll_core_m4_ck>;
					clock-names = "fck";
					bus_freq = <1000000>;
					#address-cells = <1>;
					#size-cells = <0>;
					status = "disabled";
				};
			};
		};
```
Other missing elements are in `am33xx-l4.dtsi`. I also copy pasted them following the other identical elements in the 
`epwmss` subsystems.

```
eqep0: eqep@180 {
				compatible = "ti,am33xx-eqep";
				reg = <0x180 0x80>;
				clocks = <&l4ls_gclk>;
				clock-names = "fck";
				interrupt-parent = <&intc>;
				interrupts = <79>;
				status = "disabled";
				};

eqep1: eqep@180 {
				compatible = "ti,am33xx-eqep";
				reg = <0x180 0x80>;
				clocks = <&l4ls_gclk>;
				clock-names = "fck";
				interrupt-parent = <&intc>;
				interrupts = <88>;
				status = "disabled";
				};

eqep2: eqep@0180 {
				compatible = "ti,am33xx-eqep";
				reg = <0x180 0x80>;
				clocks = <&l4ls_gclk>;
				clock-names = "fck";
				interrupt-parent = <&intc>;
				interrupts = <89>;
				status = "disabled";
				};
```
and the last final change that I did is commenting this piece of code in `am335x-bone-common-no-capemgr.dtsi`

```
// &rtc {
// 	clocks = <&clk_32768_ck>, <&l4_per_clkctrl AM3_CLKDIV32K_CLKCTRL 0>;
// 	clock-names = "ext-clk", "int-clk";
// 	system-power-controller;
// };
```
After all theses changes, I compiled and it was successful. I compiled it for 5.10 version of Linux as I was already using it in my Yocto project.

I copied the DTB file over to the `boot` folder and add a file `uEnv.txt` so that u-boot will read the correct DTB file.
### Contents of uEnt.txt
```
ftdfile=am335x-boneblack-roboticscape.dtb
```
I restarted it and it booted normally. I checked the LEDS, I2C buses and they seem to be working properly.
The `i2cdetect` also shows MPU9250 and barometer. 

I will try to make a PR after I have understood the DTS files and making sure the changes that I did are correct. If you can correct me the changes that I have made, it would also be great.

Thanks.

closed 2021-05-30 by parzival2

apper002, 2023-12-03:
After above procedure:
debian@BeagleBone:~$ sudo rc_test_drivers

Kernel: 5.10.168-ti-r71
BeagleBoard.org Debian Bullseye IoT Image 2023-09-02
Debian: 11.8

PASSED: gpio 0
PASSED: gpio 1
PASSED: gpio 2
PASSED: gpio 3
ERROR:  ti-pwm driver not loaded for hrpwm0
ERROR:  ti-pwm driver not loaded for hrpwm1
ERROR:  ti-pwm driver not loaded for hrpwm2
ERROR:  ti-eqep driver not loaded for eqep0
ERROR:  ti-eqep driver not loaded for eqep1
ERROR:  ti-eqep driver not loaded for eqep2
PASSED: pru-rproc
ERROR:  uart1 driver not loaded
ERROR:  uart2 driver not loaded
ERROR:  uart4 driver not loaded
ERROR:  uart5 driver not loaded
PASSED: i2c1
PASSED: i2c2
PASSED: spi
PASSED: LED
PASSED: ADC iio

Currently running on a:
MODEL_BB_BLACK_RC
Robot Control library Version:
1.0.5

Still some wrong syntax in am335x-roboticscape.dtsi.