beagleboard/bb.org-overlays #220: No LEDs after upgrade on a BBGW

JoseGoncalves, 2021-08-31:
**Describe the bug**
After upgrading a BBGW with latest bb-cape-overlays, bb-usb-gadgets, & bb-u-boot-am335x-evm packages I don't have access to BB LEDs at `/sys/class/leds`.
Additionally, doing the same upgrade procedure in a BBGG does not cause any issue.

**Describe your hardware**
Beaglebone Green Wireless

**Please run this built-in script**
```sudo /opt/scripts/tools/version.sh```
```
eeprom:[A335BNLTGW1ABBGW16060242]
model:[TI_AM335x_BeagleBone_Green_Wireless]
dogtag:[BeagleBoard.org Debian Buster Console Image 2021-02-22]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot SPL 2019.04-g10968f0 (Feb 19 2021 - 21:40:51 +0000)]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-g10968f0]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-bonegreen-wireless-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0.bb.org-overlays]
UBOOT: Loaded Overlay:[BB-BBGW-WL1835-00A0.bb.org-overlays]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0.bb.org-overlays]
UBOOT: Loaded Overlay:[BB-FASTER-UAV]
kernel:[4.19.94-ti-r59-inov9]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[dtb_overlay=BB-FASTER-UAV.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210821.0-0~buster+20210821]
pkg:[bb-customizations]:[1.20210810.1-0~buster+20210810]
pkg:[bb-usb-gadgets]:[1.20210816.0-0~buster+20210816]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~buster+20200813]
pkg:[kmod]:[26-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev gpio admin spi iio i2c docker tisdk weston-launch xenomai cloud9ide pwm eqep remoteproc]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait ipv6.disable=1 coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   42.164114] Bluetooth: hci0: change remote baud rate command in firmware
[   51.418276] remoteproc remoteproc0: wkup_m3 is available
[   51.689516] remoteproc remoteproc0: powering up wkup_m3
[   51.689549] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   51.689831] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   54.047752] remoteproc remoteproc1: 4a334000.pru is available
[   54.049967] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   54.047752] remoteproc remoteproc1: 4a334000.pru is available
[   54.047940] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   54.049967] remoteproc remoteproc2: 4a338000.pru is available
[   54.050136] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.909301] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    1.326150] pinctrl-single 44e10800.pinmux: pin PIN30 already requested by 481a6000.serial; cannot claim for leds
[    1.336608] pinctrl-single 44e10800.pinmux: pin-30 (leds) status -22
[    1.343026] pinctrl-single 44e10800.pinmux: could not request pin 30 (PIN30) from group pinmux_bt_pins  on device pinctrl-single
dmesg | grep gpio-of-helper
[    0.921772] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

JoseGoncalves, 2021-09-01:
Reverting to sources from 1a16e046b441a9b1671e0787157d24e545a516a6, recompiling and manually installing in `/lib/firmware` the files `AM335X-PRU-RPROC-4-19-TI-00A0.dtbo`, `BB-ADC-00A0.dtbo`, `BB-BBGW-WL1835-00A0.dtbo` and `BB-BONE-eMMC1-01-00A0.dtbo` fixes this issue.
With this setup, this is the output of `sudo /opt/scripts/tools/version.sh`:
```
eeprom:[A335BNLTGW1ABBGW16060242]
model:[TI_AM335x_BeagleBone_Green_Wireless]
dogtag:[BeagleBoard.org Debian Buster Console Image 2021-02-22]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot SPL 2019.04-g10968f0 (Feb 19 2021 - 21:40:51 +0000)]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-g10968f0]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-bonegreen-wireless-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0.bb.org-overlays]
UBOOT: Loaded Overlay:[BB-BBGW-WL1835-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0.bb.org-overlays]
UBOOT: Loaded Overlay:[BB-FASTER-UAV]
kernel:[4.19.94-ti-r59-inov9]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[dtb_overlay=BB-FASTER-UAV.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210821.0-0~buster+20210821]
pkg:[bb-customizations]:[1.20210810.1-0~buster+20210810]
pkg:[bb-usb-gadgets]:[1.20210816.0-0~buster+20210816]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~buster+20200813]
pkg:[kmod]:[26-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev gpio admin spi iio i2c docker tisdk weston-launch xenomai cloud9ide pwm eqep remoteproc]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait ipv6.disable=1 coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   51.444701] remoteproc remoteproc0: wkup_m3 is available
[   51.559764] remoteproc remoteproc0: powering up wkup_m3
[   51.559791] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   51.560042] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   53.091718] remoteproc remoteproc1: 4a334000.pru is available
[   53.093899] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   53.091718] remoteproc remoteproc1: 4a334000.pru is available
[   53.091915] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   53.093899] remoteproc remoteproc2: 4a338000.pru is available
[   53.094040] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.908702] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.921182] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

JoseGoncalves, 2021-09-03:
This issue is related to recent changes in `BB-BBGW-WL1835-00A0.dts`.
Removing the `leds` section in the latest `BB-BBGW-WL1835-00A0.dts`, i.e.:

```patch
diff --git a/src/arm/BB-BBGW-WL1835-00A0.dts b/src/arm/BB-BBGW-WL1835-00A0.dts
index 1a92689..60c787b 100644
--- a/src/arm/BB-BBGW-WL1835-00A0.dts
+++ b/src/arm/BB-BBGW-WL1835-00A0.dts
@@ -52,18 +52,6 @@
                gpio = <&gpio0 26 0>;
                enable-active-high;
        };
-
-       leds {
-               pinctrl-names = "default";
-               pinctrl-0 = <&bt_pins>;
-               compatible = "gpio-leds";
-
-               wl18xx_bt_en {
-                       label = "wl18xx_bt_en";
-                       gpios = <&gpio1 28 GPIO_ACTIVE_HIGH>;
-                       default-state = "off";
-               };
-       };
 };
 
 &am33xx_pinmux {
```
also fixes this issue for me.

level323, 2023-07-21:
I hit this problem also. Due to multiple reasons, I'm unable to dig deeper into solving this in a constructive way for the wider community, but will note that one way to work around the problem is to hold the following package via the following command (must be run BEFORE running `apt upgrade`):

`apt-mark hold bb-cape-overlays`

This will prevent the upgrade bb-cape-overlays and packages that depend on newer versions when you run `apt upgrade`