beagleboard/bb.org-overlays #221: BB-SPIDEV1-00A0.dtbo conflict

Un1Gfn, 2021-09-24:
BB-SPIDEV1-00A0.dtbo conflict

```plain
pinctrl-single 44e10800.pinmux: pin PIN84 already requested by 48024000.serial; cannot claim for 48030000.spi
pinctrl-single 44e10800.pinmux: pin-84 (48030000.spi) status -22
pinctrl-single 44e10800.pinmux: could not request pin 84 (PIN84) from group pinmux_bb_spi0_pins  on device pinctrl-single
omap2_mcspi 48030000.spi: Error applying setting, reverse things back
omap2_mcspi: probe of 48030000.spi failed with error -22
```

<details><summary>full dmesg</summary>

    [    0.000000] Booting Linux on physical CPU 0x0
    [    0.000000] Linux version 5.11.2-1-ARCH (builduser@leming) (armv7l-unknown-linux-gnueabihf-gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35) #1 PREEMPT Sun Feb 28 22:36:49 UTC 2021
    [    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
    [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
    [    0.000000] OF: fdt: Machine model: TI AM335x BeagleBone Green Wireless
    [    0.000000] Memory policy: Data cache writeback
    [    0.000000] cma: Reserved 48 MiB at 0x9c800000
    [    0.000000] Zone ranges:
    [    0.000000]   Normal   [mem 0x0000000080000000-0x000000009fefffff]
    [    0.000000]   HighMem  empty
    [    0.000000] Movable zone start for each node
    [    0.000000] Early memory node ranges
    [    0.000000]   node   0: [mem 0x0000000080000000-0x000000009fefffff]
    [    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000009fefffff]
    [    0.000000] On node 0 totalpages: 130816
    [    0.000000]   Normal zone: 1150 pages used for memmap
    [    0.000000]   Normal zone: 0 pages reserved
    [    0.000000]   Normal zone: 130816 pages, LIFO batch:31
    [    0.000000] CPU: All CPU(s) started in SVC mode.
    [    0.000000] AM335X ES2.1 (sgx neon)
    [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
    [    0.000000] pcpu-alloc: [0] 0
    [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129666
    [    0.000000] Kernel command line: console=tty0 console=ttyO0,115200n8 root=PARTUUID=6afc7d6f-01 rw rootwait
    [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
    [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
    [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
    [    0.000000] Memory: 439364K/523264K available (14336K kernel code, 1707K rwdata, 3436K rodata, 1024K init, 297K bss, 34748K reserved, 49152K cma-reserved, 0K highmem)
    [    0.000000] random: get_random_u32 called from __kmem_cache_create+0x34/0x3a8 with crng_init=0
    [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
    [    0.000000] ftrace: allocating 43822 entries in 86 pages
    [    0.000000] ftrace: allocated 86 pages with 4 groups
    [    0.000000] rcu: Preemptible hierarchical RCU implementation.
    [    0.000000]   Trampoline variant of Tasks RCU enabled.
    [    0.000000]   Rude variant of Tasks RCU enabled.
    [    0.000000]   Tracing variant of Tasks RCU enabled.
    [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
    [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
    [    0.000000] IRQ: Found an INTC at 0x(ptrval) (revision 5.0) with 128 interrupts
    [    0.000000] TI gptimer clocksource: always-on /ocp/interconnect@44c00000/segment@200000/target-module@31000
    [    0.000002] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
    [    0.000027] clocksource: dmtimer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
    [    0.000400] TI gptimer clockevent: 24000000 Hz at /ocp/interconnect@48000000/segment@0/target-module@40000
    [    0.001583] Console: colour dummy device 80x30
    [    0.002194] printk: console [tty0] enabled
    [    0.002236] WARNING: Your 'console=ttyO0' has been replaced by 'ttyS0'
    [    0.002260] This ensures that you still see kernel messages. Please
    [    0.002281] update your kernel commandline.
    [    0.002345] Calibrating delay loop... 995.32 BogoMIPS (lpj=1990656)
    [    0.048492] pid_max: default: 32768 minimum: 301
    [    0.048770] LSM: Security Framework initializing
    [    0.048945] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
    [    0.048989] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
    [    0.050284] CPU: Testing write buffer coherency: ok
    [    0.050406] CPU0: Spectre v2: using BPIALL workaround
    [    0.051591] Setting up static identity map for 0x80100000 - 0x80100060
    [    0.051809] rcu: Hierarchical SRCU implementation.
    [    0.053496] devtmpfs: initialized
[... 401 lines cut ...]

load mmc 1:1 ${fdt_addr_r}     /boot/dtbs/am335x-bonegreen-wireless.dtb
   # 88862 bytes read
fdt addr $fdt_addr_r

setenv overlayaddr 0x88070000
load mmc 1:1 ${overlayaddr}    /boot/BB-SPIDEV0-00A0.dtbo
   # 1515 bytes read
fdt resize 102400
fdt apply ${overlayaddr}

load mmc 1:1 ${ramdisk_addr_r} /boot/initramfs-linux.img
   # 6876350 bytes read
setenv ramdisk_size ${filesize}

part uuid mmc 1:1 uuid
setenv bootargs "console=tty0 console=${console} root=PARTUUID=${uuid} rw rootwait"
bootz ${kernel_addr_r} ${ramdisk_addr_r}:${ramdisk_size} ${fdt_addr_r}
```

May I disable the SD card reader with some u-boot magic, or I have to either downgrade kernel or switch to some other distro?

PS: My ultimate goal is actually to [flash](https://libreboot.org/docs/install/spi.html#hardware-configuration) my X200 BIOS (winbond [W25Q32BVSSIG](http://www.datasheetcafe.com/25q32bvsig-datasheet-w25q32bvssig-winbond/)) with BBGW instead of frying it with 5V CH341A

Thanks.

RobertCNelson, 2021-09-24:
Your fighting with the grove header on this board: uart2:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm/boot/dts/am335x-bonegreen-common.dtsi?h=v5.15-rc2#n24

```
&am33xx_pinmux {
	uart2_pins: uart2_pins {
		pinctrl-single,pins = <
			AM33XX_PADCONF(AM335X_PIN_SPI0_SCLK, PIN_INPUT, MUX_MODE1)	/* spi0_sclk.uart2_rxd */
			AM33XX_PADCONF(AM335X_PIN_SPI0_D0, PIN_OUTPUT, MUX_MODE1)	/* spi0_d0.uart2_txd */
		>;
	};
};

&uart2 {
	pinctrl-names = "default";
	pinctrl-0 = <&uart2_pins>;
	status = "okay";
};
```
You can just remove it..

and there is a "hardware" bug on this board:

Just "pop" the capacitor..

https://stackoverflow.com/questions/45620504/cannot-configure-spi0-on-beaglebone-green-wireless

Regards,

Un1Gfn, 2021-09-25:
(the previous one was SPIDEV0, title fixed)

SPIDEV1 seems to work

Using am335x-bonegreen.dtb ([dts](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm/boot/dts/am335x-bonegreen.dts?h=v5.11)) ~instead of am335x-bonegreen-wireless.dtb~ and applying BB-SPIDEV**1**-00A0.dtbo ~instead of BB-SPIDEV0-00A0.dtbo~, I get access to `/dev/spidev1.x`

```console
# dmesg | grep -i pin
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129666
[    0.073109] pinctrl core: initialized pinctrl subsystem
[    2.959422] pinctrl-single 44e10800.pinmux: 142 pins, size 568

# flashrom -p linux_spi:dev=/dev/spidev1.0
flashrom v1.2 on Linux 5.11.2-1-ARCH (armv7l)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Using default 2000kHz clock. Use 'spispeed' parameter to override.
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.

# flashrom -p linux_spi:dev=/dev/spidev1.1
flashrom v1.2 on Linux 5.11.2-1-ARCH (armv7l)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Using default 2000kHz clock. Use 'spispeed' parameter to override.
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
```

Can I use it without trouble?

Wi-Fi is obviously offline since they are in conflict w/ spi1 as follows:

```
AM3358BZCZ100.A13/SPI1_SCLK ‣ SPI1_SCLK ‣ TXS0108E_U.B1
AM3358BZCZ100.B13/SPI1_D0 ‣ SPI1_D0 ‣ TXS0108E_U.B2
AM3358BZCZ100.D12/SPI1_D1 ‣ SPI1_D1 ‣ TXS0108E_U.B3
```

, however, will the Wi-Fi chip haunt spi1 even if it is dead?