beagleboard/bb.org-overlays #223: Can't claim spi0 in uboot overlay.  Still claimed by cape-universal?

mfallavol, 2022-01-05:
I am trying to port two overlays that worked under 3.8 to the 4.1x kernel.  The first is to enable the QCA7000 driver which is now in the 4.1 kernel source. 

I started with the Debian Buster IoT Image 2020-04-06, downloaded the kernel source and rebuilt the kernel with the QCA driver and then updated the kernel on the uSD.  That all seemed to go okay so I started working on converting the 3.8 overlay.  

Here is my QCA overlay which compiles and seems to load fine:
`/*
 * Copyright (C) 2014-2022 Valent Power
 *
 * qcaspi Device Tree Overlay
 *
 */
/dts-v1/;
/plugin/;

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/pinctrl/am33xx.h>
#include <dt-bindings/board/am335x-bbw-bbb-base.h>
 
/ {
	/*
	 * Helper to show loaded overlays under: /proc/device-tree/chosen/overlays/
	 */
	fragment@0 {
		target-path="/";
		__overlay__ {

			chosen {
				overlays {
					VP-QCA7K-00A2 = __TIMESTAMP__;
				};
			};
		};
	};

	/*
	 * Free up the pins used by the cape from the pinmux helpers.
	 */
	fragment@1 {
		target = <&ocp>;
		__overlay__ {
			cape-universal { status = "disabled"; };

			P9_17_pinmux { status = "disabled"; };	/* P9_17: spi0_cs0 */
			P9_18_pinmux { status = "disabled"; };	/* P9_18: spi0_d1 (MOSI) */
			P9_21_pinmux { status = "disabled"; };	/* P9_21: spi0_d0 (MISO) */
			P9_22_pinmux { status = "disabled"; };	/* P9_22: spi0_sclk */

			P8_11_pinmux { status = "disabled"; };	/* P8_11: gpio1_13 */
			P8_12_pinmux { status = "disabled"; };	/* P8_12: gpio1_12 */
			P9_27_pinmux { status = "disabled"; };	/* P9_27: gpio3_19 (IRQ) */
		};
	};

	fragment@2 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			bb_spi0_pins: pinmux_bb_spi0_pins {
				pinctrl-single,pins = <
					BONE_P9_17 (PIN_OUTPUT_PULLUP | MUX_MODE0)	/* P9_17: spi0_cs0, OUTPUT_PULLUP | MODE0 */
					BONE_P9_18 (PIN_OUTPUT_PULLUP | MUX_MODE0)	/* P9_18: spi0_d1, OUTPUT_PULLUP | MODE0 */
					BONE_P9_21 (PIN_INPUT_PULLUP | MUX_MODE0)	/* P9_21: spi0_d0, INPUT_PULLUP | MODE0 */
					BONE_P9_22 (PIN_INPUT_PULLUP | MUX_MODE0)	/* P9_22: spi0_sclk, INPUT_PULLUP | MODE0 */
				>;
			};
			qca_gpio_pins: pinmux_qca_gpio_pins {
				pinctrl-single,pins = <
					BONE_P8_11 (PIN_OUTPUT_PULLUP | MUX_MODE7)	 /* P8.11 gpio1_13(13) : QCA_nRESET : GPIO OUTPUT PULLUP (0x17) */
					BONE_P8_12 (PIN_OUTPUT_PULLUP | MUX_MODE7)	 /* P8.12 gpio1_12(12) : QCA_nBOOT : GPIO OUTPUT PULLUP (0x17) */
					BONE_P9_27 (PIN_INPUT | MUX_MODE7)		/* P9.27 gpio3_19(105) : QCA_INT : INPUT_NO-PULL  | MODE7 */
				>;
[... 165 lines cut ...]
[   52.559746] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   52.560038] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   55.695935] remoteproc remoteproc1: 4a334000.pru is available
[   55.710052] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   55.695935] remoteproc remoteproc1: 4a334000.pru is available
[   55.696125] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   55.710052] remoteproc remoteproc2: 4a338000.pru is available
[   55.710234] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    1.043257] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    3.973159] pinctrl-single 44e10800.pinmux: pin PIN13 already requested by ocp:helper; cannot claim for spi0.0
[    4.115767] pinctrl-single 44e10800.pinmux: pin-13 (spi0.0) status -22
[    4.199989] pinctrl-single 44e10800.pinmux: could not request pin 13 (PIN13) from group pinmux_qca_gpio_pins  on device pinctrl-single
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
`
It does look like my pinmux's are getting set correctly.  If I cat PINS I see that PIN12 & PIN13 are now outputs and PIN105 is an input with no pullup.

`pin 12 (PIN12) 44e10830 00000017 pinctrl-single
pin 13 (PIN13) 44e10834 00000017 pinctrl-single
pin 105 (PIN105) 44e109a4 0000002f pinctrl-single
`

mfallavol, 2022-01-05:
Well I made a bit of improvement by removing fragment@4 from the overlay.  Now my boot messages show:
```
Starting kernel ...

[    0.002146] timer_probe: no matching timers found
[    0.192405] l4_wkup_cm:clk:0010:0: failed to disable
[    1.337550] omap2_mcspi 48030000.spi: chipselect 0 already in use
[    1.343833] spi_master spi0: spi_device register error /ocp/spi@48030000/channel@0
[    1.576573] omap_voltage_late_init: Voltage driver support not added
rootfs: clean, 90385/230144 files, 561831/920576 blocks
[   20.150423] qcaspi spi0.0 eth1: qcaspi: unable to get IRQ 0 (irqval=-22).
```

I don't know if the IRQ issue is something to do with the overlay or something to do with not registering spi0.

RobertCNelson, 2022-01-05:
@mfallavol add these two, they are loaded by default for spidev access.. this will nuke them..

```
			channel@0{ status = "disabled"; };
			channel@1{ status = "disabled"; };
```

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-SPI0-CC2520-00A0.dts#L84-L85

mfallavol, 2022-01-05:
@RobertCNelson  I noticed those lines in one of the examples but when I added them I got compiler warnings so I never tried it.  Here is the warnings I just got with DTC 1.4.7: 
```
src/arm/VP-QCA7K-00A2.dtbo: Warning (unique_unit_address): /fragment@3/__overlay__/channel@0: duplicate unit-address (also used in node /fragment@3/__overlay__/qca7000@0)
```
That said, it seems to have done the trick, thank you.  

I fixed my interrupt problem by changing `interrupts = <19>;` to `interrupts = <19 0>;`

Though its too early to be sure, it looks like the qca driver is loaded correctly.  The only thing odd so far is that under the 3.8 kernel ifconfig would show the qca as **qca0** whereas it is showing up now as **eth1**.  Is there any way to change the name back to **qca0**?

RobertCNelson, 2022-01-06:
that might be systemd renaming the network interface... 3.8.x was just soo old, it's hard to say..

mfallavol, 2022-01-06:
I was able to change it by setting net.ifnames=1 on the boot command line and then adding a /etc/systemd/network/10-persistent-net.link file with the contents:
```
    [Match]
    Path=platform-48030000.spi

    [Link]
    Name=qca0
```
The secret was to make sure I updated the initramfs.

closed 2022-01-06 by mfallavol