beagleboard/bb.org-overlays #224: How to setup misc GPIOs not associated with a driver (4.1x kernel)

mfallavol, 2022-01-06:
I have a rather large overlay for a project that sets up an SPI bus, UART, RTC etc but I also need to configure some GPIOs that I control via an application using BBBioLib.  I did all of this with the 3.8 kernel and it all worked fine.  I've converted the overlay and it compiles and seems to load okay and all of the pins seem to be set to the correct mux except for the GPIO pins.

Here is the entire overlay:
```
/*
 * Copyright (C) 2012 Texas Instruments Incorporated - http://www.ti.com/
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License version 2 as
 * published by the Free Software Foundation.
 *
 * MAF 00A0 2019-08-19
 * Created this separate overlay for just the EvaMain reset line
 * and the heartbeat.
 *
 * MAF 00A1 2020-04-21
 * Adding in UART1 from 3.8 kernel source because we are out of slots
 * so I need to combine overlays.
 *
 * MAF 00A2 2020-04-21
 * UART1 was successful so adding BB-SPIDEV1 as well.
 *   SPI on pins P9.28 P9.29 P9.30 P9.31
 *
 * MAF 00A3 2021-03-11
 * EM-nRESET behavior inverted and renamed to SAM_RESET_REQ
 * Added RTC fragment
 *
 * MAF 00A3 2021-03-11
 * Added buzzer to P9.15
 *
 * MAF 00A4 2021-05-12
 * Added chip select for FPGA on P8.24
 *
 * MAF 00A5 2021-12-29
 * Converting for 4.19 kernel
 * 
 */

/dts-v1/;
/plugin/;

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/pinctrl/am33xx.h>
#include <dt-bindings/board/am335x-bbw-bbb-base.h>

/ {
	/*
	 * Helper to show loaded overlays under: /proc/device-tree/chosen/overlays/
	 */
	fragment@0 {
		target-path="/";
		__overlay__ {

			chosen {
				overlays {
					VP-EVA-00A5 = __TIMESTAMP__;
				};
			};
		};
	};

	/*
	 * Free up the pins used by the cape from the pinmux helpers.
	 */
	fragment@1 {
		target = <&ocp>;
		__overlay__ {
			cape-universal { status = "disabled"; };

			P9_15_pinmux { status = "disabled"; };	/* P9_15:  */
			P9_16_pinmux { status = "disabled"; };	/* P9_16:  */
			P9_23_pinmux { status = "disabled"; };	/* P9_23:  */
			P9_25_pinmux { status = "disabled"; };	/* P9_25:  */

			P9_24_pinmux { status = "disabled"; };	/* P9_24: uart1_txd */
			P9_26_pinmux { status = "disabled"; };	/* P9_26: uart1_rxd */

			P8_24_pinmux { status = "disabled"; };	/* P8_24:  */
			P9_28_pinmux { status = "disabled"; };	/* P9_28: mcasp0_ahclkr.spi1_cs0 */
			P9_29_pinmux { status = "disabled"; };	/* P9_29: mcasp0_fsx.spi1_d0 */
			P9_30_pinmux { status = "disabled"; };	/* P9_30: mcasp0_axr0.spi1_d1 */
			P9_31_pinmux { status = "disabled"; };	/* P9_31: mcasp0_aclkx.spi1_sclk */
		};
	};

	fragment@2 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			bb_uart1_pins: pinmux_bb_uart1_pins {
				pinctrl-single,pins = <
					BONE_P9_24 (PIN_OUTPUT | MUX_MODE0)	// uart1_txd.uart1_txd
					BONE_P9_26 (PIN_INPUT | MUX_MODE0)	// uart1_rxd.uart1_rxd
				>;
			};
		};
	};

	fragment@3 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			bb_spi1_pins: pinmux_bb_spi1_pins {
				pinctrl-single,pins = <
					BONE_P9_31 (PIN_INPUT | MUX_MODE3)	/* P9_31 (A13) mcasp0_aclkx.spi1_sclk */
					BONE_P9_29 (PIN_INPUT | MUX_MODE3)	/* P9_29 (B13) mcasp0_fsx.spi1_d0 */
					BONE_P9_30 (PIN_INPUT | MUX_MODE3)	/* P9_30 (D12) mcasp0_axr0.spi1_d1 */
					BONE_P9_28 (PIN_INPUT | MUX_MODE3)	/* P9_28 (C12) mcasp0_ahclkr.spi1_cs0 */
				>;
			};
		};
	};

	fragment@4 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			eva_gpio_pins: pinmux_eva_gpio_pins {
				pinctrl-single,pins = <
					BONE_P8_24 (PIN_OUTPUT_PULLUP | MUX_MODE7)	/* P8.24 (gpio1 1)  EP_CS2 : GPIO OUTPUT PULLUP (0x17) */
					BONE_P9_15 (PIN_OUTPUT_PULLDOWN | MUX_MODE7)	/* P9.15 (gpio1 16) BUZZER : GPIO OUTPUT PULLDOWN (0x07) */
					BONE_P9_16 ( PIN_INPUT | MUX_MODE7)		/* P9.16 (gpio1 19) EP_nINTR : GPIO INPUT NO-PULL (0x3F) */
					BONE_P9_23 (PIN_OUTPUT_PULLDOWN | MUX_MODE7)	/* P9.23 (gpio1 17) EM-HB : GPIO OUTPUT PULLDOWN (0x07) */
					BONE_P9_25 (PIN_OUTPUT_PULLUP | MUX_MODE7)	/* P9.25 (gpio3 21) SAM_RESET_REQ : GPIO OUTPUT PULLUP (0x17) */
				>;
			};
		};
	};

	fragment@10 {
		target = <&ocp>;
		__overlay__ {
			eva {
				compatible = "eva-gpio";
				status = "okay";
				pinctrl-names = "default";
				pinctrl-0 = <&eva_gpio_pins>;

				gpios = <
					&gpio1 1  0
					&gpio1 16 0
					&gpio1 19 0
					&gpio1 17 0
					&gpio3 21 0
				>;
			};
		};
	};

	fragment@11 {
		target = <&uart1>;
		__overlay__ {
			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <&bb_uart1_pins>;
		};
	};

	fragment@12 {
		target = <&spi1>;
		__overlay__ {
			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <&bb_spi1_pins>;

			#address-cells = <1>;
			#size-cells = <0>;

			channel@0 {
				#address-cells = <1>;
				#size-cells = <0>;

				compatible = "spidev";
				symlink = "bone/spi/1.0";

				reg = <0>;
				spi-max-frequency = <16000000>;
				spi-cpha;
			};

			channel@1 {
				#address-cells = <1>;
				#size-cells = <0>;

				compatible = "spidev";
				symlink = "bone/spi/1.1";

				reg = <1>;
				spi-max-frequency = <16000000>;
			};
		};
	};

	fragment@20 {
		target-path="/";
		__overlay__ {
			aliases {
				rtc0 = &extrtc;
				/* The OMAP RTC implementation in the BBB is
				 * buggy, so that it cannot be used as a
				 * battery-backed RTS, so that it loses its
				 * contents when power is removed from the
				 * Beaglebone...
				 *
				 * We move the omap built-in RTC to rtc1, so
				 * that userspace defaults to using the DS1307.
				 *
				 * The omap RTC must remain enabled because it
				 * is also used during the reboot process on the
				 * BBB.
				 */
				rtc1 = "/ocp/rtc@44e3e000";
			};
		};
	};

	fragment@21 {
		target = <&i2c2>;
		__overlay__ {
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			/* pinctrl-0 = <&i2c2_pins>; */

			clock-frequency = <100000>;

			extrtc: mcp7941@6F { /* Real time clock defined as a child of the i2c2 bus */
				compatible = "maxim,mcp7941x";
				reg = <0x6F>;
			};
		};
	};	
};
```
The section that I'm concerned about fragment 10:
```
 	fragment@10 {
		target = <&ocp>;
		__overlay__ {
			eva {
				compatible = "eva-gpio";
				status = "okay";
				pinctrl-names = "default";
				pinctrl-0 = <&eva_gpio_pins>;

				gpios = <
					&gpio1 1  0
					&gpio1 16 0
					&gpio1 19 0
					&gpio1 17 0
					&gpio3 21 0
				>;
			};
		};
	};
```

Is this the correct way to specify the GPIO pins?

RobertCNelson, 2022-01-06:
@mfallavol so gpio never got it's own standard driver.. thus, we all abuse "gpio-led"... 

best of all you can set a startup state or trigger..

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/leds/leds-gpio.yaml

Regards,

mfallavol, 2022-01-06:
Thanks @RobertCNelson I wondered about that.  So gpio-leds for outputs, I assume then gpio-keys for inputs?

RobertCNelson, 2022-01-06:
exactly!

mfallavol, 2022-01-07:
Thanks again @RobertCNelson.  M fragment now looks like this:
```
	fragment@10 {
		target = <&ocp>;
		__overlay__ {
			leds {
				compatible = "gpio-leds";

				P8_24 {
					gpios = <&gpio1 1 GPIO_ACTIVE_HIGH>;
					default-state = "on";
				};

				P9_15 {
					gpios = <&gpio1 16 GPIO_ACTIVE_HIGH>;
					default-state = "off";
				};

				P9_23 {
					gpios = <&gpio1 17 GPIO_ACTIVE_HIGH>;
					default-state = "off";
				};

				P9_25 {
					gpios = <&gpio3 21 GPIO_ACTIVE_HIGH>;
					default-state = "on";
				};
			};

			gpio-keys {
				compatible = "gpio-keys";

				P9_16 {
					label = "EP_nINTR";
					gpios = <&gpio1 19 0>;
					linux,code = <100>;
				};
			};
		};
	};
```
It seems to be working now.  I can see my outputs under /sys/class/leds but I can't find anything similar for my input (P9_16).  That said if I export the GPIO it shows the direction as 'in'.

The only other thing that is a little bothersome is that the pin modes shown by cat'ing $PINS does not seem to reflect the expected mode but only for these GPIO pins.  All of the other pins I set in the overlay seem to be show as expected.