beagleboard/bb.org-overlays #225: How to free up emmc pin (8.24) for GPIO?

mfallavol, 2022-01-24:
I need to use P8.24 for GPIO.  The default mode for this pin is 31h (mmc1_dat1)

I've modified uEnv.txt to disable eemc: `disable_uboot_overlay_emmc=1`
and I see the appropriate message on boot:
```uboot_overlays: uboot loading of [/lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo] disa                                       bled by /boot/uEnv.txt [disable_uboot_overlay_emmc=1]...```

My overlay attempts to free up the pin with:
```P8_24_pinmux { status = "disabled"; };```
but I'm wondering if that is the correct way to do it for an emmc pin because I do the same thing with all of my other GPIOs and they change modes appropriately but P8.24 is still 31h and I can't take control of it.  show-pins shows:
```P8.24 / eMMC d1                    1 fast rx  up  1 mmc 1 d1```
where-as my other GPIO outputs show up like this:
```
P9.23                             17 fast rx down 7 gpio 1.17 lo >>  P9_23
P9.25 / audio osc                107 fast rx down 7 gpio 3.21 hi >>  P9_25
```

Is there something I'm missing?

RobertCNelson, 2022-01-24:
That's the correct way... but note, not every "eMMC" can be disabled... Using the eMMC pins should be considered a last resort..

Regards,

mfallavol, 2022-01-24:
Thanks @RobertCNelson but is there a reason this would have changed with the 4.x kernel and UBOOT overlays vs the 3.8 kernel?  This same BBB with 3.8.13 works.  (With a different overlay of course.)

RobertCNelson, 2022-01-24:
Nope, if it worked before, i'd expect it to work now (with the exact same hardware)...  I forget which model of eMMC on some boards can't be disabled..

Regards,