beagleboard/bb.org-overlays #226: Universal cape: selection of the GPO lines level does not work

avives-keonn, 2022-03-02:
Hello,

I'm using the standard universal cape to control a set of GPO lines from a custom board without any cape.

The standard /sys/class/gpio/gpioNN/ access to gpio functionality does not seem to work. This is the first attempt at kernel 5.x and I may be missing some changes.
The gpio structure is created as usual, I can change the direction of the GPIO line, but any change to the _value_ file is not reflected into the hardware level.

There are a couple of lines at boot time I'm not sure which effect may have

```
[   10.655542] gpio-of-helper ocp:cape-universal: Failed to get gpio property of 'P8_03'
[   10.655574] gpio-of-helper ocp:cape-universal: Failed to create gpio entry
```

I'm using a Bone black with the latest minimal debian image, without major changes so far.
am335x-debian-11.2-minimal-armhf-2022-03-01-2gb.img.xz

The uEnv.txt is pretty simple at the moment

```
uname_r=5.10.100-ti-r37

###Master Enable
enable_uboot_overlays=1
###

#disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
disable_uboot_overlay_adc=1

###
###Cape Universal Enable
enable_uboot_cape_universal=1
###

console=ttyS0,115200n8
cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet
```

And the result of the beagle-version is
```
eeprom:[A335BNLT00C04417BBBK1664]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Bullseye Minimal Image 2022-03-01]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot SPL 2021.10-gc2b239f7 (Dec 17 2021 - 16:57:21 +0000)]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot SPL 2021.10-gc2b239f7 (Dec 17 2021 - 16:57:21 +0000)]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0.kernel]
kernel:[5.10.100-ti-r37]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210821.0-0~bullseye+20210821]
pkg:[bb-customizations]:[1.20220119.0-0~bullseye+20220125]
pkg:[bb-usb-gadgets]:[1.20220112.3-0~bullseye+20220112]
pkg:[bb-wl18xx-firmware]:[1.20211222.2-0~bullseye+20211222]
pkg:[kmod]:[28-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20210315-3]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input netdev gpio admin i2c tisdk weston-launch bluetooth cloud9ide]
cmdline:[console=ttyS0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   29.970381] remoteproc remoteproc0: wkup_m3 is available
[   30.046671] remoteproc remoteproc0: powering up wkup_m3
[... 61 lines cut ...]
[   11.334008] gpio-of-helper ocp:cape-universal: Allocated GPIO id=49 name='P9_16'
[   11.334181] gpio-of-helper ocp:cape-universal: Allocated GPIO id=50 name='P9_17'
[   11.334349] gpio-of-helper ocp:cape-universal: Allocated GPIO id=51 name='P9_18'
[   11.334602] gpio-of-helper ocp:cape-universal: Allocated GPIO id=52 name='P9_19'
[   11.334792] gpio-of-helper ocp:cape-universal: Allocated GPIO id=53 name='P9_20'
[   11.334964] gpio-of-helper ocp:cape-universal: Allocated GPIO id=54 name='P9_21'
[   11.335127] gpio-of-helper ocp:cape-universal: Allocated GPIO id=55 name='P9_22'
[   11.335335] gpio-of-helper ocp:cape-universal: Allocated GPIO id=56 name='P9_23'
[   11.335498] gpio-of-helper ocp:cape-universal: Allocated GPIO id=57 name='P9_24'
[   11.335999] gpio-of-helper ocp:cape-universal: Allocated GPIO id=58 name='P9_25'
[   11.336190] gpio-of-helper ocp:cape-universal: Allocated GPIO id=59 name='P9_26'
[   11.336356] gpio-of-helper ocp:cape-universal: Allocated GPIO id=60 name='P9_27'
[   11.336529] gpio-of-helper ocp:cape-universal: Allocated GPIO id=61 name='P9_28'
[   11.336688] gpio-of-helper ocp:cape-universal: Allocated GPIO id=62 name='P9_29'
[   11.336880] gpio-of-helper ocp:cape-universal: Allocated GPIO id=63 name='P9_30'
[   11.337066] gpio-of-helper ocp:cape-universal: Allocated GPIO id=64 name='P9_31'
[   11.337252] gpio-of-helper ocp:cape-universal: Allocated GPIO id=65 name='P9_41'
[   11.337429] gpio-of-helper ocp:cape-universal: Allocated GPIO id=66 name='P9_91'
[   11.337590] gpio-of-helper ocp:cape-universal: Allocated GPIO id=67 name='P9_42'
[   11.337760] gpio-of-helper ocp:cape-universal: Allocated GPIO id=68 name='P9_92'
[   11.337771] gpio-of-helper ocp:cape-universal: ready
END
```

Thanks in advance

RobertCNelson, 2022-03-02:
HI @avives-keonn thanks for testing!  Sorry our v5.10.x has been a regression nightmare, i'll take a look at this.. What pin are you specifically looking at? just for reference, so i can triple check with a logic analyzer..

Regards,

avives-keonn, 2022-03-02:
Hello,

We've tested P9_12, P9_23 and P8_17 with the same result, we actually use around 25 lines. Also it seems 5.15 has the same problem
What would be the latest LTS kernel we should be using?

Thanks,

avives-keonn, 2022-03-02:
By the way, UART line 1 work fine by directly using /dev/ttyS1, the /dev/ttyOx are not created.

RobertCNelson, 2022-03-02:
4.19.x-ti i'd consider our main fall back for fully functionally and fully tested...

https://forum.beagleboard.org/t/debian-10-x-11-x-kernel-updates/30928

avives-keonn, 2022-03-02:
Hi,

It works perfectly.
In terms of CVE patches, is 4.19.x-ti also a good recommendation?

Thanks,

RobertCNelson, 2022-03-02:
Hi @avives-keonn not really..... https://git.ti.com/gitweb/?p=ti-linux-kernel/ti-linux-kernel.git;a=shortlog;h=refs/heads/ti-linux-4.19.y

Ti moved on to v5.4.x, v5.10.x..

Are you using ti's remoteproc pru or the mainline "uio pru" version?

looks at your log... nope neither..

So, you can switch to the "v4.19.x-bone" version.. While this doesn't have the full "TI" patchset, the am335x is pretty good on mainline (minus the remoteproc pru..)   this is currently: 4.19.221-bone74

my 4.19.x branch sync's with 4.19.x-rt... https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/4.19/  so it looks like i need to bump it to 4.19.230-bone75... ;)

Regards,

avives-keonn, 2022-03-02:
Hi,

Thanks a lot!
We will switch to the v4.19.x-bone version, as we are require to be parched as much as possible.
What would be the best way to say alert to new releases?

Thanks,

RobertCNelson, 2022-03-02:
The easy way on Debian 10.x/11.x is:

```
sudo apt update
sudo apt install bbb.io-kernel-4.19-bone
```

This will follow all stable kernel updates, thru "apt update/apt upgrade"..

Regards,

RobertCNelson, 2022-03-02:
You can pretty much guess a new update about once a month...

https://github.com/RobertCNelson/bb-kernel/commits/am33x-v4.19

avives-keonn, 2022-03-02:
Noted with thanks

closed 2022-03-02 by avives-keonn