beagleboard/bb.org-overlays #229: Issues claiming ports...

ensimek, 2022-12-23:
Hi guys, 
Have new BBB where i want to claim all P8[0] bank as inputs and P8[1] as outputs (outputs not done yet). 
Disabled all overlays + universal one too. However I am getting bunch of conflicts like:
[    1.499283] pinctrl-single 44e10800.pinmux: could not request pin 8 (PIN8) from group pinmux_P8_19_default_pin  on device pinctrl-single
[    1.522002] pinctrl-single 44e10800.pinmux: pin PIN32 already requested by ocp:helper; cannot claim for ocp:P8_21_pinmux
[    1.532997] pinctrl-single 44e10800.pinmux: pin-32 (ocp:P8_21_pinmux) status -22

Running: Linux beaglebone 4.19.94-ti-r42 #1buster SMP PREEMPT Tue Mar 31 19:38:29 UTC 2020 armv7l GNU/Linux

root@beaglebone:/opt/scripts/tools# ./version.sh
git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLT00C02212SBB24899]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
kernel:[4.19.94-ti-r42]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/TBOX_INPUTS-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[enable_uboot_cape_universal=0]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210821.0-0~buster+20210821]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~buster+20200322]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.949624] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    1.157165] pinctrl-single 44e10800.pinmux: pin PIN6 already requested by ocp:helper; cannot claim for ocp:P8_03_pinmux
[    1.168107] pinctrl-single 44e10800.pinmux: pin-6 (ocp:P8_03_pinmux) status -22
[    1.175494] pinctrl-single 44e10800.pinmux: could not request pin 6 (PIN6) from group pinmux_P8_03_default_pin  on device pinctrl-single
[    1.197150] pinctrl-single 44e10800.pinmux: pin PIN2 already requested by ocp:helper; cannot claim for ocp:P8_05_pinmux
[    1.208048] pinctrl-single 44e10800.pinmux: pin-2 (ocp:P8_05_pinmux) status -22
[    1.215419] pinctrl-single 44e10800.pinmux: could not request pin 2 (PIN2) from group pinmux_P8_05_default_pin  on device pinctrl-single
[    1.237008] pinctrl-single 44e10800.pinmux: pin PIN36 already requested by ocp:helper; cannot claim for ocp:P8_07_pinmux
[    1.247984] pinctrl-single 44e10800.pinmux: pin-36 (ocp:P8_07_pinmux) status -22
[    1.255442] pinctrl-single 44e10800.pinmux: could not request pin 36 (PIN36) from group pinmux_P8_07_default_pin  on device pinctrl-single
[    1.277330] pinctrl-single 44e10800.pinmux: pin PIN39 already requested by ocp:helper; cannot claim for ocp:P8_09_pinmux
[    1.288306] pinctrl-single 44e10800.pinmux: pin-39 (ocp:P8_09_pinmux) status -22
[    1.295764] pinctrl-single 44e10800.pinmux: could not request pin 39 (PIN39) from group pinmux_P8_09_default_pin  on device pinctrl-single
[    1.317862] pinctrl-single 44e10800.pinmux: pin PIN13 already requested by ocp:helper; cannot claim for ocp:P8_11_pinmux
[    1.328850] pinctrl-single 44e10800.pinmux: pin-13 (ocp:P8_11_pinmux) status -22
[    1.336308] pinctrl-single 44e10800.pinmux: could not request pin 13 (PIN13) from group pinmux_P8_11_default_pin  on device pinctrl-single
[    1.358554] pinctrl-single 44e10800.pinmux: pin PIN9 already requested by ocp:helper; cannot claim for ocp:P8_13_pinmux
[    1.369448] pinctrl-single 44e10800.pinmux: pin-9 (ocp:P8_13_pinmux) status -22
[    1.376820] pinctrl-single 44e10800.pinmux: could not request pin 9 (PIN9) from group pinmux_P8_13_default_pin  on device pinctrl-single
[    1.399121] pinctrl-single 44e10800.pinmux: pin PIN15 already requested by ocp:helper; cannot claim for ocp:P8_15_pinmux
[    1.410101] pinctrl-single 44e10800.pinmux: pin-15 (ocp:P8_15_pinmux) status -22
[    1.417561] pinctrl-single 44e10800.pinmux: could not request pin 15 (PIN15) from group pinmux_P8_15_default_pin  on device pinctrl-single
[    1.440135] pinctrl-single 44e10800.pinmux: pin PIN11 already requested by ocp:helper; cannot claim for ocp:P8_17_pinmux
[    1.451103] pinctrl-single 44e10800.pinmux: pin-11 (ocp:P8_17_pinmux) status -22
[    1.458560] pinctrl-single 44e10800.pinmux: could not request pin 11 (PIN11) from group pinmux_P8_17_default_pin  on device pinctrl-single
[    1.481011] pinctrl-single 44e10800.pinmux: pin PIN8 already requested by ocp:helper; cannot claim for ocp:P8_19_pinmux
[    1.491910] pinctrl-single 44e10800.pinmux: pin-8 (ocp:P8_19_pinmux) status -22
[    1.499283] pinctrl-single 44e10800.pinmux: could not request pin 8 (PIN8) from group pinmux_P8_19_default_pin  on device pinctrl-single
[    1.522002] pinctrl-single 44e10800.pinmux: pin PIN32 already requested by ocp:helper; cannot claim for ocp:P8_21_pinmux
[    1.532997] pinctrl-single 44e10800.pinmux: pin-32 (ocp:P8_21_pinmux) status -22
[    1.540456] pinctrl-single 44e10800.pinmux: could not request pin 32 (PIN32) from group pinmux_P8_21_default_pin  on device pinctrl-single
[    1.562983] pinctrl-single 44e10800.pinmux: pin PIN4 already requested by ocp:helper; cannot claim for ocp:P8_23_pinmux
[... 15 lines cut ...]
[    1.781441] pinctrl-single 44e10800.pinmux: pin-53 (ocp:P8_33_pinmux) status -22
[    1.788904] pinctrl-single 44e10800.pinmux: could not request pin 53 (PIN53) from group pinmux_P8_33_default_pin  on device pinctrl-single
[    1.812472] pinctrl-single 44e10800.pinmux: pin PIN52 already requested by ocp:helper; cannot claim for ocp:P8_35_pinmux
[    1.823449] pinctrl-single 44e10800.pinmux: pin-52 (ocp:P8_35_pinmux) status -22
[    1.830909] pinctrl-single 44e10800.pinmux: could not request pin 52 (PIN52) from group pinmux_P8_35_default_pin  on device pinctrl-single
[    1.855076] pinctrl-single 44e10800.pinmux: pin PIN48 already requested by ocp:helper; cannot claim for ocp:P8_37_pinmux
[    1.866087] pinctrl-single 44e10800.pinmux: pin-48 (ocp:P8_37_pinmux) status -22
[    1.873549] pinctrl-single 44e10800.pinmux: could not request pin 48 (PIN48) from group pinmux_P8_37_default_pin  on device pinctrl-single
[    1.898031] pinctrl-single 44e10800.pinmux: pin PIN46 already requested by ocp:helper; cannot claim for ocp:P8_39_pinmux
[    1.909029] pinctrl-single 44e10800.pinmux: pin-46 (ocp:P8_39_pinmux) status -22
[    1.916490] pinctrl-single 44e10800.pinmux: could not request pin 46 (PIN46) from group pinmux_P8_39_default_pin  on device pinctrl-single
[    1.941358] pinctrl-single 44e10800.pinmux: pin PIN44 already requested by ocp:helper; cannot claim for ocp:P8_41_pinmux
[    1.952346] pinctrl-single 44e10800.pinmux: pin-44 (ocp:P8_41_pinmux) status -22
[    1.959806] pinctrl-single 44e10800.pinmux: could not request pin 44 (PIN44) from group pinmux_P8_41_default_pin  on device pinctrl-single
[    1.984910] pinctrl-single 44e10800.pinmux: pin PIN42 already requested by ocp:helper; cannot claim for ocp:P8_43_pinmux
[    1.995896] pinctrl-single 44e10800.pinmux: pin-42 (ocp:P8_43_pinmux) status -22
[    2.003357] pinctrl-single 44e10800.pinmux: could not request pin 42 (PIN42) from group pinmux_P8_43_default_pin  on device pinctrl-single
[    2.028708] pinctrl-single 44e10800.pinmux: pin PIN40 already requested by ocp:helper; cannot claim for ocp:P8_45_pinmux
[    2.039694] pinctrl-single 44e10800.pinmux: pin-40 (ocp:P8_45_pinmux) status -22
[    2.047153] pinctrl-single 44e10800.pinmux: could not request pin 40 (PIN40) from group pinmux_P8_45_default_pin  on device pinctrl-single
dmesg | grep gpio-of-helper
[    0.963063] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

RobertCNelson, 2022-12-23:
```
uboot_overlay_options:[enable_uboot_cape_universal=0]
```

it's not the 'value' that enables it.. it's the fact the variable is 'set'..

```
#enable_uboot_cape_universal=1
```

Regards,

ensimek, 2022-12-23:
> ```
> uboot_overlay_options:[enable_uboot_cape_universal=0]
> ```
> 
> it's not the 'value' that enables it.. it's the fact the variable is 'set'..
> 
> ```
> #enable_uboot_cape_universal=1
> ```
> 
> Regards,

O wow, i got so confused with it. Commented out and problem is gone. However I am not sure if this is related but regardless how i set P8_3 input (pull down or pull up) it always values to 1 even on open circuit.. haven't checked all P[0] inputs but some of them when measured to the GND are 0 and some are 1. I thought if i set all of them the same way 0x27 or 0x37 they will be the same but aren't. Am I doing something wrong here?

RobertCNelson, 2022-12-23:
internal pull-up resistor might be set..

i wouldn't trust any value with an open-circuit..

ensimek, 2022-12-23:
Unfortunately project on the desk but i will try to wire it up to have 'real values'. So these internal pullups. Guess i cannot control them if 0x27 doesn't work?

ensimek, 2022-12-23:
So this is how i see it:
/opt/scripts/device/bone/show-pins.pl
P8.03 / eMMC d6                    6  R9 fast rx down 7 gpio 1.06        ocp/helper (pinctrl_tbox_0_pins)

It says its mode 7 and pulldown which is correct. However when i export 38 and do cat value, i would expect 0, but it shows me 1 while nothing is connected to P8_3 PIN