beagleboard/bb.org-overlays #230: Strict ordering of custom uio devices *after* uio_pruss?

wohali, 2023-07-07:
I read through #138 which is similar to this, but not strictly the same thing. Hoping @mvduin can shed some light...

No matter what I try (am on @RobertCNelson 's latest 5.10 / bullseye image), I can't get the pruss uio nodes to load before a different, custom dts `compatible = "uio"` block (Or, conversely, I can't delay the new `uio` block to be enumerated after the pruss uio block.)

This is causing headaches with the old `am335x_pru_package` stuff where include `#define`s such as:

    #define PRU_EVTOUT_0 0

and

    #define PRUSS_UIO_DRV_PRUSS_BASE "/sys/class/uio/uio0/maps/map0/addr"

are no longer accurate -- as my custom block is now `uio0` and `uio_pruss` is at `uio[1-8]`.

I've tried:
* reordering things in `/boot/uEnv.txt`
* using `dtb_overlay` vs. `uboot_overlay_addr#` vs. other
* commenting out `uboot_overlay_pru=AM335X-PRU-UIO-00A0.dtbo` and copying that dts's contents into my own dts instead, then changing the order of declarations inside the combined dts file

Nothing seems to do the trick. I _suspect_ it's because `uio_pruss` is compiled as a module, and the dts-declared uio is loaded before the module is loaded, but I haven't tried a custom kernel compile to validate that theory.

I'd **really** rather not have to patch `am335x_pru_package` for this application. And porting everything to rproc/etc is a non-starter.

Is there any hope?

mvduin, 2023-07-07:
Yeah it's pretty awful that libprussdrv just rudely hardcodes a number that's dynamically allocated by the kernel.

Both uio_pdrv_genirq and uio_pruss are compiled as module, so it should be possible to arrange for the loading of uio_pdrv_genirq to be deferred. The problem is likely that uio_pdrv_genirq gets unconditionally loaded super early, I think as a result of it having an of_id match on `""` (filled in by kernel parameter, but modprobe doesn't know that). You'd need to blacklist it or omit it from initrd (if you're using that).

Patching libprussdrv to not make braindead assumption would definitely be the best solution though. If your system is recent enough it'll have an udev rule that creates symlinks to the devices, e.g. to access the memory space you'd use `/dev/uio/pruss/module` instead of `/dev/uio0` and to access the interrupts you'd use `/dev/uio/pruss/irq2..9` instead of `/dev/uio0..7`:
```
@beaglebone:~$ ls -ld /dev/uio/pruss/*
lrwxrwxrwx 1 root root 10 May  9 13:51 /dev/uio/pruss/irq2 -> ../../uio0
lrwxrwxrwx 1 root root 10 May  9 13:51 /dev/uio/pruss/irq3 -> ../../uio1
lrwxrwxrwx 1 root root 10 May  9 13:51 /dev/uio/pruss/irq4 -> ../../uio2
lrwxrwxrwx 1 root root 10 May  9 13:51 /dev/uio/pruss/irq5 -> ../../uio3
lrwxrwxrwx 1 root root 10 May  9 13:51 /dev/uio/pruss/irq6 -> ../../uio4
lrwxrwxrwx 1 root root 10 May  9 13:51 /dev/uio/pruss/irq7 -> ../../uio5
lrwxrwxrwx 1 root root 10 May  9 13:51 /dev/uio/pruss/irq8 -> ../../uio6
lrwxrwxrwx 1 root root 10 May  9 13:51 /dev/uio/pruss/irq9 -> ../../uio7
lrwxrwxrwx 1 root root 10 May  9 13:51 /dev/uio/pruss/module -> ../../uio0
```
For the sysfs directory you can use something like this: (completely untested)
```c
#define _GNU_SOURCE
#include <stdio.h>
#include <sys/sysmacros.h>
#include <sys/stat.h>
char *PRUSS_UIO_DRV_PRUSS_BASE = NULL;
struct stat sb;
stat("/dev/uio/pruss/module", &sb);  // FIXME check for errors
asprintf( &PRUSS_UIO_DRV_PRUSS_BASE, "/sys/char/%u:%u/maps/map0/addr", major(sb.st_dev), minor(sb.st_dev) );
```

If you don't want to rely on symlinks created by an udev rule your best option would probably be to just scan `/sys/class/uio/` or use `libudev`.

wohali, 2023-07-07:
Thanks for the tips. It's a hack, but it works, and minimal code changes needed on my side:

```
$ cat /etc/modprobe.d/uio-blacklist.conf
blacklist uio_pdrv_genirq
```

followed by a generous helping of

```
update-initramfs -u -k `uname -r
```

and a startup script that does

```
modprobe uio_pdrv_genirq
```

Thanks for the nudge in the right direction.

If https://github.com/beagleboard/am335x_pru_package  would be interested in a PR to remove all the hardcoded values, and use a smarter approach (walk the `/sys` or `/dev/uio` tree to find the correct devices, with appropriate fallbacks for older kernels) I'd be willing to work on it, time permitting. Somehow I don't think that's happening. 😉

closed 2023-07-07 by wohali