beagleboard/bb.org-overlays #231: BB-I2C2-RTC-DSX.dtbo not functioning with 5.10.168-X kernel

rbed23, 2023-11-23:
I have tried _unsuccessfully_ to setup/run the DS1307 overlay on either kernels:
- 5.10.168-bone-rt-r71
- 5.10.168-ti-rt-72 (desired, bc incorporates the reset-gpios, reset-assert-us, reset-deassert-us for PCB RevC.3 ETH PHY fix)

It **does** run successfully with the following:
- 5.4.161-bone-rt-r63

uname -r
```
5.4.161-bone-rt-r63
```

dmesg | grep rtc
```
[    1.913571] rtc-ds1307 2-0068: registered as rtc0
[    1.916027] omap_rtc 44e3e000.rtc: already running
[    1.917340] omap_rtc 44e3e000.rtc: registered as rtc1
[    2.567727] PM: bootloader does not support rtc-only!
[    2.574055] rtc-ds1307 2-0068: setting system clock to 2023-11-23T18:47:03 UTC (1700765223)
```

/opt/scripts/tools/version.sh
```
git:/opt/scripts/:[4ec0c8ed26872ace02789973ffce8477bbe1fbca]
eeprom:[A335BNLTEIA02308SBI01267]
model:[TI_AM335x_BeagleBone_Black]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot SPL 2019.04-dirty (Apr 16 2021 - 18:24:02 -0700)]:[location: dd MBR]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-dirty]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot.dts]
UBOOT: Loaded Overlay:[BB-ADC-00A0.bb.org-overlays]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0.bb.org-overlays]
UBOOT: Loaded Overlay:[BB-I2C2-RTC-DS1307]
kernel:[5.4.161-bone-rt-r63]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-I2C2-RTC-DS1307.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210821.0-0~focal+20210821]
pkg:[bb-customizations]:[1.20231007.0-0~focal+20231007]
pkg:[bb-usb-gadgets]:[1.20231003.1-0~focal+20231003]
pkg:[bb-wl18xx-firmware]:[1.20230414.0-0~focal+20230414]
pkg:[kmod]:[27-1ubuntu2.1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input netdev i2c bluetooth gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 text bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait]
dmesg | grep remote
[    1.938209] remoteproc remoteproc0: wkup_m3 is available
[    2.537363] remoteproc remoteproc0: powering up wkup_m3
[    2.537393] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217148
[    2.537655] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.595956] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.597064] gpio-of-helper ocp:cape-universal: Allocated GPIO id=0 name='P9_19'
[    1.597426] gpio-of-helper ocp:cape-universal: Allocated GPIO id=1 name='P9_20'
[    1.597439] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

/boot/uEnv.txt
```
uname_r=5.4.161-bone-rt-r63

# enable uboot overlays
enable_uboot_overlays=1

##Disable HDMI
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1

##Disable Wireless
disable_uboot_overlay_wireless=1

##Disable eMMC
disable_uboot_overlay_emmc=1

##Enable I2C
optargs=quiet drm.debug=7 capemgr.enable_partno=BB-I2C0,BB-I2C2

## Disable GUI
optargs=text

## I2C2 RTC OVERLAY
dtb_overlay=/lib/firmware/BB-I2C2-RTC-DS1307.dtbo
```

## Unsuccessful 5.10.168-X runs:

dmesg | grep rtc
```
[    4.011445] omap_rtc 44e3e000.rtc: already running
[    4.013695] omap_rtc 44e3e000.rtc: registered as rtc0
[    4.013761] omap_rtc 44e3e000.rtc: setting system clock to 2023-11-23T19:13:38 UTC (1700766818)
[    4.324039] rtc-ds1307 2-0068: registered as rtc1
[   64.231322] PM: bootloader does not support rtc-only!
```

/opt/scripts/tools/version.sh
```git:/opt/scripts/:[4ec0c8ed26872ace02789973ffce8477bbe1fbca]
eeprom:[A335BNLTEIA02308SBI01267]
model:[TI_AM335x_BeagleBone_Black]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot SPL 2019.04-dirty (Apr 16 2021 - 18:24:02 -0700)]:[location: dd MBR]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-dirty]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot.dts]
UBOOT: Loaded Overlay:[BB-ADC-00A0.bb.org-overlays]
UBOOT: Loaded Overlay:[BB-I2C2-RTC-DS1307]
kernel:[5.10.168-ti-rt-r72]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-I2C2-RTC-DS1307.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210821.0-0~focal+20210821]
pkg:[bb-customizations]:[1.20231007.0-0~focal+20231007]
pkg:[bb-usb-gadgets]:[1.20231003.1-0~focal+20231003]
pkg:[bb-wl18xx-firmware]:[1.20230414.0-0~focal+20230414]
pkg:[kmod]:[27-1ubuntu2.1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input netdev i2c bluetooth gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 text bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait]
dmesg | grep remote
[    4.039903] remoteproc remoteproc0: wkup_m3 is available
[   41.960020] remoteproc remoteproc0: powering up wkup_m3
[   41.960061] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217148
[   41.961741] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   74.759188] remoteproc remoteproc1: 4a334000.pru is available
[   74.828109] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   74.759188] remoteproc remoteproc1: 4a334000.pru is available
[   74.828109] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pinctrl-single
[    3.808183] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 002: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

rbed23, 2023-11-23:
sudo find / -name BB-I2C2-RTC-DS1307* -type f
```
/lib/firmware/BB-I2C2-RTC-DS1307.dtbo
```

/lib/firmware/BB-I2C2-RTC-DS1307 (DTS)
```
/dts-v1/;

/ {

	fragment@0 {
		target-path = [2f 00];

		__overlay__ {

			chosen {

				overlays {
					BB-I2C2-RTC-DS1307 = "Sat Aug 21 14:57:17 2021";
				};
			};
		};
	};

	fragment@1 {
		target-path = [2f 00];

		__overlay__ {

			aliases {
				rtc0 = "/fragment@2/__overlay__/ds1307@68";
				rtc1 = "/ocp/rtc@44e3e000";
			};
		};
	};

	fragment@2 {
		target = <0xffffffff>;

		__overlay__ {
			status = "okay";
			#address-cells = <0x01>;
			#size-cells = <0x00>;

			ds1307@68 {
				compatible = "maxim,ds1307";
				reg = <0x68>;
				phandle = <0x01>;
			};
		};
	};

	__symbols__ {
		extrtc = "/fragment@2/__overlay__/ds1307@68";
	};

	__fixups__ {
		i2c2 = "/fragment@2:target:0";
	};
};
``

rbed23, 2023-11-23:
also - just noticed the [README](https://github.com/beagleboard/bb.org-overlays/tree/master#supported-linux-kernels) on the beagleboard bb.org-overlays ([git](https://github.com/beagleboard/bb.org-overlays)) states 'supported to 5.4.X'

:roll_eyes: shame on me -

RobertCNelson, 2023-11-23:
The issue, this reference changed... rtc1 = "/ocp/rtc@44e3e000";

RobertCNelson, 2023-11-23:
On Mobile so I can't check everything.. here is with 5.10.x https://git.beagleboard.org/beagleboard/BeagleBoard-DeviceTrees/-/blob/v5.10.x-ti-unified/src/arm/overlays/BB-I2C1-RTC-DS3231.dts?ref_type=heads#L41 we've moved to kernel  specific as there is to much churn between releases

rbed23, 2023-11-24:
@RobertCNelson thanks for pointing that out -

Referencing the [git line L41](https://git.beagleboard.org/beagleboard/BeagleBoard-DeviceTrees/-/blob/v5.10.x-ti-unified/src/arm/overlays/BB-I2C1-RTC-DS3231.dts?ref_type=heads#L41) above + the `/lib/firmware/BB-I2C2-RTC-DS1307` file I copied earlier, I made the following modification and it is now working on my BBB 5.10.168-ti-rt-r72

diff /lib/firmware/BB-I2C2-RTC-DS1307.dts /lib/firmware/BB-I2C2-RTC-DS1307-modified.dts
```
25,26c25,27
< 				rtc0 = "/fragment@2/__overlay__/ds1307@68";
< 				rtc1 = "/ocp/rtc@44e3e000";
---
> 				rtc0 = "/fragment@2/__overlay__/ds1307@68";
> 				#rtc1 = "/ocp/rtc@44e3e000";
>                               rtc1 = "/ocp/interconnect@44c00000/segment@200000/target-module@3e000/rtc@0";
```

dmesg | grep rtc
```
[    4.015344] omap_rtc 44e3e000.rtc: already running
[    4.017576] omap_rtc 44e3e000.rtc: registered as rtc1
[    4.328058] rtc-ds1307 2-0068: registered as rtc0
[    4.329143] rtc-ds1307 2-0068: setting system clock to 2023-11-24T17:19:12 UTC (1700846352)
[   71.621585] PM: bootloader does not support rtc-only!
```

jeffmelville, 2024-09-04:
Just stumbled into this same issue and fix with 5.10.