beagleboard/bb.org-overlays #235: BBAI64 - Failed loading overlay /boot/firmware/overlays/BBORG_LOAD-00A2.dtbo

scroston, 2025-01-22:
Good Morning,

I have a BBAI64 with a Load Cape attached running Linux BeagleBone-AI64 6.6.58-ti-rt-arm64-r17 #1 SMP PREEMPT_RT.

** Booting bootflow 'mmc@4f80000.bootdev.part_1' with extlinux
BeagleBone AI-64 eMMC (extlinux.conf)
1:      copy eMMC to microSD
2:      eMMC (debug)
3:      eMMC (default)
Enter choice: 3
3:      eMMC (default)
Retrieving file: /Image
Retrieving file: /initrd.img
append: root=/dev/mmcblk0p2 ro rootfstype=ext4 rootwait net.ifnames=0 quiet
Retrieving file: /ti/k3-j721e-beagleboneai64.dtb
Retrieving file: /boot/firmware/overlays/BBORG_LOAD-00A2.dtbo
Failed loading overlay /boot/firmware/overlays/BBORG_LOAD-00A2.dtbo
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
Working FDT set to 88000000
   Loading Ramdisk to 8f608000, end 8ffffadc ... OK
   Loading Device Tree to 000000008f5e8000, end 000000008f607165 ... OK
Working FDT set to 8f5e8000

Starting kernel ...

extLinux Updated to:

label eMMC (default)
    kernel /Image
    append root=/dev/mmcblk0p2 ro rootfstype=ext4 rootwait net.ifnames=0 quiet
    fdtdir /
    fdtoverlays /boot/firmware/overlays/BBORG_LOAD-00A2.dtbo
    initrd /initrd.img

It indicates it found the file, but failed on loading it. I retrieved the Device Tree files from github  branch-> dtb-6.6. 

Is there a newer version of BBORG_LOAD-00A2.dtbo for the AI-64? What am I missing here?

v/r
Scott Croston

RobertCNelson, 2025-01-22:
```
label eMMC (default)
kernel /Image
append root=/dev/mmcblk0p2 ro rootfstype=ext4 rootwait net.ifnames=0 quiet
fdtdir /
fdtoverlays /boot/firmware/overlays/BBORG_LOAD-00A2.dtbo
initrd /initrd.img
```

"/boot/firmware" is the fat32 mounted path in linux.

If you mount just the partition, you'll see it's:

```
/overlays/BBORG_LOAD-00A2.dtbo
```

So the fix is to use:

```
label eMMC (default)
kernel /Image
append root=/dev/mmcblk0p2 ro rootfstype=ext4 rootwait net.ifnames=0 quiet
fdtdir /
fdtoverlays /overlays/BBORG_LOAD-00A2.dtbo
initrd /initrd.img
```

Which will give you:

```
1:	Linux NVMe
Retrieving file: /Image
Retrieving file: /initrd.img
append: console=ttyS2,115200n8 root=/dev/nvme0n1p1 ro rootfstype=ext4 rootwait net.ifnames=0 quiet
Retrieving file: /ti/k3-j721e-beagleboneai64.dtb
Retrieving file: /overlays/BBORG_LOAD-00A2.dtbo
failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND
Failed to apply overlay /overlays/BBORG_LOAD-00A2.dtbo, skipping
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
```

We need to update BBORG_LOAD-00A2.dts...

Regards,

scroston, 2025-01-28:
Got it. Thank you very much. I'll await the update.

Regards

jadonk, 2025-01-29:
cc @lorforlinux 

This should be in our cape compatibility backlog.

lorforlinux, 2025-01-29:
@jadonk Thanks for pointing it out, I have added it to my to-do list.

RobertCNelson, 2025-01-29:
This cape arrived this week, it's just toggling 8 gpio pins, we can do it with gpioset/gpioclear

jadonk, 2025-01-30:
If we can use names to make it compatible between boards, it would be great to reuse the instructions.

scroston, 2025-05-07:
Is there any updates on the fix?